libretro/Lakka-LibreELEC #1496: vcgencmd missing dependency (Pi4 AARCH64)

duduke, 2021-09-09:
`vcgencmd` under LAKKA versions v3.3 and v3.4 on AARCH64 cannot run due to a missing dependency.
When running `/usr/bin/vcgencmd` it reports:
`-sh: vcgencmd: not found`
I believe this is due to it being compiled against `ld-linux-armhf.so.3` which is missing from /usr/lib.

dmrlawson, 2021-09-10:
Yep. This is due to using the 32bit precompiled version from https://github.com/raspberrypi/firmware

I was trying to make a new package from https://github.com/raspberrypi/userland to compile it ourselves, I'll have another go

ToKe79, 2021-09-15:
@duduke Hi, can you test build from here (choose your flavor):
https://nightly.builds.lakka.tv/PRs/pr_1500_000431/
Thank you!

PR #1500 (2021-09-17): RPi: build GPU tools
solves incompatibility on aarch64 targets with binaries provided by
bcm2835-driver for 32-bit ARM.

fixes #1496
commit 2b52be53 (2021-09-17): RPi: build GPU tools (#1500)

solves incompatibility on aarch64 targets with binaries provided by
bcm2835-driver for 32-bit ARM.

fixes #1496
closed 2021-09-17 by Ntemis