libretro/Lakka-LibreELEC #1497: Add xpadneo driver for xbox controller support

coreybruce, 2021-09-12:
Please add the xboxneo driver for xbox controllers support so we can properly use them over Bluetooth out of thew box as we are unable to install the driver ourselves since it doesn't include the tools to do so.

https://github.com/atar-axis/xpadneo

ToKe79, 2021-09-12:
I'm building images with https://github.com/ToKe79/Lakka-LibreELEC/commit/55432a24fb155091cacf5ed6fac8a2f633b9d671, can you test then, when done? Which device you can test on?

ToKe79, 2021-09-13:
@coreybruce HI, can you please test an image from here: http://nightly.builds.lakka.tv/PRs/pr_1498_000428/
Thanks!

coreybruce, 2021-09-13:
> I'm building images with [ToKe79@55432a2](https://github.com/ToKe79/Lakka-LibreELEC/commit/55432a24fb155091cacf5ed6fac8a2f633b9d671), can you test then, when done? Which device you can test on?

Raspberry pi 4 8gb

coreybruce, 2021-09-13:
> @coreybruce HI, can you please test an image from here: http://nightly.builds.lakka.tv/PRs/pr_1498_000428/
> Thanks!

Sure thing, I'll let you know how it goes! thanks for the quick reply to fix the issue :D

ToKe79, 2021-09-13:
Yes, if you have RPi4 with aarch64 currently. There are images for all devices, so if you can also test a different platform (e.g. PC), that would be also helpful. I do not have an Xbox gamepad, so I cannot test myself. But if you confirm with RPi4, we can assume that others will work as well.

coreybruce, 2021-09-14:
The devel build seems to be unstable, every time I go down in the menu's it will freeze and the pi will show heaps of IO activity so I am unable to even test if it works properly, I did try and manually update to it from stable via the img.gz file in updates as well but the xbox controller would connect for a couple of seconds and disconnect.

Did you build it and include the devel image with the xpadneo DKMS driver?

ToKe79, 2021-09-14:
yes, the images at the above URL include the `hid-xpadneo.ko` module. looks like the generic bluetooth drivers is taking over, so I need to patch the kernel. stay tuned.
however I am not sure what caused the heap of IO activity. that was after the clean flash?

ToKe79, 2021-09-14:
This image also includes patch of the kernel to stop claiming xbox gamepad by the generic hid driver:
https://nightly.builds.lakka.tv/members/vudiq/xpadneo-test/
This build is for RPi4 (aarch64).
Please, let me know, if the patch helped, I will then check/adapt the kernel-patch for other kernel versions (if necessary) and push it.

coreybruce, 2021-09-14:
Might of actually been because I upped the gpu memory allocation from the default or due to forced turbo being on (might wanna look into that) as on the stable release I was getting freezes and after changing those two which I will have to look into but I think it's the memory allocation which was the culprit it caused stability on the config.txt.

I only changed it from 300ish value (forgot the exact number) to 512 which caused a issue.

I will try again with the dev build again and not change those and let you know sorry.

coreybruce, 2021-09-14:
Hmm seems to cause some lag and freezing trying to connect it sometimes but I did get it to stay connected (at least according to the UI) but the xbox one controller is still flashing and is supposed to vibrate with a solid xbox light also also it does not even show up as a controller option but this is progress.

I know the driver does work perfectly since I use it on Linux distros like Manjaro and Manjaro Arm64 for the pi 4 and it works just fine so I am not sure what the issue is exactly but I am happy to help track it down.

ToKe79, 2021-09-14:
If you can connect to your RasPi via ssh, please, collect the logs (see [Automated collection of relevant logs](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka#automated-collection-of-relevant-logs)). I would suggest clean install (no tweaks), enable SSH, connect to cmd line, then start `createlog`. Try to connect the gamepad, then exit RetroArch by selecting `Restart RetroArch` in the main menu (do not Reboot or Shutdown). It should then return to `createlog` and give you options where/how you want to store the logs.

Unfortunately I cannot test myself without an Xbox gamepad :-/

The IO peak can be constant logging of events - hopefully there will be some data in the logs.

ToKe79, 2021-09-14:
Also please check here: https://atar-axis.github.io/xpadneo/#troubleshooting
Maybe some of those troubleshooting hints will help.

coreybruce, 2021-09-14:
Here is the log

https://transfer.sh/E8U3mL/log-2021-09-14-09.18.35.zip

coreybruce, 2021-09-14:
ok I got it working!

Make sure to have this auto configured in lakka

```
echo "options hid_xpadneo trigger_rumble_mode=2" | tee /etc/modprobe.d
/99-xpadneo-bluetooth.conf
```

```nano /sys/module/bluetooth/parameters/disable_ertm```
Change it from N to Y
If it says N, write Y to the file and try again. You may need to remove your partially paired controller from your Bluetooth settings completely before pairing again.

And when you try and connect it again it will vibrate and properly connect! :grinning: :+1:

coreybruce, 2021-09-14:
here is the new log while it is frozen on the UI 

https://transfer.sh/2jceS8/log-2021-09-14-09.42.44.zip

ToKe79, 2021-09-14:
Thank you.

Repairing: First you would have to unpair the gamepad, unfortunately there is no such feature in RetroArch menu (yet). You have to unpair it manually via command line (e.g. ssh) using `bluetoothctl` with the command `remove AA:BB:CC:DD:EE:FF`, where `AA:...:FF` is the MAC address of the game pad (see `paired-devices` output in `bluetoothctl`).

Regarding
```
echo "options hid_xpadneo trigger_rumble_mode=2" | tee /etc/modprobe.d/99-xpadneo-bluetooth.conf
```
I can add `options his_xpadneo trigger_rumble_mode=2` to the image to `/usr/lib/modprobe.d/xpadneo.conf`, hopefully it will be included then, otherwise it has to be done manually as you did (just fyi: `/etc/modprobe.d` is symlink to `/storage/.config/modprobe.d`).

It should be however persitent, I would do
```echo "options hid_xpadneo trigger_rumble_mode=2" > /storage/.config/modprobe.d/99-xpadneo-bluetooth.conf```

Changing `/sys/module/bluetooth/parameters/disable_ertm` to `Y` could be possible via `udev` rule, I will check. This is not persitent and has to be done every time after boot (but udev rule might solve it).

To have bluetooth on after reboot, you have to enable it in Settings -> Services, IIRC. It is started when you open the bluetooth menu, but it is not enabled to be started at boot.

I will check the logs later. The freezing in UI could be caused by calls to system programs such as bluetoothctl during the pairing (waiting for child process to finish, etc.).

ToKe79, 2021-09-14:
after googling:
```
echo "options bluetooth disable_ertm=1" >> /storage/.config/modprobe.d/99-xpadneo-bluetooth.conf
```
should also do the persistent change.
I made a new image, which should include:
- xpadneo module
- patch to kernel (so generic hid does not try to claim the gamepad)
- additions to modprobe conf file

The udpated image is here again:
https://nightly.builds.lakka.tv/members/vudiq/xpadneo-test/

A test with clean install (e.g. nothing in `/storage/.config`) should verify, if the changes I made are good. Thank you in advance.

coreybruce, 2021-09-14:
Ok cool so do I still need to run 
> echo "options bluetooth disable_ertm=1" >> /storage/.config/modprobe.d/99-xpadneo-bluetooth.conf

on the new image or is that already done?

ToKe79, 2021-09-14:
Everything should work out of the box, no need for `tee`/`echo`...

The Bluetooth service switch should be under the same menu item as is e.g. SSH, Samba. Bluetooth should be there as well. When switched on, Bluetooth service will start at boot and paired devices should just pair.

coreybruce, 2021-09-14:
Hey that works great! there was a few second freeze when I first connected it but than it all worked fine, even after a restart it connected automatically :smile:

coreybruce, 2021-09-14:
"The Bluetooth service switch should be under the same menu item as is e.g. SSH, Samba. Bluetooth should be there as well. When switched on, Bluetooth service will start at boot and paired devices should just pair." 

Hmm in this menu right?
[image]

ToKe79, 2021-09-14:
Great news, thank you for your time :-)
Yes, this is the menu I meant - in your photo the bluetooth service is enabled.
Few seconds freeze is from my point of view normal during the first pairing process, as there are some calls to system commands and configuration is written and the pipe waits for the return of the child process, I guess. However the next auto-pairing should be just smooth, as it is done by the already running daemon (bluetooth.service) in separate process.
Just remember, when you pair your gamepad with a different device, you will have to repair again - remove the pairing manually via command line, as I described previously, and then pair again.

closed 2021-09-14 by ToKe79

coreybruce, 2021-09-15:
Will there be a option added in the UI to eventually remove Bluetooth devices?

ToKe79, 2021-09-15:
The issue was to add the driver, this is done. Any issues with the driver itself should be first compared to behavior outside Lakka (e.g. on a regular Linux distro or other *ELEC-distros - LibreELEC, EmuELEC, ...). If the behavior is same, bug report should be made to the driver developer. If the issue is not present outside of Lakka, it would help to kind of triage the bug / compare the setups outside Lakka vs Lakka and then report findings. We cannot give support for devices that we do not own and cannot test ourselves.

ToKe79, 2021-09-15:
And for the removal of the device - there is already an issue open (#1416) and I have a patch ready, but very little time to test. If you can help, here is the image:
https://nightly.builds.lakka.tv/members/vudiq/bt_remove_device/rpi4.aarch64/
To remove the pairing, press `Start` button on the device in the list (where you usually do the pairing), you want to remove/unpair.

coreybruce, 2021-09-15:
> The issue was to add the driver, this is done. Any issues with the driver itself should be first compared to behavior outside Lakka (e.g. on a regular Linux distro or other *ELEC-distros - LibreELEC, EmuELEC, ...). If the behavior is same, bug report should be made to the driver developer. If the issue is not present outside of Lakka, it would help to kind of triage the bug / compare the setups outside Lakka vs Lakka and then report findings. We cannot blindly give support for devices that we do not own and cannot test ourselves.

Yeah the bugs I've presented with the additional fixed we had to do manually aren't something I've experienced on other Linux distro's with my main tests being on Manjaro and Manjaro for the raspberry pi 4 where I didn't have to do any additional things to make it work other than just install the driver with no drop outs or other issues besides the typical Xbox controller taking a while to show up which happens on other non Linux devices anyways but I'd like to help hunt down the bug and help fix anything that's not right including any stability issues befoee this gets pushed as a main update.

coreybruce, 2021-09-15:
> And for the removal of the device - there is already an issue open (#1416) and I have a patch ready, but very little time to test. If you can help, here is the image: [...]

Sure I can do testing and I'll even give any feed back on how the UI/button layout works on impairing and if it works well or not.

ToKe79, 2021-09-15:
Thanks. I think there is currently no feedback in the UI that the device was removed, so the test case should look like this:
1) pair a gamepad with Lakka
2) check if the gamepad works in Lakka
3) reboot Lakka to check whether the pairing stayed
4) switch off Lakka
5) pair the same controller with a different device (e.g. your Xbox)
6) switch on Lakka
7) set the gamepad to pairing mode
8) go to the bluetooth menu
9) move the cursor on the device you need to remove
10) press start button on that item (you need to use a different gamepad or keyboard - space key)
11) press ok button (either right or bottom face button, depends on your setting, or enter on keyboard) - the gamepad should pair (there can be 10 seconds lag/freeze)
I hope I can test also myself, but I appreciate your feedback as well :-)

ToKe79, 2021-09-16:
Please leave any feedback to the "unpair" feature in the related issue - #1416

coreybruce, 2021-09-17:
Hmm I let it update to that image and can confirm it is on this new image which is "Lakka-RPi4.aarch64-3.4-devel-20210916133317-e5fa7df" but I can't seem to bring up the option to unpair it, will I need to do a fresh install because that's a pain if I do with all my games on it lol

ToKe79, 2021-09-17:
Hi, you do not need to reflash the whole image, just copy the update file to /storage/.update folder (it is a hidden folder). you can do it via ssh (this will download the latest nightly build, that has this patch included, and reboot in order to apply the update):
```
cd /storage/.update && wget https://nightly.builds.lakka.tv/2021-09-17/RPi4.aarch64/Lakka-RPi4.aarch64-nightly-20210917-86441fd.tar && reboot
```
But no need to test - I have tested yesterday myself and the patch is now in Lakka source tree. There is now also a visual feedback after the pairing removed and informs the user that RetroArch needs to be restarted in order to pair a new controller. Currently the "unpair" command can be also used on unpaired devices in the list, I may fix this later, but it does not break anything I know of. Thank you for your time!

coreybruce, 2021-09-17:
Ah ok cool that's what I did and I can confirm it is using the new image in use but I can't seem to get that option to unpair. I think having another button and making a visible button on the on the screen to show users what button is to unpair it or even option screen for the device that is paired with the word paired on what devices you have connected to lakka so the user has visual feedback on what they have connected. :)

ToKe79, 2021-09-17:
Just to clarify - there is no "Unpair" button in the UI - you have to go to Settings -> Bluetooth -> select a device you want to unpair and then press the `Start` button on a gamepad or `Space` on the keyboard. Then you will get a message, the the device pairing was removed and if you want to pair that device again, you have to restart RetroArch.

coreybruce, 2021-09-17:
> Just to clarify - there is no "Unpair" button in the UI - you have to go to Settings -> Bluetooth -> select a device you want to unpair and then press the `Start` button on a gamepad or `Space` on [...]

Ahh right I see now, it works but it's not very obvious at all when you unpair a device but at least it does work, I assume this will be further worked on to be more obvious and user friendly for users with these testing?

ToKe79, 2021-10-11:
Upstream what? The patch for RetroArch (to unpair a device from the list of the Bluetooth devices) is currently only in Lakka tree applied at build time and it is kind of a hack just to improve "quality of life", in upstream it would need to rework the Bluetooth menu - having sub items for connect / unpair with their own list of devices. Currently the patch does not check if the device is paired, it just sends disconnect & unpair commands to the Bluetooth driver for the selected device - just to save the user the hassle to connect to the file system / shell of the Lakka device and do this manually.

For the driver we made no changes so there is nothing for upstream. Anyway - it seems that the driver is disrupting the experience of other Bluetooth gamepads, users have to blacklist the driver, so maybe unless this is fixed we might consider removing/replacing it.

coreybruce, 2021-10-11:
Talking about the tests we have been doing if the will go upstream to stable or not. Ah ok that's what I wanted to check, is this something being worked on instead of the hack/work around or so far no because of issues? I am always happy to help test anything

ToKe79, 2021-10-11:
The driver is now included in Lakka since v3.5 as described here: https://github.com/libretro/Lakka-LibreELEC/issues/1524#issuecomment-939733685
The patch regarding the unpairing feature is not fit for upstream (i.e. to be included in RetroArch source), but is applied at build time of Lakka.
Regarding the driver, there have been issue reports from users with non Xbox gamepads, which can be fixed on the users' side by blacklisting the driver, e.g. https://github.com/libretro/Lakka-LibreELEC/issues/1524, https://github.com/libretro/Lakka-LibreELEC/issues/1523, https://github.com/libretro/Lakka-LibreELEC/issues/1528
Not saying that the cause is in all cases the driver, but is the prime suspect.

PR #1498 (2021-09-14): add xpadneo driver for Xbox One wireless gamepad
fixes #1497