libretro/Lakka-LibreELEC #1502: Menu fails to load on generic PC using AMD GPU with Vulkan driver

GPDP1, 2021-09-19:
I am using Lakka on a PC outfitted with an Intel Core i5-7500 with integrated Intel HD 630 graphics. It worked with both GL and Vulkan drivers without major issues, but the integrated graphics were too weak to go beyond a certain rendering resolution on most cores, and it also buckled against heavy shaders such as CRT-Royale. As such, I decided to try adding a GPU, namely an AMD R7 250 I got for cheap. The card performs like a champ using the GL driver, performing well beyond the Intel graphics with 3D cores such as Mupen64Plus-Next using GLideN64, as well as Dolphin and PCSX2, allowing me to raise the internal resolution quite a bit higher while maintaining full speed. However, it fails to work with Vulkan for some reason. It just stays stuck on the Lakka boot logo, and goes nowhere, though funnily enough, if I unplug the HDMI cable from the AMD card and plug it onto the onboard HDMI port without restarting (in other words, switching to using the Intel GPU), the menu loads fine.

According to the log, it appears to keep trying to use the Intel GPU even while the cable is plugged into the AMD card:

> [INFO] RetroArch 1.9.9 (Git 36888612eb)
> [INFO] === Build =======================================
> [INFO] CPU Model Name: Intel(R) Core(TM) i5-7500 CPU @ 3.40GHz
> [INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX
> [INFO] Built: Sep 17 2021
> [INFO] Version: 1.9.9
> [INFO] Git: 36888612eb
> [INFO] =================================================
> [INFO] [Input]: Found input driver: "udev".
> [INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
> [INFO] Version of libretro API: 1
> [INFO] Compiled against API: 1
> [INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
> [INFO] [Video]: Video @ fullscreen
> [INFO] [Vulkan]: Vulkan dynamic library loaded.
> [INFO] [Vulkan]: Found vulkan context: khr_display
> [INFO] [Vulkan]: Detecting screen resolution 0x0.
> [INFO] [Vulkan]: Found GPU at index 0: Intel(R) HD Graphics 630 (KBL GT2)
> [INFO] [Vulkan]: Using GPU index 0.
> [ERROR] [Vulkan]: Failed to create KHR_display surface.
> [ERROR] [Vulkan]: Failed to set video mode.
> [ERROR] [Video]: Cannot open video driver ... Exiting ...
> [ERROR] Fatal error received in: "init_video()"
> [INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
> [INFO] [Core]: Unloading core..
> [INFO] [Core]: Unloading core symbols..
> [INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.

The GL driver does appear to see the AMD card just fine, by comparison. Using the latest nightly did not fix the problem.

Ntemis, 2021-09-19:
Is not that vulkan doesnt load is the fact that you need a tool called prime
https://wiki.archlinux.org/title/PRIME

GPDP1, 2021-09-21:
For what it's worth, I did fix this issue. Turns out Lakka was defaulting to the mesa radeon driver, and apparently my specific card (a Radeon R7 250, which is of the Southern Islands family of cards) does not play well with Vulkan while using that driver, or so I have read. I switched to the alternative amdgpu driver by appending radeon.si_support=0 amdgpu.si_support=1 to the kernel command line on syslinux.cfg and grub.cfg as documented [here](https://wiki.archlinux.org/title/AMDGPU#Enable_Southern_Islands_(SI)_and_Sea_Islands_(CIK)_support), and the Vulkan video driver now works with my card, though unfortunately not without issues. It works just fine for most cores, with the exception of Dolphin (though apparently that is a known issue with the core in general - it does not work with my Intel GPU, either) and, most distressingly, the ParaLLEl-RDP plugin on both the Mupen64Plus-Next and ParaLLEl cores. Both Angrylion (using the Vulkan driver) and GLideN64 (using GL, obviously) work fine, and ParaLLEl-RDP still works with the Intel card, so I can confirm it's not a regression. The only relevant bit the log throws out is the following:

> [libretro INFO] paraLLEl-RDP: Using RDRAM size of 8388608 bytes.
> [libretro ERROR] This device probably does not support 8/16-bit storage. Make sure you're using up-to-date drivers!

Driver issue, then?

evbo, 2022-01-10:
I have an NVidia (GTX750) graphics card and have a similar error. What's the trick to enabling Vulkan correctly on Lakka? Does it vary by GPU? 

> [ERROR] [Vulkan]: Failed to enumerate physical devices.
[ERROR] [Vulkan]: Failed to create KHR_display surface.
[ERROR] [Vulkan]: Failed to set video mode.
[ERROR] [Video]: Cannot open threaded video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"

In the link it gives two instructions: installing prime without proprietary drivers and installing prime _with_ them. Which is recommended? @Ntemis Could we please at least reopen this issue as an opportunity to improve docs if further install steps are necessary?

gouchi, 2022-01-10:
@GPDP1 
`[libretro ERROR] This device probably does not support 8/16-bit storage. Make sure you're using up-to-date drivers!`

For AMD GPU Vulkan issue it might come from this [issue](https://github.com/libretro/parallel-n64/issues/702). You may try to make some test with those [environment variables](https://github.com/libretro/parallel-n64/tree/master/mupen64plus-video-paraLLEl/parallel-rdp#environment-variables-for-development--testing) ?

@evbo 
For NVidia we don't support Vulkan as there is no open source Vulkan driver for NVidia.

evbo, 2022-01-10:
@gouchi thanks I was worried that might be the case! Is there any work around? For instance, someway to configure lakka to use proprietary drivers I install myself?

gouchi, 2022-01-11:
@evbo Yes you may try  this [version](https://lakka.tv/articles/2019/01/16/lakka-generic-nvidia-experimental-builds/) if you need Vulkan.

evbo, 2022-01-11:
@gouchi thank you, is manually replacing Nouveau with a proprietary NVidia driver a bad idea? For instance, like on Ubuntu?:
https://www.linuxcapable.com/how-to-install-or-upgrade-nvidia-drivers-on-ubuntu-21-10-impish-indri/

I'm tempted to try, but if LibElec is not as straight forward as Ubuntu in this regard that would be good to know :)

gouchi, 2022-03-02:
@GPDP1 Is it still an issue with [nightly build ](https://nightly.builds.lakka.tv/)? [Lakka v.4.x](https://nightly.builds.lakka.tv/members/vudiq/Lakka-4.0-devel/Generic.x86_64/) ? [Lakka-LE-master](https://nightly.builds.lakka.tv/members/vudiq/Lakka-LE-master/Generic.x86_64/) ?

Thank you.

GPDP1, 2022-03-05:
@gouchi Unfortunately the fan on my Radeon card up and died on me, and so while the card technically works still, I decided to remove it until I replace it with another, so I cannot tell you. All I can say is before the fan died, I did eventually get ParaLLEl-RDP working with the Radeon card by adding PARALLEL_RDP_ALLOW_EXTERNAL_HOST=0 to /storage/.cache/services/retroarch.conf, as outlined in the following thread:

https://forums.libretro.com/t/parallel-rdp-plugin-for-n64-not-available-on-pc/34776/10

gouchi, 2022-03-05:
Thank you for the feedback.

I will close this issue as there is a workaround and that Mesa Vulkan driver will improve over time if you card support this Vulkan property.

closed 2022-03-05 by gouchi