libretro/Lakka-LibreELEC #1507: Vulkan crashes libretro dolphin core on x86-64

65a, 2021-09-25:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Version: Lakka nightlies for past few weeks
Hardware: Intel NUC
Graphics: Intel(R) Iris(R) Pro Graphics 580 (SKL GT4)
Resolution: 1920x1080

How to crash:
Set video driver to Vulkan
Set dolphin driver in dolphin.ini to Vulkan
Disabled cores switching video driver

Launch any GameCube game
Core crashes retroarch by causing segmentation fault / SIGSEGV

Notable logging output. Possibly unrelated, but Vulkan thinks the resolution is 0x0
```
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [SRAM]: Skipping SRAM load..
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 31270.51 Hz.
[INFO] [Video]: Video @ 1920x1080
[INFO] [Video]: Using HW render, vulkan driver forced.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: App: Dolphin-Emu (version 5)
[INFO] [Vulkan]: Engine: Dolphin-Emu (version 2)
[INFO] [Vulkan]: Found vulkan context: khr_display
[INFO] [Vulkan]: Detecting screen resolution 0x0.
[New Thread 0x7f4fb51c1700 (LWP 668)]
[INFO] [Vulkan]: Found GPU at index 0: Intel(R) Iris(R) Pro Graphics 580 (SKL GT4)
[INFO] [Vulkan]: Using GPU index 0.
```

thread apply all bt
```
(gdb) thread apply all bt

Thread 13 (Thread 0x7f4fb51c1700 (LWP 668)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0x25b56f8)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x25b56a8, cond=0x25b56d0)
    at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0x25b56d0, mutex=0x25b56a8) at pthread_cond_wait.c:655
#3  0x00007f4fbcc7eee1 in ?? () from /usr/lib/libvulkan_intel.so
#4  0x00007f4fbcc7ea1b in ?? () from /usr/lib/libvulkan_intel.so
#5  0x00007f4fc01b5fa4 in start_thread (arg=<optimized out>) at pthread_create.c:486
#6  0x00007f4fbe612b2f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 12 (Thread 0x7f4fb59c2700 (LWP 667)):
#0  0x00007f4fbe60af03 in __GI___select (nfds=13, readfds=0x7f4fb59c1db0, writefds=0x0, 
    exceptfds=0x0, timeout=0x0) at ../sysdeps/unix/sysv/linux/select.c:41
#1  0x00007f4fa5749b21 in ?? () from /tmp/cores/dolphin_libretro.so
#2  0x00007f4fbe907de0 in ?? () from /usr/lib/libstdc++.so.6
#3  0x00007f4fc01b5fa4 in start_thread (arg=<optimized out>) at pthread_create.c:486
#4  0x00007f4fbe612b2f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 10 (Thread 0x7f4fbdae5700 (LWP 665)):
#0  0x00007f4fbe608bdf in __GI___poll (fds=0x7f4fac001470, nfds=2, timeout=5000)
    at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007f4fbffe6730 in ?? () from /usr/lib/libusb-1.0.so.0
#2  0x00007f4fbffe6f4a in libusb_handle_events_timeout_completed ()
   from /usr/lib/libusb-1.0.so.0
#3  0x00007f4fa54ac22d in ?? () from /tmp/cores/dolphin_libretro.so
#4  0x00007f4fbe907de0 in ?? () from /usr/lib/libstdc++.so.6
#5  0x00007f4fc01b5fa4 in start_thread (arg=<optimized out>) at pthread_create.c:486
#6  0x00007f4fbe612b2f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 9 (Thread 0x7f4fa7fff700 (LWP 664)):
#0  0x00007f4fbe608bdf in __GI___poll (fds=0x7f4fa7ffee60, nfds=2, timeout=-1)
    at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007f4fbffe9e96 in ?? () from /usr/lib/libusb-1.0.so.0
#2  0x00007f4fc01b5fa4 in start_thread (arg=<optimized out>) at pthread_create.c:486
#3  0x00007f4fbe612b2f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 8 (Thread 0x7f4fbe2e6700 (LWP 663)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0x2251558)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x2258940, cond=0x2251530)
    at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0x2251530, mutex=0x2258940) at pthread_cond_wait.c:655
#3  0x000000000043bdb4 in ?? ()
#4  0x0000000000558825 in ?? ()
#5  0x00007f4fc01b5fa4 in start_thread (arg=<optimized out>) at pthread_create.c:486
#6  0x00007f4fbe612b2f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7f4fbe32b600 (LWP 645)):
--Type <RET> for more, q to quit, c to continue without paging--c
#0  0x0000000000000000 in ?? ()
#1  0x00007f4fa553ee7c in ?? () from /tmp/cores/dolphin_libretro.so
#2  0x00007f4fa5540c12 in ?? () from /tmp/cores/dolphin_libretro.so
#3  0x00007f4fa5310e1f in ?? () from /tmp/cores/dolphin_libretro.so
#4  0x0000000000577db2 in ?? ()
#5  0x000000000057b387 in ?? ()
#6  0x0000000000585fa0 in ?? ()
#7  0x0000000000576f99 in ?? ()
#8  0x00000000004318d8 in ?? ()
#9  0x0000000000436d54 in ?? ()
#10 0x000000000043d8f9 in ?? ()
#11 0x000000000043eaf9 in ?? ()
#12 0x000000000051ac5e in ?? ()
#13 0x000000000042a770 in ?? ()
#14 0x00000000004d5c28 in ?? ()
#15 0x00000000004298d4 in ?? ()
#16 0x000000000042f800 in ?? ()
#17 0x0000000000432bec in ?? ()
#18 0x00007f4fbe543bab in __libc_start_main (main=0x411450, argc=2, argv=0x7ffe26c98ef8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffe26c98ee8) at ../csu/libc-start.c:308
#19 0x00000000004116ba in ?? ()
(gdb) 
```

Other vulkan cores work fine, including Parallel with angrylion.

65a, 2021-09-25:
I also commented on https://github.com/libretro/dolphin/issues/165, but I'm not sure if this is the same bug. I sort of suspect KHR_Display problems which would tend toward Lakka specific. Thanks!

Ntemis, 2021-09-25:
> I also commented on libretro/dolphin#165

Please try the 3.4 release and report back

65a, 2021-09-25:
3.4 not working is why I started trying nightly releases, so that has the same behavior.

Incidentally, here are the logs from Parallel-RDP (N64) working with Vulkan successfully. The crash in dolphin core (above) happens at approximately the time working cores say "Using semaphores for WSI acquire"
```
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: khr_display
[INFO] [Vulkan]: Detecting screen resolution 0x0.
[INFO] [Vulkan]: Found GPU at index 0: Intel(R) Iris(R) Pro Graphics 580 (SKL GT4)
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Found GPU at index 0: Intel(R) Iris(R) Pro Graphics 580 (SKL GT4)
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Using semaphores for WSI acquire.
[INFO] [Vulkan]: Using GPU: Intel(R) Iris(R) Pro Graphics 580 (SKL GT4)
[INFO] [Vulkan]: Queue family 0 supports 1 sub-queues.
[INFO] [Vulkan]: Using resolution 1920x1080
[INFO] [Vulkan]: Using RGB565 format.
[INFO] [Vulkan]: Loading stock shader.
```

Ntemis, 2021-09-25:
We need to investigate if the core is broken. 3.4 and nightly have the same
version. Can you login to retroarch discord in lakka channel?

65a, 2021-09-25:
Not a discord user. Seems core is broken, can reproduce this issue on Arch (not Lakka) with recent kernel and AMD graphics instead of intel.

65a, 2021-09-25:
Going to try a debug build on Arch and see if I can contribute a better stack trace to an issue for core.

joelddiaz, 2021-11-06:
Seeing what might be a similar failure on a Fedora 35 system on Wayland with Vulkan with a recent Retroarch build from source, and a nightly dolphin core.
```
[INFO] [Config]: Loading config from: "/home/joel/.config/retroarch/config/dolphin_libretro.cfg".                                              
[INFO] RetroArch 1.9.12 (Git 80e7e1adde)                                                                                                                                     
[INFO] === Build =======================================                                                                                                                     
[INFO] CPU Model Name: Intel(R) Core(TM) i5-8500 CPU @ 3.00GHz                                                                                                               
[INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX AVX2                                                                                                
[INFO] Built: Oct 28 2021                                                                                                                                                    
[INFO] Version: 1.9.12                                                                                                                                                       
[INFO] Git: 80e7e1adde                                                                                                                                                       
[INFO] =================================================                                                                                                                     
[INFO] [Input]: Found input driver: "linuxraw".                                                                                                                              
[INFO] [Core]: Loading dynamic libretro core from: "/home/joel/.config/retroarch/cores/dolphin_libretro.so"  
...
[INFO] [Environ]: GET_PREFERRED_HW_RENDER, video driver name: vulkan.
[INFO] [Environ]: GET_PREFERRED_HW_RENDER - Context callback set to RETRO_HW_CONTEXT_VULKAN.
[INFO] [Environ]: SET_HW_RENDER, context type: vulkan.
[INFO] Requesting Vulkan context.
[INFO] Reached end of SET_HW_RENDER.
[INFO] [Environ]: SET_HW_RENDER_CONTEXT_NEGOTIATION_INTERFACE.
...
[libretro INFO] 04:25:108 Core/Core.cpp:1049 N[COMMON]: Want determinism <- false
[libretro INFO] 04:25:108 Core/Core.cpp:239 I[BOOT]: Starting core = GameCube mode
[libretro INFO] 04:25:108 Core/Core.cpp:240 I[BOOT]: CPU Thread separate = Yes
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [SRAM]: Skipping SRAM load.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 32061.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Using HW render, vulkan driver forced.
[WARN] [Wayland]: Compositor doesn't support zwp_idle_inhibit_manager_v1 protocol!
[WARN] [Wayland]: Compositor doesn't support zxdg_decoration_manager_v1 protocol!
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: App: Dolphin-Emu (version 5)
[INFO] [Vulkan]: Engine: Dolphin-Emu (version 2)
[INFO] [Vulkan]: Found vulkan context: vk_wayland
[INFO] [Vulkan]: Detecting screen resolution 0x0.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

Segmentation fault (core dumped)
```

BigFidel, 2021-12-15:
This is happening with me as well.
X99 Chipset - i7 6850k - Radeon 580X Latest Lakka Nightly as of this writing.
Open GL results in a white screen with working sound and input (game is running) while vulkan automatically crashes and fails to load any content.

gravitone, 2022-04-02:
Yes, issue is still present in the most recent Lakka 4.x builds, nightlies.

gouchi, 2022-04-02:
So we might need to add `VK_KHR_display` support like it has [been added in PPSSPP](https://github.com/hrydgard/ppsspp/commit/e3540ccd17193e131d42290a9dd4d6bca6e5f634) so that it can work DRM/KMS ?

gravitone, 2022-04-02:
That might be the case. I'm not familiar enough with Kms/DRM to confirm
this properly. The core might have other issues currently. Did a quick
check on my windows pc and it appears to segfault the same way as on lakka
with vulkan.

gravitone, 2022-04-02:
This is the current log info I have from the lakka/pi build.

[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [SRAM]: Skipping SRAM load.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 32029.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 60.00
Hz, Game = 50.00 Hz)
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Using HW render, vulkan driver forced.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: App: Dolphin-Emu (version 5)
[INFO] [Vulkan]: Engine: Dolphin-Emu (version 2)
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 4.2".
[INFO] [Vulkan]: Using GPU index 0.
Segmentation fault

ToKe79, 2022-04-21:
vulkan should work now with dolphin in latest nightlies, can we confirm this please?

dmrlawson, 2022-04-22:
Didn't work for me on x86_64. Logs [here](https://pastebin.com/dTmSWHj7).

This was on Lakka-Generic.x86_64-nightly-v4.x-20220422-c96aa36

Ntemis, 2022-06-10:
Closing down. Try the latest 4.x version and open a new one

closed 2022-06-10 by Ntemis