libretro/Lakka-LibreELEC #1508: Unable to start RA successfully (nightly-devel Generic.x86_64)

dverbeeck, 2021-09-27:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 20210925, nightly-20210926, nightly-20210927
- What system hardware are you using? Generic x86_64
- What did you do? Updated from 20210923
- What did you expect to happen? System to boot normally and launch RA frontend
- What happened instead? System never progresses past console mode startup

I have been running nightly devel on an HP ZBook G3 fairly happily for months. The last few nightlies have resulted in the system being unable to start RA. I have seen some regressions reported due to the bump to 1.9.10, though my nightlies have been on 1.9.10 for at least a few days now if not longer, and I do not see any issues tied directly to that release of RA occuring.

In fact, when I roll back to the previous stable nightly (20210923), it's running the same git commit (d11ffdbb6b)

issue is possibly DRM/GBM related... Looks like Mesa could be broken. Logs simply cut off after gl init. I've included a log from successful startup of 20210923 as well for comparison, from the same system.
[retroarch__2021_09_26__20_55_30-nightly0927.log](https://github.com/libretro/Lakka-LibreELEC/files/7232644/retroarch__2021_09_26__20_55_30-nightly0927.log)

[retroarch__2021_09_26__20_56_26-nightly0923.log](https://github.com/libretro/Lakka-LibreELEC/files/7232645/retroarch__2021_09_26__20_56_26-nightly0923.log)

falken1975, 2021-09-27:
Reported the same here! Last working version was nightly from 24 September

Ntemis, 2021-09-27:
What is the gpu model?

Ntemis, 2021-09-28:
We need a full dmesg log please

ToKe79, 2021-09-28:
Please refer to [Troubleshooting Lakka](http://lakka.tv/doc/Troubleshooting-Lakka/) on how to collect relevant logs. Thank you!

ToKe79, 2021-09-28:
If you can test with clean build - https://nightly.builds.lakka.tv/members/vudiq/generic.x86_64-clean-master-20210928/
Maybe the cached pre-built packages do not work correctly together (e.g. libdrm / mesa updates were at different points in time).
Thanks!

by-gnome, 2021-09-28:
@ToKe79 Works!

dverbeeck, 2021-09-28:
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/7247260/dmesg.txt)

dmesg capture from 20210928 nightly, going to try the clean/master image now

dverbeeck, 2021-09-28:
Same result w/ clean, bit of a longer dump from the segfault this time
[dmesg_clean.txt](https://github.com/libretro/Lakka-LibreELEC/files/7247301/dmesg_clean.txt)

dverbeeck, 2021-09-28:
Additional breadcrumbs potentially, from verbose log of restarting the .service

This laptop does have hybrid graphics (i915) but I have that disabled in BIOS, so the dri failures are probably not a factor here(?)

[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
**MESA-LOADER: failed to open i965: /usr/lib/dri/i965_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/dri, suffi
x _dri)
failed to load driver: i965
MESA-LOADER: failed to open kms_swrast: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/li
b/dri, suffix _dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/dri, s
uffix _dri)
failed to load swrast driver**
function is no-op
function is no-op
function is no-op
function is no-op
Segmentation fault

dverbeeck, 2021-09-29:
I scrubbed core configs, playlists and presets for sanity and can confirm issue is still present in 20210929

Doesn't appear to be a physical hw issue. Nightly retroarch build compiled using libretro-super runs fine on the same physical hardware if I restore my Fedora 34 image, would providing the logs from that help out here at all?

by-gnome, 2021-09-29:
@ToKe79
Sorry, I overlooked, it doesn't work, RA started on discrete graphics after several unsuccessful attempts with Intel HD Graphics 4600.

[dmesg_pc_i5-4690.txt](https://github.com/libretro/Lakka-LibreELEC/files/7252185/dmesg_pc_i5-4690.txt)
[retroarch__2021_09_29__16_13_04.log](https://github.com/libretro/Lakka-LibreELEC/files/7252180/retroarch__2021_09_29__16_13_04.log)
[retroarch__2021_09_29__16_13_06.log](https://github.com/libretro/Lakka-LibreELEC/files/7252182/retroarch__2021_09_29__16_13_06.log)

by-gnome, 2021-09-29:
OK for Intel UHD Graphics 630.

[dmesg_pc_i3-8100.txt](https://github.com/libretro/Lakka-LibreELEC/files/7252391/dmesg_pc_i3-8100_2.txt)
[retroarch_pc_i3-8100.log](https://github.com/libretro/Lakka-LibreELEC/files/7252393/retroarch_pc_i3-8100_2.log)

by-gnome, 2021-09-29:
> Quadro K3100M

This discrete graphics (NVIDIA GK104) most likely works. The problem with integrated Intel HD Graphics 4600.

dverbeeck, 2021-09-29:
Ahh, can I simply disable that gpu via kernel cmdline for now? I know i915
allows that, not too familiar with the Intel HD driver though.

ToKe79, 2021-09-29:
I added i965 DRI driver - https://nightly.builds.lakka.tv/members/vudiq/generic.x86_64-clean-master-20210929+i965_dri/

dverbeeck, 2021-09-29:
> I added i965 DRI driver - https://nightly.builds.lakka.tv/members/vudiq/generic.x86_64-clean-master-20210929+i965_dri/

Wow that's some timing :)

Just double checked and Hybrid Graphics were set to Auto, not "disable" so it was indeed still trying to use that GPU. The most recent nightly now works, w/ the K3100M. I'm thinking that would outperform the integrated in a dance off anyway but I will definitely test your clean build out with that as well. Thanks for the quick action!

ToKe79, 2021-09-29:
I'm not sure if my change defies the addition of crocus (gallium) driver we have added - which should be "better" than the i965 (gallium) driver... Also strange that it is trying to load i965_dri, when you claim your's i915... Thank you for your time!

dverbeeck, 2021-09-29:
@ToKe79 Confirmed working! I switched BIOS setting back to hybrid enabled

20210929 nightly still failed as expected
your 0929+i965_dri build happily starts up!

Some GPU info for sanity's sake, it does appear to use the i915 driver so still curious as to the root cause. But, tis fixed!! Thanks again!

00:02.0 VGA compatible controller: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06) (prog-if 00 [VGA controller])
	DeviceName: 256
	Subsystem: Hewlett-Packard Company 4th Gen Core Processor Integrated Graphics Controller
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 32
	Region 0: Memory at ce400000 (64-bit, non-prefetchable) [size=4M]
	Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
	Region 4: I/O ports at 6000 [size=64]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: [90] MSI: Enable+ Count=1/1 Maskable- 64bit-
		Address: fee00018  Data: 0000
	Capabilities: [d0] Power Management version 2
		Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [a4] PCI Advanced Features
		AFCap: TP+ FLR+
		AFCtrl: FLR-
		AFStatus: TP-
	Kernel driver in use: i915

Lakka:~ # dmesg|grep -i i915
[    3.772539] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.774322] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    3.803151] [drm] Initialized i915 1.6.0 20200917 for 0000:00:02.0 on minor 0
[    3.817578] fbcon: i915drmfb (fb0) is primary device
[    4.989686] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device

by-gnome, 2021-09-29:
> I added i965 DRI driver

Looks like solved: [retroarch_pc_i5-4690.log](https://github.com/libretro/Lakka-LibreELEC/files/7253110/retroarch_pc_i5-4690_2.log)

ToKe79, 2021-09-29:
looks like the issue was that `i915` was in mesa in `-Ddri-drivers=` instead of in `-Dgallium-drivers` - building again (clean).

ToKe79, 2021-09-29:
Can you please confirm this image is working?
https://nightly.builds.lakka.tv/members/vudiq/generic.x86_64-clean-master-20210929+i915_gallium/
We have deprecated the i965 driver in favor of crocus (for the older intel GPUs), therefore we do not want to include it in future.
Thanks!

dverbeeck, 2021-09-29:
@ToKe79 segfaulting again w/ 20210929+i915_gallium

[dmesg_gallium.txt](https://github.com/libretro/Lakka-LibreELEC/files/7254682/dmesg_gallium.txt)

ToKe79, 2021-09-29:
can you also provide RA log [with `LIBGL_DEBUG=verbose`](http://lakka.tv/doc/Troubleshooting-Lakka/#option-b-creating-a-log-file) or with [the `createlog` script](http://lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs)?

dverbeeck, 2021-09-29:
@ToKe79

[Lakka-Generic.x86_64-3.5-devel-20210929190102-08f4712.log](https://github.com/libretro/Lakka-LibreELEC/files/7254933/Lakka-Generic.x86_64-3.5-devel-20210929190102-08f4712.log)

ToKe79, 2021-09-29:
can you add `MESA_LOADER_DRIVER_OVERRIDE=i915` for one test and `MESA_LOADER_DRIVER_OVERRIDE=crocus` as second test before `LIBGL_DEBUG=verbose`?
E.g.
```
$ MESA_LOADER_DRIVER_OVERRIDE=i915 LIBGL_DEBUG=verbose retroarch -v
```

dverbeeck, 2021-09-29:
[i915.log](https://github.com/libretro/Lakka-LibreELEC/files/7255274/i915.log
[crocus.log](https://github.com/libretro/Lakka-LibreELEC/files/7255275/crocus.log)
)

The crocus override allows RA to start successfully!

closed 2021-10-07 by ToKe79