libretro/Lakka-LibreELEC #1518: Add RPi4-RetroDreamer

ToKe79, 2021-09-30:


Ntemis, 2021-10-16:
@ToKe79 can we close this?

dmrlawson, 2021-10-17:
Tested this. arm_64bit config was missing so I pushed a fix to the branch.

Flycast and duckstation don't launch:

Flycast:
```
[INFO] RetroArch 1.9.10 (Git d11ffdbb6b)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Oct 14 2021
[INFO] Version: 1.9.10
[INFO] Git: d11ffdbb6b
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/flycast_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/Flycast/Flycast.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/Flycast/Crazy Taxi (USA).cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/Flycast/Crazy Taxi (USA).cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/Crazy Taxi (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/Crazy Taxi (USA).state".
[INFO] [Environ]: GET_LOG_INTERFACE.
retroarch: symbol lookup error: /tmp/cores/flycast_libretro.so: undefined symbol: __aarch64_ldadd4_acq_rel
```

Duckstation:
```
[INFO] RetroArch 1.9.10 (Git d11ffdbb6b)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Oct 14 2021
[INFO] Version: 1.9.10
[INFO] Git: d11ffdbb6b
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/duckstation_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/DuckStation/DuckStation.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/DuckStation/psx.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/DuckStation/Tony Hawk's Pro Skater 2 (USA).cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2.
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Remaps]: Game-specific remap found at "/storage/remappings/DuckStation/Tony Hawk's Pro Skater 2 (USA).rmp".
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/Tony Hawk's Pro Skater 2 (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/Tony Hawk's Pro Skater 2 (USA).state".
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[libretro INFO] [LibretroHostInterface] Port 1 = DigitalController
[libretro INFO] [LibretroHostInterface] Port 2 = None
[libretro INFO] [LibretroHostInterface] Port 3 = None
[libretro INFO] [LibretroHostInterface] Port 4 = None
[libretro INFO] [LibretroHostInterface] Port 5 = None
[libretro INFO] [LibretroHostInterface] Port 6 = None
[libretro INFO] [LibretroHostInterface] Port 7 = None
[libretro INFO] [LibretroHostInterface] Port 8 = None
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[libretro INFO] [Main] retro_load_game(/storage/roms/psx/Tony Hawk's Pro Skater 2 (USA).cue)
[libretro INFO] [HostInterface] Boot Filename: /storage/roms/psx/Tony Hawk's Pro Skater 2 (USA).cue
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[libretro INFO] [HostInterface] Creating 'Cubeb' audio stream, sample rate = 44100, channels = 2, buffer size = 2048
[libretro INFO] [System] Loading CD image '/storage/roms/psx/Tony Hawk's Pro Skater 2 (USA).cue'...
[libretro INFO] [System] Auto-detected console NTSC-U region for '/storage/roms/psx/Tony Hawk's Pro Skater 2 (USA).cue' (region NTSC-U)
[libretro INFO] [System] Console Region: NTSC-U/C (US, Canada)
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] [HostInterface] Searching for a NTSC-U/C (US, Canada) BIOS in '/tmp/system'...
[libretro WARN] [FindBIOSImageInDirectory] Skipping '.placeholder': incorrect size
[INFO] [Environ]: SET_MESSAGE: ERROR: No BIOS image found for NTSC-U/C (US, Canada) region
[libretro ERROR] [ReportError] No BIOS image found for NTSC-U/C (US, Canada) region
[INFO] [Environ]: SET_MESSAGE: ERROR: Failed to load NTSC-U BIOS.
[libretro ERROR] [ReportError] Failed to load NTSC-U BIOS.
[libretro INFO] [LibretroHostInterface] Running game changed:  (/)
[INFO] [Environ]: SET_MESSAGE: ERROR: System failed to boot. The log may contain more information.
[libretro ERROR] [ReportError] System failed to boot. The log may contain more information.
[ERROR] [CONTENT LOAD]: Failed to load content
```

(it appears to not find the bios file, but it works again after downgrading to latest 3.x nightly)

I suspect these core issues are not unique to RetroDreamer and are common to all RPi4 variants, so I think with the arm_64bit fix this can be regarded as done