libretro/Lakka-LibreELEC #1519: Fix parallel_n64 for arm

ToKe79, 2021-09-30:
build crashes during linking
```
build.Lakka-AMLGX.arm/toolchain/lib/gcc/armv8a-libreelec-linux-gnueabihf/10.2.0/../../../../armv8a-libreelec-linux-gnueabihf/bin/ld.gold: error: ./mupen64plus-core/src/r4300/new_dynarec/arm/linkage_arm.o: multiple definition of 'stop'
build.Lakka-AMLGX.arm/toolchain/lib/gcc/armv8a-libreelec-linux-gnueabihf/10.2.0/../../../../armv8a-libreelec-linux-gnueabihf/bin/ld.gold: ./mupen64plus-core/src/r4300/r4300.o: previous definition here
collect2: error: ld returned 1 exit status
```

Ntemis, 2021-10-08:
https://github.com/libretro/mupen64plus-libretro
Core is depreciated, you can safely  clean this one out

dmrlawson, 2021-10-11:
I think this might be https://github.com/libretro/parallel-n64/

closed 2021-10-12 by ToKe79