libretro/Lakka-LibreELEC #1522: No network connection unless the system is connected to a tv

pitsi, 2021-10-03:
- Which version of Lakka are you using?
3.4 and I hope to upgrade to 3.5. Sorry but I can't use nightly.

- What system hardware are you using?
Rpi zero (non w) and rpi 3b+. The 3b+ uses the aarch64 build that was first released with lakka 3.2 and I use an external wireless adapter for the zero. Both are wirelessly connected to my network.

- What did you do? 
When a new version of lakka is released, I connect to each one via ssh, download each image in the update folder, reboot and let the system do the upgrade. Then I connect again and poweroff each system.
I do not have them permanently connected to my tv, because I use them on occasion.

- What did you expect to happen?
I expected each system to show up in my router's connected devices. That is how get their ips and use them for the ssh connections.
I place both systems on or near my desk when I have to update them, like 50-100cm away from the router, so bad wifi coverage is not an issue here.

- What happened instead?
None of the devices showed up. Then I connected the 3b+ via ethernet but it did not show up either. Then I tried booting libreelec on the 3b+, which I have on a different sd card. It showed up on my router's connected devices for both wired and wireless interfaces. I connected to it via ssh, powered it off and connected it to the tv, hoping to see something pop up in the boot messages/
It booted as usual and when it reached the retroarch interface, it showed up as a connected device in the router. I could ssh to it with no issues. The same thing happened with the zero as well.

- More info
I do the described update procedure every time there is an upgrade to do. I use lakka on both since 2.3.2 (if not from a previous version at the zero). I did a fresh install on both systems when lakka 3.0 was released and one more on the 3b+ when the aarch64 build was released. I did not have that issue for all the upgrades inbetween (3.0 to 3.1 to 3.2 to 3.3 to 3.4).
To be honest, I can't remember if there was an issue after the upgrade to 3.4.
The room that my tv is at has poor wifi coverage. Fortunately, I use a great wifi adapter for the zero (tp link w722n v1, the one with atheros chipset) and that solves the wifi coverage issue by a lot, but the 3b+ built in wifi chip is not that great.

Right now I have connected the zero to the tv and I am downloading the upgrade there. I hope it will finish soon.

Sorry for not attaching any log, but a network connection is required to do so.

pitsi, 2021-10-03:
The update to 3.5 on the zero finished a few minutes ago. I removed it from the tv, got it back on my desk and now it works like it used to, without being connected to the tv. I rebooted it 2-3 times to be sure, and it kept showing up as connected.

On the other hand, even connecting to the 3b+ is a tough task because of the bad wifi coverage and the lag. I typed `cd .update` in putty and the "phrase" was completed 10-15 seconds later! I typed `poweroff` and even though the devices was powered off, putty never realised that the connection was ended and was just staying open!

If anyone has any idea on how to update the 3b+ to 3.5 without using its network interfaces, please share.

pitsi, 2021-10-03:
I removed the sd card and connected it to the pc. I was hoping to download the update via wget, but it fails
```
$ wget https://le-builds.lakka.tv/RPi3.aarch64/Lakka-RPi3.aarch64-3.5.img.gz
--2021-10-03 12:00:16--  https://le-builds.lakka.tv/RPi3.aarch64/Lakka-RPi3.aarch64-3.5.img.gz
Resolving le-builds.lakka.tv (le-builds.lakka.tv)... 104.21.52.91, 172.67.197.143, 2606:4700:3032::ac43:c58f, ...
Connecting to le-builds.lakka.tv (le-builds.lakka.tv)|104.21.52.91|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 495240427 (472M) [application/octet-stream]
--2021-10-03 12:00:16--  https://le-builds.lakka.tv/RPi3.aarch64/Lakka-RPi3.aarch64-3.5.img.gz
Reusing existing connection to le-builds.lakka.tv:443.
HTTP request sent, awaiting response... 200 OK
Length: 495240427 (472M) [application/octet-stream]
Lakka-RPi3.aarch64-3.5.img.gz: Permission denied

Cannot write to ‘Lakka-RPi3.aarch64-3.5.img.gz’ (Permission denied).

```

ToKe79, 2021-10-03:
Change ownership of the permissions of the update folder to 777.

pitsi, 2021-10-03:
It did not let me do it as a regular user, but it let me do it as root. Then I switched back to regular user and now I am downloading it. I will report when it finishes updating.
Thanks a lot :)

Meanwhile, I made a big mistake. My neighbor's wifi is also visible at the room of the tv  It also has the same bad coverage. 
So I connected to it and tried updating lakka via retroarch. It did connect and downloaded the index file for the update. But when I clicked the update to 3.5, it stuck to 0% and it popped up a message that the download failed. 
The 3b+ will probably reconnect to it next time it boots, so how can I delete it or something?

ToKe79, 2021-10-03:
The update process should delete any corrupted update file, so rebooting should be safe.

pitsi, 2021-10-03:
Sorry, I wasn't clear. How can I delete the neighbor's wifi network from the stored ones?
Good thing though that the corrupted file is deleted automatically.

closed 2021-10-03 by pitsi