libretro/Lakka-LibreELEC #1523: Menu Toggle Button Broken

paqman3d, 2021-10-03:
- Which version of Lakka are you using?

Lakka 3.5

- What system hardware are you using?

Asus Tinkerboard

- What did you do?

Press the HOME button on my Retrobit Sega Genesis Bluetooth Controller

- What did you expect to happen?

Enter the home menu

- What happened instead?

Nothing. Just upgraded from 3.4 when this worked. Checked Input Settings/Hotkeys and the menu toggle is set to "Button 10" like it was on 3.4. Re-entering the button in the hotkey menu assured me it works, but it just won't register. Instead I have to press and hold start for 2 seconds to enter the menu. Not exactly sure what broke lol.

The autoconfig is set for the 8bitdo M30 and all other buttons work fine.

paqman3d, 2021-11-11:
Okay, so I ended up installing 3.5.2 on my Pi3 and notice the same issue with the 8bitdo SN30's menu button. However, unlike the Retrobit Sega pad, the SN30 didn't work at all (including wired) until I blacklisted the xpadneo stuff. 

I'm providing the output for lsmod from my pi3, but like I said -- same issue:

`Lakka (official): 3.5.2 (RPi3.aarch64)
Lakka:~ # lsmod
Module                  Size  Used by
evdev                  28672  2
joydev                 28672  1
hid_xpadneo            24576  0
ff_memless             20480  1 hid_xpadneo
hidp                   32768  1
udf                    98304  1
bnep                   28672  2
hci_uart               49152  1
btbcm                  24576  1 hci_uart
bluetooth             471040  29 hidp,hci_uart,btbcm,bnep
ecdh_generic           16384  1 bluetooth
ecc                    32768  1 ecdh_generic
snd_seq_dummy          16384  0
snd_seq                81920  2 snd_seq_dummy
snd_seq_device         20480  1 snd_seq
brcmfmac              258048  0
brcmutil               20480  1 brcmfmac
cfg80211              864256  1 brcmfmac
bcm2835_codec          49152  0
v4l2_mem2mem           40960  1 bcm2835_codec
bcm2835_isp            32768  0
videobuf2_dma_contig    24576  2 bcm2835_codec,bcm2835_isp
videobuf2_memops       20480  1 videobuf2_dma_contig
bcm2835_mmal_vchiq     40960  2 bcm2835_codec,bcm2835_isp
sr_mod                 32768  1
videobuf2_v4l2         32768  3 bcm2835_codec,v4l2_mem2mem,bcm2835_isp
cdrom                  45056  2 udf,sr_mod
videobuf2_common       61440  4 bcm2835_codec,videobuf2_v4l2,v4l2_mem2mem,bcm283                                                                                                                                                             5_isp
videodev              311296  5 bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4                                                                                                                                                             l2_mem2mem,bcm2835_isp
snd_bcm2835            28672  0
mc                     57344  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_                                                                                                                                                             common,v4l2_mem2mem,bcm2835_isp
vc_sm_cma              36864  2 bcm2835_mmal_vchiq,bcm2835_isp
sdhci_iproc            16384  0
vchiq                 339968  3 vc_sm_cma,snd_bcm2835,bcm2835_mmal_vchiq
rfkill                 36864  5 bluetooth,cfg80211
fuse                  131072  3
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
overlay               122880  7
sch_fq_codel           20480  3
`

ToKe79, 2021-11-13:
Sorry, there was a typo above, the module name is `hid_xpadneo`, not `xpadneo`, so you have to put `blacklist hid_xpadneo` to the `blacklist.conf` file.

paqman3d, 2021-11-15:
While I still haven't tested this on the Tinkerboard (it's unhooked at the moment lol), I made the changes on my pi3 and now the 8Bitdo SN30 doesn't work. It will sync fine, but there's no on screen prompt or control. When I reverted it back to the typo, it returned to the previous behavior.

paqman3d, 2021-11-19:
I tested the hid_xpadneo edit on the Tinkerboard as well and it also broke the gamepad (Retrobit sega genesis) working. Reverted back to 3.4 again. 

Also of note that the Retrobit and 8bitdo pads are paired for xinput.

RobLoach, 2021-12-09:
Could it be the autoconfigs? Are you able to do inputs and export to see if they're different? https://github.com/libretro/retroarch-joypad-autoconfig

paqman3d, 2022-02-14:
Just to update, I've acquired another Pi since I opened this issue and even on a clean install of the latest version, I'm still getting the broken menu/home button on the 8Bitdo SN30 and Retrobit Sega Genesis pads.

Things I've tried:

- On my Pi running Lakka 3.5.2, I replaced the autoconfig with an exported autoconfig from my working Lakka 3.4. No luck.

- Used an autoconfig from the Github page with no change
 
- I've tested that the menu button is correctly mapped in the Input settings. The funny thing is, the button works in the menu if I'm making keybinds. It just won't work in game.

Here's the full config from my 3.4 install that works fine:

```
input_device = "8Bitdo SN30 Pro"
input_device_display_name = "Xbox 360 Controller"
input_driver = "udev"
input_vendor_id = "1118"
input_product_id = "736"
input_b_btn = "0"
input_y_btn = "2"
input_select_btn = "6"
input_start_btn = "7"
input_up_btn = "h0up"
input_down_btn = "h0down"
input_left_btn = "h0left"
input_right_btn = "h0right"
input_a_btn = "1"
input_x_btn = "3"
input_l_btn = "4"
input_r_btn = "5"
input_l2_axis = "+2"
input_r2_axis = "+5"
input_l3_btn = "8"
input_r3_btn = "9"
input_l_x_plus_axis = "+0"
input_l_x_minus_axis = "-0"
input_l_y_plus_axis = "+1"
input_l_y_minus_axis = "-1"
input_r_x_plus_axis = "+3"
input_r_x_minus_axis = "-3"
input_r_y_plus_axis = "+4"
input_r_y_minus_axis = "-4"
input_menu_toggle_btn = "8"
input_b_btn_label = "A"
input_y_btn_label = "X"
input_select_btn_label = "Back"
input_start_btn_label = "Start"
input_up_btn_label = "D-Pad Up"
input_down_btn_label = "D-Pad Down"
input_left_btn_label = "D-Pad Left"
input_right_btn_label = "D-Pad Right"
input_a_btn_label = "B"
input_x_btn_label = "Y"
input_l_btn_label = "Left Bumper"
input_r_btn_label = "Right Bumper"
input_l2_axis_label = "Left Trigger"
input_r2_axis_label = "Right Trigger"
input_l3_btn_label = "Left Thumb"
input_r3_btn_label = "Right Thumb"
input_l_x_plus_axis_label = "Left Analog Right"
input_l_x_minus_axis_label = "Left Analog Left"
input_l_y_plus_axis_label = "Left Analog Down"
input_l_y_minus_axis_label = "Left Analog Up"
input_r_x_plus_axis_label = "Right Analog Right"
input_r_x_minus_axis_label = "Right Analog Left"
input_r_y_plus_axis_label = "Right Analog Down"
input_r_y_minus_axis_label = "Right Analog Up"
input_menu_toggle_btn_label = "Guide"
input_right_axis = `"+6"

```

ekx, 2022-04-29:
I have the same Issue on 4.2 with a 8bitdo pro 2 controller. Setting the hotkey works but when I'm in game and press the button nothing happens. This is with the controller in xinput mode.

Ntemis, 2022-06-10:
Switch to d-input and try again

closed 2022-06-10 by Ntemis