libretro/Lakka-LibreELEC #1524: Face Buttons Reversed After 3.5 Update

mediamoshpit, 2021-10-04:
- Which version of Lakka are you using?
3.5

- What system hardware are you using?
Raspberry Pi 4B

- What did you do?
Updated from 3.4 to 3.5 and pressed face buttons on xinput-based controller (8bitdo SN30).

- What did you expect to happen?
For them to be mapped correctly

- What happened instead?
A=B, X=Y throughout the menus and in games

mediamoshpit, 2021-10-04:
Remapping input actually works until an autoconfig is saved. At that point, when the manual mapping is cleared and the autoconfig is loaded, the controls reverse again.

mediamoshpit, 2021-10-04:
Downgrading to 3.4 resolved the issue for the time being.

mediamoshpit, 2021-10-05:
I believe I've found the cause of this issue. According to the xpadneo github documentation, this particular driver treats 8bitdo controllers differently by only allowing the SNES layout through a quirk flag. 

I've got several different 8bitdo controller models and would rather not do this for each of them. Is there any way to use the driver found in 3.4? Another option would be to use the controller in another mode, but then I would lose out on having force feedback.

gouchi, 2021-10-05:
May you  can try something like ?

`echo "options hid_xpadneo quirks=MAC-controller:quirk-ids" > /storage/.config/modprobe.d/8bitdo.conf
`

Otherwise, you can try to unload the module `rmmod xpadneo` you can get the module name with ` lsmod | grep xpadneo`.

If it is working you can blacklist it with 

`echo "blacklist xpadneo" > /storage/.config/modprobe.d/blacklist.conf
`

[Source](https://github.com/atar-axis/xpadneo#8bitdo-controllers) 
[Source quirk bits](https://github.com/atar-axis/xpadneo/issues/236#issuecomment-669814219)

mediamoshpit, 2021-10-05:
I got it working by connecting the controller in 'MacOS Mode' to a Mayflash Magic S pro in 'Xinput Mode. Everything works as you'd expect, even force feedback. I also tried using a Brook Wingman XB, but didn't have the same luck in any mode.

It's a shame 8Bitdo controllers no longer work with Lakka by default, but the Mayflash adapter works well enough. This will likely be my personal solution, but if need be, I'll be happy to test out any future alternatives for the benefit of others.

carnage-asada, 2021-10-07:
I am also experiencing the same issue on Lakka 3.5, Raspberry Pi 4B Arch64.
The A and B buttons on my 8Bitdo SN30 Pro+ are also switched around. Using 3.4 in the meantime.

ToKe79, 2021-10-07:
As [already pointed out](https://github.com/libretro/Lakka-LibreELEC/issues/1524#issuecomment-934358995) by @gouchi you have to fix this on your side. This is side effect of the added driver to support Xbox Wireless Gamepads.

mediamoshpit, 2021-11-01:
I wanted to wait a while before I approached the devs of xpadneo to see if anything else came to light. For one, I originally missed that it reverses the shoulder buttons on the Mayflash adapter I was using. Then, last night I had a Halloween get together where we were playing horror-themed retro games. 

I told anyone who was interested to bring their own controller and out of everyone, three people had controllers (non 8bitdo) that were reversing in one way or another. Out of those three, one was able to switch to dinput (an older Logitech). I tried blacklisting xpadneo as noted above, but it was still active after a reboot. We finally just cut things short and watched The Monster Squad. 

In the end we honestly had a good time. Our night was far from ruined. I only bring it up here to say that the problems likely go beyond 8bitdo and its worth keeping that in mind if users see issues down the road with random xinput compatible controllers. 

Myself, I'm now just looking for a way to disable it outright. I would have loved to keep it enabled for any of my friends who stop by with an Xbox One controller,  but it's just to much of a wildcard to troubleshoot with people over. Any suggestions?

ToKe79, 2021-11-01:
Thank you for your report.
The blacklisting should be probably `blacklist hid_xpadneo` - but first check if the driver/module is loaded by `lsmod`. most probably `hid_xpadneo` is correct (`modprobe xpadneo` throws error - module not found, `modprobe hid_xpadneo` loads the module).

mediamoshpit, 2021-11-01:
Well, `blacklist hid_xpadneo` definitely did the trick. However, after the system was rebooted and the controller successfully paired, no input was acknowledged. I even tried to remap, but it only ever awaited input presses from the controller as it went through the possible buttons. I then commented out the blacklist, rebooted and found that I of course had control again.

ToKe79, 2021-11-02:
The swapped face buttons is actually a known issue / expected behavior with 8BitDo controllers - see https://atar-axis.github.io/xpadneo/#8bitdo-controllers

So you can do:

```
echo options hid_xpadneo quirks=E4:17:D8:xx:xx:xx:32 > /storage/.config/modprobe.d/8bitdo-fix.conf
```
> where you replace xx:xx:xx with the values from your controller MAC (as shown in dmesg). The value 32 enables Nintendo layout. If you’ll want to add other quirk flags, simply add the values, e.g. 32 + 7 (default quirks for 8BitDo) = 39. After changing this, reload the driver or reboot.

Does this fix the issue?

mediamoshpit, 2021-11-03:
I actually tried that approach first. With 32 on the end, the controller began to rumble and wouldn't stop until the controller was powered down. During that time button presses were unresponsive.

ToKe79, 2021-11-03:
> e.g. 32 + 7 (default quirks for 8BitDo) = 39

I do not know if the quirks are documented somewhere, but have you tried `39`?

ToKe79, 2021-11-03:
Okay, here are the [quirks/flags](https://atar-axis.github.io/xpadneo/#configuration-1)

And the flags are a sum of values of various "switches":
> 1 if your controller does not support pulse parameters (i.e., 8BitDo controllers)
2 if your controller does not support trigger rumble (most clones in compat-mode)
4 if your controller does not support individual motor programming (i.e., 8BitDo controllers)
8 if your controller supports hardware profiles (auto-detected, do not set manually)
16 if your controller boots in Linux mode (auto-detected, do not change manually)
32 if you prefer to use Nintendo button mappings (i.e., 8BitDo controllers, defaults to off)
64 if your controller has a awkwardly mapped Share button (auto-detected, do not set manually)

mediamoshpit, 2021-11-04:
I'm of course happy to go through these for the sake of other users, but I have to admit that none of this is practical for my particular use case. If I have my Pi set up at any of my friend's places and one of their controllers isn't working, I can't really use ssh on their network, get their individual controller's mac addy, then start a lottery of mod quirks before we can start playing.

Also, the quirks don't appear to offer a solution to my particular Mayflash S Pro adapter (where the shoulder buttons are reversed) which I use for countless arcade sticks and controllers. Again, I'm happy to help out with troubleshooting the 8bitdo issue seeing as how I own one, but since the driver seems to have issues that affects several other xinput compatible control devices, I would greatly appreciate continued help with deactivating the xpadneo driver outright.

I'll report back soon with my findings on the mod flags.

mediamoshpit, 2021-11-04:
OK, I tried 32,33,36,37 and 39. The rumble was in a continuous active state in every instance except for 37 and 39. Unfortunately however, the face buttons remained incorrectly swapped on each occasion.

ToKe79, 2021-11-04:
If remapping / manual mapping of buttons still works, then I do not see any breaking issue. The issue can also be in the default autoconfig used by RetroArch - the config has the mapping swapped. So I hope you are OK with the things as they are - you just go to input settings, change/fix the buttons, save the config and next time the correct mapping should be applied.

We are not the developer of the said driver. So if you want to pursue the issue further, I would recommend opening an issue with the driver developer.

Thank you for your time spent on the testing.

closed 2021-11-04 by ToKe79

mediamoshpit, 2021-11-04:
That's perhaps the biggest rub in this whole issue. The controls are oddly immune to being remapped in the input menu. They go right back to being reversed. Same thing went for all the various brands of controllers I have previously mentioned. In fact, if I could've just remapped them normally, I never would've opened this issue.

mediamoshpit, 2021-11-04:
I believe I solved it. The issue seems to stem from the auto configs that are baked into Lakka by default. Remapping and saving only lead to the configs snapping back to their install state after 'reset to default controls was selected'. Editing them by hand in a text editor and saving them did the trick.

As for my Mayflash S Pro and the other controllers my friends were using that didn't have an autoconfig ship with Lakka, I can attest that the Mayflash reported itself as a generic Xbox controller and I imagine the others did as well.