libretro/Lakka-LibreELEC #1531: Test upgrade from 3.x to 4.0

ToKe79, 2021-10-12:
On following targets automatic upgrade should work (i.e. on Lakka 3.x place the update file in `/storage/.update`, reboot for upgrade, after upgrade boot to RetroArch):
- [ ] Generic (32-bit)
- [ ] Generic (64-bit)
- [ ] RPi (arm)
- [ ] RPi2 (arm)
- [ ] RPi3 (aarch64)
- [ ] RPi4 (aarch64)
- [ ] RetroDreamer (aarch64)
- [ ] PiBoy DMG (aarch64)
- [ ] Switch (aarch64)
- [ ] Allwinner H3 / Capcom Home Arcade (arm)
- [ ] NXP iMX6 (arm)

On following targets automatic upgrade will not be possible, but adding empty `.nocompat` file in the update folder (`$ touch /storage/.update/.nocompat`) should force the upgrade:
- [ ] Allwinner A64 (arm &rarr; aarch64)
- [ ] Allwinner H6 (arm &rarr; aarch64)
- [ ] Amlogic AMLGX (arm &rarr; aarch64)
- [ ] Tinkerboard / MiQi (merged into `RK3288`)
- [ ] Rockchip RK3328 (arm &rarr; aarch64)
- [ ] Rockchip RK3399 (arm &rarr; aarch64)

Latest images can be downloaded from https://nightly.builds.lakka.tv/members/vudiq/Lakka-4.0-devel/

GavinDarkglider, 2022-02-14:
@ToKe79 I managed to fix the updater in 4.x builds, but I am not going to bother making anymore changes to 3.x tree, as it appears to be completely broken on the switch at this point in time.

ToKe79, 2022-04-01:
as Lakka 4.0 is released, this becomes obsolete.

closed 2022-04-01 by ToKe79