libretro/Lakka-LibreELEC #1538: HDMI output issue (RPi4B) Lakka Stable aarch64 3.5.1

Rstation4, 2021-10-19:
Raspberry Pi 4 B (4gb of Ram)
Lakka Version: Stable RPI4.aarch64-3.5.1

There is a issue with the video output not outputting video on the hdmi port. It does however work once you unplug and re-plug the the micro end of the HDMI cable. Even if I un-commenting that first "# hdmi_force_hotplug=1" option in the config. This was a clean install of this Lakka version on a new sd card,so those config settings have been left  to their default settings.

As for testing if it would work if I comment the "Hdmi_force_hotplug=1" it will show video output after I unplug the micro end of the HDMI cable and then re-plug it.

ToKe79, 2021-10-21:
is the same issue present in the latest nightly?

ToKe79, 2021-10-21:
Please confirm with latest nightly and if the issue is still present, please test with removed patch (that enables CRT/analog video output) from here: https://nightly.builds.lakka.tv/members/vudiq/rpi-no_crt_analog_patch/

Rstation4, 2021-10-21:
Does this mean I have to do a clean install?

ToKe79, 2021-10-21:
no, you can update using the .tar in update folder.

Rstation4, 2021-10-21:
I updated it to the version you provided. It still is not displaying anything, its a blank black screen but it goes back to displaying normally when you unplug and re-plug the hdmi cable

cmdline.txt attatched below
[cmdline.txt](https://github.com/libretro/Lakka-LibreELEC/files/7392634/cmdline.txt)

config.txt
[config.txt](https://github.com/libretro/Lakka-LibreELEC/files/7392637/config.txt)

distroconfig.txt
[distroconfig.txt](https://github.com/libretro/Lakka-LibreELEC/files/7392640/distroconfig.txt)

Rstation4, 2021-10-31:
> @Rstation4 remove/comment out following lines from `distroconfig.txt`:
> 
> ```
> hdmi_max_pixel_freq:0=200000000
> hdmi_max_pixel_freq:1=200000000
> ```
> 
> These lines are there to avoid switching to 4K on capable displays and display at 1080p, but maybe it breaks compatibility with non FHD/UHD displays (at least I suspect that the Vizio is HD/720p TV).

That did not work. Since there is a newer 3.5.2 I will update to that and see how it performs.

ToKe79, 2021-10-31:
There is also a suggestion to update the RPi firmware to solve the no display issue.

Rstation4, 2021-11-02:
> There is also a suggestion to update the RPi firmware to solve the no display issue.

I just checked and I have the latest stable release of the eeprom firmware for the Raspberry Pi 4B.

ToKe79, 2021-11-02:
If you uncomment those two lines and try on the Samsung 4K, is it still laggy? It might have been laggy due to displaying in 4K. Those two lines should allow 1080p at most, so that is their reason for being in distroconfig.txt (there is no need to run Lakka at 4K on RPi devices anyway).

ab77, 2021-11-14:
Possibly same? problem here with 1080/60Hz HDMI pico projector AAXA P300 running on RPi4 aarch64.

The last known version to boot with this setup is Lakka 3.3.

I've tried 3.5, 3.5.1, 3.5.2 and now 3.6. The projector input momentarily detects 1080 60Hz signal, but then displays no signal.

Tried commenting out `hdmi_max_pixel_freq`, adding `hdmi_safe=1`, etc. to no avail.

Rstation4, 2021-11-15:
I've updated to the newer **3.6 version** and it still has this issue on my Vizio 1080P  Full HD TV. The latest stable firmware for the Raspberry Pi 4 **eeprom** is still that from the April 2021 release. I only use the stable branch **firmware** for my Raspberry Pi 4 unit.

4K at least in my area of California seems to be the norm that has replaced most HDTV sets that many of us own. 8K actually has been around for a while, though I do see more 8K TVs being available. Even though I see why trying to focus on making RA/Lakka compatible with 4k Tv's over the decline of HDTV's but in the case of SBC devices 4K is not needed for them at least for Emulation.

Will update when I try commenting out those 2 line's again.

Rstation4, 2021-11-24:
I have some logs from Lakka OS when I had it runnng on  3 devices, one Full HD TV (old 1080P tv), a 4K computer monitor (samsung U28E570D) and a 4K samsung TV (I've tend to see people I know own these types or something similar in TV OS). It seems not just display has issues but something which might be tied into display settings has also affected framerates or refresh rates as there was performance issues on 4K computer monitor with Lakka OS running.

Log 1
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/7592111/retroarch.log)

ToKe79, 2022-02-24:
seems like the issue is due to updated kernel/firmware. [this image](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-3.7-dev-citra+kernel-rollback/) has older kernel/firmware (as in Lakka 3.5) and one user already confirmed this helped them. as a bonus there is citra core (but I'm not sure if the core will even work).

ToKe79, 2022-03-07:
is it still an issue with [latest release](https://le-builds.lakka.tv/) and [dev build](https://nightly.builds.lakka.tv/latest/)?

ab77, 2022-03-26:
I've tried v4.0 and this version still has the issue described in [comment](https://github.com/libretro/Lakka-LibreELEC/issues/1538#issuecomment-968390126).

I've also tried using HDMI to CSI-2 bridge for a laugh, which works fine with v3.3 and it also came up with a blank screen with Lakka v4.0. I didn't try the dev. build.

ToKe79, 2022-03-27:
It is a known issue, you can use this workaround by @Brunnis: https://github.com/raspberrypi/firmware/issues/1648#issuecomment-986517797

closed 2022-06-10 by Ntemis