libretro/Lakka-LibreELEC #1544: Retroflag GPi Raspberry pi Zero 2 W AArch64 image

coreybruce, 2021-11-03:
Hi I hope this isn't the incorrect place to post and I know it is brand new but I just wanted to ask if we could get a Retroflag GPi Raspberry pi Zero 2 W AArch64 image to support the Pi Zero W in the Retroflag GPi as it also supports the new board and got the Retroflag GPi for the PI Zero 2 W.

Thanks again guys for all your help.

Cicirifu, 2021-11-05:
Ive tried implementing this but haven't gotten fully there, so Im sharing my findings

* AArch64 requires use of the VC4 KMS drivers, unless I am mistaken. Thus, my jouney began.
* This means none of the config.txt display configuration works;
* You need to implement hardware support for the GPi LCD;
* The GPi screen requires falling edge HSync/Vsync triggers as well as Mode 6 RGB666;
* Turns out the VC4 driver does not signal these DRM hints to the GPU. Had to implement this as well.
* and after all that: 90 degree rotation is broken, for which I have no fix yet. 

Gonna build myself a 32-bit instead and use the firmware GPU driver instead of the open source VC4 one...

SethBarberee, 2021-11-05:
@Cicirifu We did try to use the VC4 driver on GPI Case 32-bit bit ran into the aforementioned rotation issues...

* GPi LCD actually works fine with VC4 (aside from the rotation). Most of the settings are already handled in the overlay
* There is still a bug open on Retroarch for getting the hints from the driver. When retroarch can get those hints, it can handle the rotation for us.

dmrlawson, 2021-11-05:
> * AArch64 requires use of the VC4 KMS drivers, unless I am mistaken. Thus, my journey began.

There's no requirement for using kms with aarch64. If you look at PiBoy and RetroDreamer we're using aarch64 with fkms for now, which allows us to set up the LCDs. I don't really know anything about the GPi case but maybe this would work?

Cicirifu, 2021-11-05:
I could not get display output however until I added support for the inverted Display Enable. I began with a fairly recent kernel but perhaps I made a mistake there. 

As for FKMS, yeah, thatd work, but that still leaves you with the rotation issues.

coreybruce, 2021-11-05:
Hmm interesting findings, hopfully these issues can be resolved. Whats the difference between the arm7l 32bit pi zero and pi zero w image to the Retroflag GPi and doing a pi zero w 2 aarch64 image?

Cicirifu, 2021-11-05:
The GPICase build unlike the Pi1/2/3 builds does not use vc4-kms or vc4-fkms but the firmware driver. This allows the screen to be more easily configured. Other than the display issues, it's fairly straight forward to just take the Pi3 device as a base as the Pi02 is essentially a Pi3.

Cicirifu, 2021-11-05:
For the record here are the patches and overlay I needed to get it to work.

`diff --git a/drivers/gpu/drm/panel/panel-simple.c b/drivers/gpu/drm/panel/panel-simple.c
index 8becb0e31814..b9c387e93955 100644
--- a/drivers/gpu/drm/panel/panel-simple.c
+++ b/drivers/gpu/drm/panel/panel-simple.c
@@ -3015,6 +3015,45 @@ static const struct panel_desc okaya_rs800480t_7x0gp = {
 	.bus_format = MEDIA_BUS_FMT_RGB666_1X18,
 };
 
+static const struct drm_display_mode retroflag_gpi_mode = {
+	.clock = 6400,
+	.hdisplay = 240,
+	.hsync_start = 240 + 38,
+	.hsync_end = 240 + 38 + 10,
+	.htotal = 240 + 38 + 10 + 20,
+	.vdisplay = 320,
+	.vsync_start = 320 + 20,
+	.vsync_end = 320 + 20 + 4,
+	.vtotal = 320 + 20 + 4 + 4,
+	.flags = DRM_MODE_FLAG_PVSYNC | DRM_MODE_FLAG_PHSYNC,
+};
+
+static const struct display_timing retroflag_gpi_timing = {
+	.pixelclock = { 6400000 },
+	.hactive = { 240 },
+	.hfront_porch = { 38 },
+	.hback_porch = { 20 },
+	.hsync_len = { 10 },
+	.vactive = { 320 },
+	.vfront_porch = { 20 },
+	.vback_porch = { 4 },
+	.vsync_len = { 4 },
+	.flags = DRM_MODE_FLAG_PVSYNC | DRM_MODE_FLAG_PHSYNC,
+};
+
+static const struct panel_desc retroflag_gpi = {
+	.modes = &retroflag_gpi_mode,
+	.num_modes = 1,
+	.bpc = 8,
+	.size = {
+		.width = 30,
+		.height = 40,
+	},
+	.bus_format = MEDIA_BUS_FMT_RGB666_1X24_CPADHI,
+	.bus_flags = DRM_BUS_FLAG_DE_LOW | DRM_BUS_FLAG_SYNC_DRIVE_NEGEDGE,
+	.connector_type = DRM_MODE_CONNECTOR_DPI,
+};
+
 static const struct drm_display_mode olimex_lcd_olinuxino_43ts_mode = {
 	.clock = 9000,
 	.hdisplay = 480,
@@ -4243,6 +4282,9 @@ static const struct of_device_id platform_of_match[] = {
 	}, {
 		.compatible = "ontat,yx700wv03",
 		.data = &ontat_yx700wv03,
+	}, {
+		.compatible = "retroflag_gpi",
+		.data = &retroflag_gpi,
 	}, {
 		.compatible = "ortustech,com37h3m05dtc",
 		.data = &ortustech_com37h3m,
diff --git a/drivers/gpu/drm/vc4/vc4_dpi.c b/drivers/gpu/drm/vc4/vc4_dpi.c
index dde2c6014748..3aa326f1bce3 100644
--- a/drivers/gpu/drm/vc4/vc4_dpi.c
+++ b/drivers/gpu/drm/vc4/vc4_dpi.c
@@ -192,6 +192,8 @@ static void vc4_dpi_encoder_enable(struct drm_encoder *encoder)
 		dpi_c |= VC4_SET_FIELD(DPI_FORMAT_18BIT_666_RGB_1, DPI_FORMAT);
 	}
 
+	dpi_c |= DPI_OUTPUT_ENABLE_INVERT | DPI_HSYNC_NEGATE | DPI_VSYNC_NEGATE;
+
 	if (mode->flags & DRM_MODE_FLAG_NHSYNC)
 		dpi_c |= DPI_HSYNC_INVERT;
 	else if (!(mode->flags & DRM_MODE_FLAG_PHSYNC))
`

(I was lazy with fixing vc4_dpi)

With this being my overlay

`/*
 * vc4-kms-v3d-overlay.dts
 */

/dts-v1/;
/plugin/;

#include <dt-bindings/pinctrl/bcm2835.h>

/ {
	compatible = "brcm,bcm2835";

  	fragment@0 {
		target-path = "/";
		__overlay__ {
			panel: panel {
				compatible = "retroflag_gpi", "simple-panel";
				orientation = <90>;
				
				port {
					panel_in: endpoint {
						remote-endpoint = <&dpi_out>;
					};
				};
			};
		};
	};

	fragment@2 {
		target = <&dpi>;
		__overlay__  {
			status = "okay";

			port {
				dpi_out: endpoint@0 {
					remote-endpoint = <&panel_in>;
				};
			};
		};
	}; 

	fragment@3 {
		target = <&gpio>;
		__overlay__ {
			dpi24: dpi24 {
				brcm,pins = <0 1 2 3 4 5 6 7 8 9
					         12 13 14 15 16 17 20
					         21 22 23 24 25>;
				brcm,function = <6>;
                brcm,pull = <0>;
			};
		};
	};

	fragment@4 {
		target = <&vc4>;
		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&dpi24>;
		};
	};

	fragment@5 {
		target = <&fb>;
		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&dpi24>;
		};
	};
};`

coreybruce, 2021-11-05:
> The GPICase build unlike the Pi1/2/3 builds does not use vc4-kms or vc4-fkms but the firmware driver. This allows the screen to be more easily configured. Other than the display issues, it's fairly straight forward to just take the Pi3 device as a base as the Pi02 is essentially a Pi3.

Ahh I see

coreybruce, 2021-11-06:
So how's it all going? Been any progress? 😁

Cicirifu, 2021-11-07:
No luck on AArch64, but ARM should be fairly trivial for the people here to add. I made a build for personal use after the AArch64 attempts failed. Dumped all the changes here: https://github.com/Cicirifu/Lakka-LibreELEC/commits/gpi-zero2 if you want to compile it for yourself. Left some AArch64 patches in there but those can be safely ignored.

coreybruce, 2021-11-08:
Oh that's fine, if it's just arm7l than that doesn't matter, as long as it all works than I am happy with it :D

coreybruce, 2021-11-09:
> No luck on AArch64, but ARM should be fairly trivial for the people here to add. (Friendly reminder I'm not part of the Lakka developers)
> 
> I made a build for personal use after the AArch64 attempts failed. Dumped all the changes here: https://github.com/Cicirifu/Lakka-LibreELEC/commits/gpi-zero2 if you want to compile it for yourself. Left some AArch64 patches in there but those can be safely ignored.

Once I get my pi zero W 2 delivered I will test it out but like I said I don't mind if it starts off as a Arm7l build as long as it all works :)

Thanks again for the test build image

Cicirifu, 2021-11-09:
> > No luck on AArch64, but ARM should be fairly trivial for the people here to add. (Friendly reminder I'm not part of the Lakka developers)
> > I made a build for personal use after the AArch64 attempts failed. Dumped all the changes here: https://github.com/Cicirifu/Lakka-LibreELEC/commits/gpi-zero2 if you want to compile it for yourself. Left some AArch64 patches in there but those can be safely ignored.
> 
> Once I get my pi zero W 2 delivered I will test it out but like I said I don't mind if it starts off as a Arm7l build as long as it all works :)
> 
> Where do I download the image?

You build it yourself. I'm not sure if anyone here is okay with me sharing download links, and I don't want to have them floating around the internet anyway.

coreybruce, 2021-11-10:
Alright so do just use make to build it? what the go with the building process? :)

rudism, 2021-11-10:
@coreybruce [the docs](http://www.lakka.tv/doc/Compiling-Lakka/) are a good place to start. Personally I used [docker](https://github.com/libretro/Lakka-LibreELEC/blob/master/Dockerfile) to do it.

coreybruce, 2021-11-11:
> @coreybruce [the docs](http://www.lakka.tv/doc/Compiling-Lakka/) are a good place to start. Personally I used [docker](https://github.com/libretro/Lakka-LibreELEC/blob/master/Dockerfile) to do it.

It says do 

```
DISTRO=Lakka PROJECT=RPi DEVICE=RPi ARCH=arm make image
```
Do i need to change anything to make a pi zero 2 w image instead?

coreybruce, 2021-11-11:
Hmm ran into this error

```
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:656: make] Error 1
make[3]: Leaving directory '/mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-RPi.arm-3.5-devel/make-4.2.1/.x86_64-pc-linux-gnu'
make[2]: *** [Makefile:798: all-recursive] Error 1
make[2]: Leaving directory '/mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-RPi.arm-3.5-devel/make-4.2.1/.x86_64-pc-linux-gnu'
make[1]: *** [Makefile:534: all] Error 2
make[1]: Leaving directory '/mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-RPi.arm-3.5-devel/make-4.2.1/.x86_64-pc-linux-gnu'
FAILURE: scripts/build make:host has failed!
[001/307] [FAIL] build   make:host

The following logs for this failure are available:
  stdout: /mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-RPi.arm-3.5-devel/.threads/logs/1/stdout
  stderr: /mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-RPi.arm-3.5-devel/.threads/logs/1/stderr

parallel: This job failed:
package_worker 1 1 307 'build make:host'
Parallel build failure - see log for details. Time of failure: Thu Nov 11 14:38:04 AEST 2021
make: *** [Makefile:12: image] Error 1
```

duduke, 2021-11-11:
@coreybruce According to @Cicirifu commits in his github, I think you should run: 
`DISTRO=Lakka PROJECT=RPi DEVICE=Pi02GPi ARCH=arm make image`
Just got my 02w, gonna try it later today.

coreybruce, 2021-11-11:
> @coreybruce According to @Cicirifu commits in his github, I think you should run: `DISTRO=Lakka PROJECT=RPi DEVICE=Pi02GPi ARCH=arm make image` Just got my 02w, gonna try it later today.

Ok so I tried that but got this error

```
    /mnt/S/Pr/Lakka-LibreELEC    gpi-zero2  DISTRO=Lakka PROJECT=RPi DEVICE=Pi02GPi ARCH=arm make image    ✔ 
./scripts/image mkimage

 =================================================================================
 Configuration for Lakka (community)
 =================================================================================

 Buildoptions:
 ======================================================
 - CPU (ARCH):				 cortex-a53 (arm)
 - FLOAT:				 hard
 - FPU:					 neon-fp-armv8
 - CPU features:			 32bit neon
 - LTO (Link Time Optimization) support: yes
 - GOLD (Google Linker) Support:	 yes
 - LLVM support:			 no
 - DEBUG:				 no

 Graphic configuration:
 ======================================================
 - OpenGL (GLX) support (provider):	 no (no)
 - OpenGLES support (provider):		 yes (bcm2835-driver)

 Hardware decoder configuration:
 ======================================================
 - Kodi Player driver:			 bcm2835-driver
 - VAAPI Support:			 no
 - VDPAU Support:			 no

 Input device configuration:
 ======================================================
 - Remote support:			 no
 - CEC Adapter support:			 
 - CEC Framework support:		 no
 - Kodi Joystick support:		 yes

 Misc. hardware configuration:
 ======================================================
 - ALSA support:			 yes
 - Pulseaudio support:			 
 - Bluetooth support:			 yes
 - Include driver:			 rpi-cirrus-config
 - Include driver:			 bcm2835-driver
 - Include firmware:			 misc-firmware
 - Include firmware:			 wlan-firmware
 - Include firmware:			 brcmfmac_sdio-firmware-rpi

 Network service configuration:
 ======================================================
 - Avahi (Zeroconf) support:		 yes
 - NFS mounting support:		 yes
 - SAMBA mounting support:		 no
 - SAMBA server support:		 no
 - SFTP server support:			 yes
 - OpenVPN support:			 no
 - WireGuard support:			 yes

 OS configuration:
 ======================================================
 - OEM Support:				 yes
 - Default ROOT Password:		 root
 - Bootloader:				 bcm2835-bootloader
 - UDevil support:			 no
 - Installer support:			 no
 - Include package:			 rsync
[... 249 lines cut ...]
/usr/bin/ld: glob/libglob.a(glob.o): in function `glob_in_dir':
glob.c:(.text+0x78b): undefined reference to `__stat'
/usr/bin/ld: glob/libglob.a(glob.o): in function `glob':
glob.c:(.text+0xee8): undefined reference to `__stat'
/usr/bin/ld: glob.c:(.text+0x1437): undefined reference to `__stat'
/usr/bin/ld: glob.c:(.text+0x176e): undefined reference to `__stat'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:656: make] Error 1
make[3]: Leaving directory '/mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-Pi02GPi.arm-3.5-devel/make-4.2.1/.x86_64-pc-linux-gnu'
make[2]: *** [Makefile:798: all-recursive] Error 1
make[2]: Leaving directory '/mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-Pi02GPi.arm-3.5-devel/make-4.2.1/.x86_64-pc-linux-gnu'
make[1]: *** [Makefile:534: all] Error 2
make[1]: Leaving directory '/mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-Pi02GPi.arm-3.5-devel/make-4.2.1/.x86_64-pc-linux-gnu'
FAILURE: scripts/build make:host has failed!
[001/328] [FAIL] build   make:host

The following logs for this failure are available:
  stdout: /mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-Pi02GPi.arm-3.5-devel/.threads/logs/1/stdout
  stderr: /mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-Pi02GPi.arm-3.5-devel/.threads/logs/1/stderr

parallel: This job failed:
package_worker 1 1 328 'build make:host'
Parallel build failure - see log for details. Time of failure: Thu Nov 11 19:38:47 AEST 2021
make: *** [Makefile:12: image] Error 1
```

Cicirifu, 2021-11-11:
Build through Docker and it'll get rid of all these annoying "it works on my machine" build errors.

`docker build -t lakka_build_env .`
and then my simple build helper in `build.sh`

coreybruce, 2021-11-11:
> docker build -t lakka_build_env .

Ok thanks for the help, I'll try that also I am on Manjaro Gnome

Cicirifu, 2021-11-11:
Depends entirely on your personal setup. In your case, probably a yes. I tend to leave using sudo as an exercise to the user.

coreybruce, 2021-11-11:
Hmm ok it should probs should be standandazed if it does or doesn't, I had to add sudo to the script to make it work but even in docker I ran into a issue I think, it says I don't have zlib so I installed it and it still gives me the error

```
./build.sh                                           ✔  8s  
./scripts/image mkimage

 =================================================================================
 Configuration for Lakka (community)
 =================================================================================

 Buildoptions:
 ======================================================
 - CPU (ARCH):				 cortex-a53 (arm)
 - FLOAT:				 hard
 - FPU:					 neon-fp-armv8
 - CPU features:			 32bit neon
 - LTO (Link Time Optimization) support: yes
 - GOLD (Google Linker) Support:	 yes
 - LLVM support:			 no
 - DEBUG:				 no

 Graphic configuration:
 ======================================================
 - OpenGL (GLX) support (provider):	 no (no)
 - OpenGLES support (provider):		 yes (bcm2835-driver)

 Hardware decoder configuration:
 ======================================================
 - Kodi Player driver:			 bcm2835-driver
 - VAAPI Support:			 no
 - VDPAU Support:			 no

 Input device configuration:
 ======================================================
 - Remote support:			 no
 - CEC Adapter support:			 
 - CEC Framework support:		 no
 - Kodi Joystick support:		 yes

 Misc. hardware configuration:
 ======================================================
 - ALSA support:			 yes
 - Pulseaudio support:			 
 - Bluetooth support:			 yes
 - Include driver:			 rpi-cirrus-config
 - Include driver:			 bcm2835-driver
 - Include firmware:			 misc-firmware
 - Include firmware:			 wlan-firmware
 - Include firmware:			 brcmfmac_sdio-firmware-rpi

 Network service configuration:
 ======================================================
 - Avahi (Zeroconf) support:		 yes
 - NFS mounting support:		 yes
 - SAMBA mounting support:		 no
 - SAMBA server support:		 no
 - SFTP server support:			 yes
 - OpenVPN support:			 no
 - WireGuard support:			 yes

 OS configuration:
 ======================================================
 - OEM Support:				 yes
 - Default ROOT Password:		 root
 - Bootloader:				 bcm2835-bootloader
 - UDevil support:			 no
 - Installer support:			 no
 - Include package:			 rsync
 - Include package:			 xpadneo
 - Include package:			 python-raspberry-gpio
[... 91 lines cut ...]
  The CMAKE_C_COMPILER:

    /mnt/Storage/Project/Lakka-LibreELEC/build.Lakka-Pi02GPi.arm-3.5-devel/toolchain/bin/armv8a-libreelec-linux-gnueabi-gcc

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/build.Lakka-Pi02GPi.arm-3.5-devel/zlib-1.2.11/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/build.Lakka-Pi02GPi.arm-3.5-devel/zlib-1.2.11/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeError.log".
FAILURE: scripts/install zlib has failed!
[058/328] [FAIL] install zlib

The following logs for this failure are available:
  stdout: /home/ubuntu/build.Lakka-Pi02GPi.arm-3.5-devel/.threads/logs/62/stdout
  stderr: /home/ubuntu/build.Lakka-Pi02GPi.arm-3.5-devel/.threads/logs/62/stderr

parallel: This job failed:
package_worker 3 62 328 'install zlib'
Parallel build failure - see log for details. Time of failure: Thu Nov 11 12:05:53 UTC 2021
make: *** [Makefile:12: image] Error 1
```

coreybruce, 2021-11-11:
> If this were run in Docker then "/mnt/Storage/Project/..." should not be a thing that exists, so I'm not sure where that is coming from. Something to debug on your end.
> 
> And again for the sudo thing, it varies per user. Not everyone needs sudo. Better to understand when you need sudo and when you don't than blindly telling people to slap sudo in front of it.

It's inside ```/mnt/Storage/Project/Lakka-LibreELEC``` which is where build.sh is, I do see the projects folder there. I don't have any experience with docker so I don't know hoe to debug this and will need help if I am doing something wrong.

Cicirifu, 2021-11-11:
You have it stored there, but your build inside Docker operates in a completely different filesystem. A virtual machine of sorts. Nothing inside of it should have or be given knowledge of folders outside of it, because it will never be able to find it. 

But this conversation is getting tiresome so I advise you to google for "Retroflag GPi Discord" and you'll likely stumble upon one that may or may not have a link to my build that is being tested by people I know.

coreybruce, 2021-11-11:
> You have it stored there, but your build inside Docker operates in a completely different filesystem. A virtual machine of sorts. Nothing inside of it should have or be given knowledge of folders [...]

Look you can't recommended a way to do something and than abandon a user when it's "tiresome" that's the point of support, it's a two way street of me trying to help you so you can help me and I really want to get this going properly and build the image but I need you guys help for this to happen. You didn't also tell me I needed to log out to properly take affect and work properly which is why it was asking for sudo and now it isn't.

Cicirifu, 2021-11-11:
I tried being helpful by falling back to a more pragmatic, faster, solution of hinting you towards the image, which is what you initially asked for, while still paying mind to my own concern of sharing random downloads. This because it was obvious the manual build route was going nowhere and would require more support than a Github Issue should be used for. (This is not a chatroom nor a helpdesk)

Furthermore, I don't owe you support in any way, shape, or form. I spent a week of evenings on this and I was kind enough to share it. I am not a Lakka developer. I am not a RetroArch developer. I am a GPi user and a software engineer with spare time. Build instructions in general aren't written with non-developers in mind either.

> You didn't also tell me I needed to log out to properly take affect and work properly which is why it was asking for sudo and now it isn't.

This would be because I have no idea what you're on about and you're acting like this is something obvious.

This is where I'm gonna end my participation in this discussion.

coreybruce, 2021-11-11:
Oh I'm very sorry, I thought thought you were a dev but didn't check, I didn't realize you weren't a developer, I still wouldn't mind downloading a image from you tho.

"This would be because I have no idea what you're on about and you're acting like this is something obvious, when this is something entirely specific to your situation."

I was talking about what was needed to make docker properly work, it's okay hopefully I will find some help to get this going, thanks very much for the help tho. :)

coreybruce, 2021-11-11:
> No luck on AArch64, but ARM should be fairly trivial for the people here to add. (Friendly reminder I'm not part of the Lakka developers)
>
> I made a build for personal use after the AArch64 attempts [...]

One more thing tho, from my investigation I think there is a issue with either the docker container or the script it uses in the docker container because I figured out how to go into the container and it has the package it claims is missing which is odd..

coreybruce, 2021-11-11:
@dmrlawson @SethBarberee @rudism any of you guys have a idea on whats happening? you might know more than I do and I'm willing to learn and help! :D

coreybruce, 2021-11-11:
Also want to add that this is the script for the gpi case 

https://github.com/RetroFlag/retroflag-picase

gouchi, 2021-11-11:
You can try this [testing image](https://nightly.builds.lakka.tv/members/gouchi/1544/Lakka-Pi02GPi.arm-3.5-devel-20211111161748-ec3765b.img.gz).

Thank you.

gouchi, 2021-11-11:
No as I have just compiled [Cicirifu tree](https://github.com/Cicirifu/Lakka-LibreELEC/commits/gpi-zero2).

Cicirifu, 2021-11-11:
It comes with the same shutdown script as the regular GPICase / Pi0 build, and there's no reason for it not to work. Works fine on my end and the people I shared this with.

coreybruce, 2021-11-11:
Awesome can't wait to try, again thank you very much and I will do what ever I can to help with any issues on my end.

duduke, 2021-11-11:
I've tried the image @gouchi shared. Most of the things seem to be working.
The emulation speed is much much worse compared to the RecalBox beta.
I've compared aburner2 on FBNeo on both RecalBox beta and Lakka, and Lakka is much slower.
I've verified that i'm not throttling: 
```
vcgencmd get_throttled
throttled=0x0
```
The CPU seems to stay at 600Mhz even when the FBNeo is running:
```
cgencmd measure_clock arm
frequency(48)=600000000

```

Any ideas?

Cicirifu, 2021-11-11:
`Lakka:~ # vcgencmd measure_clock arm
frequency(48)=1000000000`

Running at 1GHz in menu and benchmarking Golden Sun in mGBA.

coreybruce, 2021-11-12:
ok so from my testing it works well, I tested GB, GBC and GBA as well as Mega drive, NES and SNES. there was a issue with aspect ration being thin and squashed but I fixed it by setting the aspect ration to 4:3 in the video settings. everything seems to all be working all at full speed but I did notice a little audio pops on SNES when testing Super Mario world.

The default rgui is ok and works fine especially after some customization but it would be nice to have a UI options similar to emulation stations UI would work well for this small resolution as this UI is the only one that work correctly and xmb looks to small on the screen.

Over all this is almost ready for people to use. :D

Cicirifu, 2021-11-12:
I have gotten one additional report on my end of someone stuck on 600MHz. While I don't have the same issue, my current image is slightly different. However, before sharing links I did benchmarking as well and none of this showed, running mGBA and PCSX ReARMed at full speed in known demanding games, matching performance I expected. (I have a GPi running a Pi3A with Lakka 2.3.2)

So, anyhow, might be worth looking into. Im expecting it might be distroconfig.txt related or perhaps firmware version. I do not have the ability to investigate at this time.

-------- Original Message --------

duduke, 2021-11-12:
@Cicirifu can you post your config.txt and distroconfig.txt please?
Thanks.

coreybruce, 2021-11-12:
Yeah with the default image I didn't run into game speed issues

Cicirifu, 2021-11-12:
I suspect the 600MHz thing is a red herring. Possible just the CPU scaling. Ive gotten two more confirmations of the 600MHz thing now but no performance complaints. Golden Sun on mGBA or FF7/8/9 on PCSX ReARMed working essentially flawlessly, or certainly not speeds Id expect from a single-core 600MHz. It's possibly just the CPU governor indicating it doesn't need to run all cores at full speed? I've managed to trigger this behaviour on my end as well.

duduke, 2021-11-12:
I'm still getting a slowdown with LAKKA.
Specifically AfterBurner 2 on FBNeo - Slow on LAKKA, Fast on the RecalBox beta.
This is the clock speed on RecalBox:
```
# vcgencmd measure_clock arm
frequency(48)=1000104000
```

Cicirifu, 2021-11-12:
Is it possible FBNeo is compiled incorrectly / not optimally?

duduke, 2021-11-12:
Could be, but i've also tested with MAME which is also slower than RecalBox.
Maybe the cores are compiled with the wrong CFLAGS?

Cicirifu, 2021-11-12:
Worth looking into. It should be compiling with 32-bit cortex-a53, neon-fp-armv8 and FBNeo specially with profile=performance and HAVE_NEON=1.

Ill do a CFLAGS comparison with Recalbox tomorrow

coreybruce, 2021-11-13:
Hmm interesting, if that is the case and it does need to be recompiled will we get a new test image?

Cicirifu, 2021-11-13:
Did find some issues on the CPU speed front. Turns out cpufreq wasnt present, meaning all scaling was effectively borked. The included Linux kernel config is based on RPI2's, so this was probably a mistake. Enabling `CONFIG_ARM_RASPBERRYPI_CPUFREQ` didnt end up enabling cpufreq for me and since I am not knowledgeable when it comes to Linux kernel configs, I just took Recalbox's config and presto. Cpufreq was available and aburner2 bumped up from a nice 26 to 42 FPS. Unfortunately I didn't check what FPS Recalbox manages. :')

Cicirifu, 2021-11-13:
Yup. Getting same FPS on Recalbox. So if someone could look into the Kernel config, that'd be sweet. I committed the Recalbox one to my fork for now but that shouldn't be blindly copied over to Lakka's branch.

coreybruce, 2021-11-15:
Should we get @gouchi to make a new test image with the changed or not yet?

Cicirifu, 2021-11-15:
> Also to report, I think the samba service isn't working, how do I check on lakka?
>
> it uses systemctl doesn't it?

Oh. Right. My test image doesnt have Samba in it because everyone on my end keeps asking to remove it as Samba is an annoying mess, where SSH and SFTP(SCP) are much less frustrating to transfer files with.

-------- Original Message --------

ToKe79, 2021-11-15:
I built the image, you can get it here: https://nightly.builds.lakka.tv/members/vudiq/rpi02gpi/
I made minor change to the kernel config file:
```diff
diff --git a/projects/RPi/devices/Pi02GPi/linux/linux.arm.conf b/projects/RPi/devices/Pi02GPi/linux/linux.arm.conf
index 2adc2f38c1..a0b0a8199e 100644
--- a/projects/RPi/devices/Pi02GPi/linux/linux.arm.conf
+++ b/projects/RPi/devices/Pi02GPi/linux/linux.arm.conf
@@ -34,7 +34,7 @@ CONFIG_KERNEL_GZIP=y
 # CONFIG_KERNEL_LZO is not set
 # CONFIG_KERNEL_LZ4 is not set
 CONFIG_DEFAULT_INIT=""
-CONFIG_DEFAULT_HOSTNAME="(none)"
+CONFIG_DEFAULT_HOSTNAME="@DISTRONAME@"
 CONFIG_SWAP=y
 CONFIG_SYSVIPC=y
 CONFIG_SYSVIPC_SYSCTL=y
@@ -167,7 +167,7 @@ CONFIG_SCHED_AUTOGROUP=y
 # CONFIG_SYSFS_DEPRECATED is not set
 CONFIG_RELAY=y
 CONFIG_BLK_DEV_INITRD=y
-CONFIG_INITRAMFS_SOURCE=""
+CONFIG_INITRAMFS_SOURCE="@INITRAMFS_SOURCE@"
 CONFIG_RD_GZIP=y
 CONFIG_RD_BZIP2=y
 CONFIG_RD_LZMA=y
```

coreybruce, 2021-11-15:
> Also to report, I think the samba service isn't working, how do I check on lakka?
> 
> it uses systemctl doesn't it?

Hmm I see, I've never had issues with it, in fact I found it a convenient options to copy stuff over and always used it with Retropie and Lakka

ToKe79, 2021-11-15:
Samba does not always work as expected due to security measures implemented after wannacry outbreak, iirc. Using SCP/SFTP client (like the opensource WinSCP) can be really more convenient to copy files over network. You use the same credentials (default root:root) as for SSH.

coreybruce, 2021-11-15:
> Samba does not always work as expected due to security measures implemented after wannacry outbreak, iirc. Using SCP/SFTP client (like the opensource WinSCP) can be really more convenient to copy files over network. You use the same credentials (default root:root) as for SSH.

Samba should only has access to the home folder which only needs normal user permission.

coreybruce, 2021-11-15:
Is anyone else still getting audio popping and crackling on some games?

Cicirifu, 2021-11-15:
Set alsathread for audio and try a 48kHz sample rate. Also note the GPi in general has rather poor audio noise on top of that.

-------- Original Message --------

coreybruce, 2021-11-15:
> Set alsathread for audio and try a 48kHz sample rate. Also note the GPi in general has rather poor audio noise on top of that.
> […](#)
> -------- Original Message --------

Alsathread as on by default but hmm it seemed to add more popping when I did that but I changed it from 43Hz to 40Hz made it do it far less

Brunnis, 2021-11-16:
> Set alsathread for audio and try a 48kHz sample rate. Also note the GPi in general has rather poor audio noise on top of that.

I just updated my GPi Case to Zero 2 and have been testing Recalbox 7.3 beta 28 on it. Audio works quite terribly out of the box due to three things:

1. Rewind feature turned on by default. Causes performance degradation which results in audio popping.
2. RetroArch is set to use its "pulse" audio driver instead of "alsathread". For some reason, synchronization/buffering seems to be off with "pulse" causing a very audible audio dropout every 26th second. alsathread fixes this.
3. The audio gain is too high resulting in some games getting very distracting clipping noise. It's extremely jarring in Star Fox. Setting RetroArch's audio gain to -10 dB fixes it.

I don't know if any of these things plague Lakka at this point. However, I'd like to point out that the audio quality between the old Zero and the Zero 2 is night and day. The audio with the Zero 2 is much cleaner, whereas on the Zero 1 there was this constant hissing and sort of varying noise that was quite audible.

coreybruce, 2021-11-16:
> I just updated my GPi Case to Zero 2 and have been testing Recalbox 7.3 beta 28 on it. Audio works quite terribly out of the box due to three things:
>
>     1. Rewind feature turned on by default. [...]

Ahh interesting, also didn't know that was a issue with the original pi zero also on this image it has the audio set to alsathreaded by default :)

[74 more replies]

closed 2022-02-28 by ToKe79