libretro/Lakka-LibreELEC #1545: incorrect regfresh for RG351MP?

auxfam, 2021-11-04:
hardware:

RG351MP  (odroidgoadvamce builds)

Lakka version:
nightiles most notably: https://nightly.builds.lakka.tv/2021-10-31/OdroidGoAdvance.arm/Lakka-OdroidGoAdvance.arm-nightly-20211031-029641c-rg351mp.img.gz

what I expect:   

`Video -> Output -> Set display-Reported refresh Rate == 60.00Hz`

What I get instead:

`Video -> Output -> Set display-Reported refresh Rate == 149.129Hz`

summary:

the present builds for RG351MP has set the refresh rate absurdly high (`49/129 Hz as above)  this seems to cause the reported frame rates when you enable ramerate notifications on to fluxtuate during menus and cores

notes:

doesn't seem to be the case with the similarly RG351P builds and hardware.

zongyuanliu-bio, 2021-11-07:
Hi, I install Lakka on RG351MP too, why I can not even boot? Have you ever had this issue before?

auxfam, 2021-11-08:
> 
> 
> Hi, I install Lakka on RG351MP too, why I can not even boot? Have you ever had this issue before?

boots just fine.  I use a 16GB sd card placed into the `TF/Int` slot .  fresh install (not upgrade).   I flash onto said card using dd (on linux) after unzipping the image.  as mentioned above this is the image used: https://nightly.builds.lakka.tv/2021-10-31/OdroidGoAdvance.arm/Lakka-OdroidGoAdvance.arm-nightly-20211031-029641c-rg351mp.img.gz

zongyuanliu-bio, 2021-11-08:
> boots just fine. I use a 16GB sd card placed into the `TF/Int` slot . fresh install (not upgrade). I flash onto said card using dd (on linux) after unzipping the image. as mentioned above this is [...]

I tried, it failed. When I first got the machine, Lakka works, then second day when I boot it, it just faild, other firmwares (ArkOS or 351ELEC) are fine, let me try again.

blindm0use, 2022-02-22:
After flashing a new SD card of a capacity not less than 2gb with Lakka (I used latest development image and a 4gb SD card) remount it on your computer and do the following:
1. Download latest stable release of 351ELEC
2. Navigate to the downloaded .gz archive and open it (can be done with 7-Zip if on Windows).
3. Open .img within the archive
4. Navigate to "0.fat"
5. Extract "rk3326-rg351mp-linux.dtb"
6. Navigate to the root of your Lakka SD card and delete every .dtb file
7. Place .dtb file from the **Step 5** into the root of Lakka SD card
8. Navigate to "extlinux" folder of Lakka SD card
9. Leave only "extlinux.conf" & "rk3326-rg351mp-linux.dtb.conf" files in the folder
10. Edit each of these two files with text editor (Notepad) and replace the 4th line starting with "APPEND  " with the following:
_**APPEND boot=UUID=2102-1725 disk=UUID=677fc905-b613-44b1-ba7a-55441415562e quiet console=ttyFIQ0 console=tty0 net.iframes=0 fbcon=rotate:0 ssh consoleblank=0 systemd.show_status=0 loglevel=0 panic=20**_
11. Safely unmount the modified Lakka SD card and insert it into your RG351MP;

**After following the steps above, you will have a fully bootable device into Lakka with correct battery management, working wifi dongle, and great RetroArch frontend. _The only problem is volume buttons not working and low volume, but these can be assigned to hotkey and a key combination as an alternative until the fix arrives._**

zyliu-OU, 2022-02-23:
I will give it a try, thanks! 👏😆

vanfanel, 2022-05-02:
@auxfam Do you know if the original issue is supposed to be fixed by now?
I am trying latest stable Lakka and refresh rate is still wrong.... Which renders it useles on the machine.

blindm0use, 2022-05-02:
We have determined with @armadillu that there might be different hardware revisions of RG351MP, which makes Lakka not bootable for some units.

But by simply swapping original dtb with [this one](https://github.com/christianhaitian/arkos/files/8238619/rk3326-rg351mp-linux.zip) system starts without a hiccup. Only the volume is not balanced in relevance to the the gain. At max vol the sound starts cracking

closed 2022-06-10 by Ntemis