libretro/Lakka-LibreELEC #1546: No audio through Composite output

paqman3d, 2021-11-12:
- What system hardware are you using?

Pi3 with 64bit Lakka 3.5.2 

- What did you do?

Hooked up system via composite out.

- What did you expect to happen?

Have sound

- What happened instead?

No sound.

Basically was testing the CRT fix and did a clean install. However, after loading any game I have no sound output at all. Is there something that changed in the config.txt on the lakka partition of the sd card perhaps? Is anything different for outputting sound through composite after the crt fix?

ToKe79, 2021-11-12:
I think you have to change the audio device to something appropriate for the analog output. I think somebody in discord mentioned setting it to `default:CARD=Headphones` worked for them.

paqman3d, 2021-11-13:
Yep, that did the trick. Happy to have CRT output working again!

closed 2021-11-13 by paqman3d