libretro/Lakka-LibreELEC #1550: f2,f3,f6,f7 and f8 are all not working.

beatle-fan, 2021-11-20:
- Which version of Lakka are you using?  3.9
- What system hardware are you using? pc (HP)
- What did you do? I went into hotkeys to map some keys
- What did you expect to happen? To map the keys
- What happened instead? They didn't allow me to map them

ToKe79, 2021-11-24:
what actions are you trying to bind/map? because I was able to map all of the above function keys.

beatle-fan, 2021-11-24:
I was trying to map volume up, down, mute and brightness up and down.

ToKe79, 2022-03-20:
@beatle-fan can you test with latest version?
anyway - after you assign a button for action, I think you have to hold the key assigned to `Hotkey Enable` - e.g. on gamepad `Hotkey Enable` assigned to `SELECT` and `Volume Up` assigned to `Right thumb stick up` requires pressing `SELECT`+`Right thumb stick up` in order to trigger volume up. or the mapping failed when you tried to assign the button - it did not register the button?

Ntemis, 2022-06-10:
Solution is already given ^^

closed 2022-06-10 by Ntemis