libretro/Lakka-LibreELEC #1551: RG351MP console can not boot Lakka 3.5.2

zongyuanliu-bio, 2021-11-20:
Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.
Tried all versions of lakka for RG351MP, even the lastest 3.6 and nightly 11192021.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 3.5.2
- What system hardware are you using? RG351MP
- What did you do? Reboot the system.
- What did you expect to happen? I want to boot with lakka logo and successfully go to main menu
- What happened instead? Boot with lakka logo and crash with many command lines.

Below are the log for system, in 01_RetroArch.log, it showed WIFI error.
[log-2021-11-20-19.14.12.zip](https://github.com/libretro/Lakka-LibreELEC/files/7575096/log-2021-11-20-19.14.12.zip)

[log-2021-11-20-19.10.50.zip](https://github.com/libretro/Lakka-LibreELEC/files/7575097/log-2021-11-20-19.10.50.zip)

zongyuanliu-bio, 2021-11-20:
RG351MP doesn't have built-in Bluetooth and WIFI modules. Maybe it's the problem reason?

Graham15, 2021-12-06:
Hi zongyuanliu-bio. I'm running Lakka 3.6 on RG351MP without any issues with or without the wireless dongle attached. Have you tried different micro sd cards?

zongyuanliu-bio, 2021-12-06:
> Hi zongyuanliu-bio. I'm running Lakka 3.6 on RG351MP without any issues with or without the wireless dongle attached. Have you tried different micro sd cards? 

Yes, I have sandisk and samsung, also with 16GB 32GB or 64GB, all failed.

Graham15, 2021-12-06:
That's really odd, especially as you can write to these cards with other firmware, so it's not your card reader.

zongyuanliu-bio, 2021-12-06:
> That's really odd, especially as you can write to these cards with other firmware, so it's not your card reader.

For the card reader, I have a USB 3.0 version, I have SD adapter version, I tried all, very strange, you all good but mine is not, the machine also don't have issue since I can install ArkOS and run smoothly. The Retroarch in ArkOS run very well, so I really don't know what happened.

zongyuanliu-bio, 2021-12-07:
> That's really odd, especially as you can write to these cards with other firmware, so it's not your card reader.
> 
> 
> 
> 
After inserted fresh flashed card, it will boot with a lakka logo, then it crash to this.
[image]

zongyuanliu-bio, 2021-12-07:
> > That's really odd, especially as you can write to these cards with other firmware, so it's not your card reader.
> > 
> > 
> > 
> > 
> After inserted fresh flashed card, it will boot with a lakka logo, then it crash to this.
> [image]

If I click reset button, sometime it will give other results.[image]

Graham15, 2021-12-07:
You have probably already tried this. Try 3.6 or a nightly build. Download the firmware again in case downloaded version has become corrupt. If you have an sd card in slot 2, remove and boot only off slot 1. Not sure what else you can try apart from googling this error and waiting for a dev to respond. Good luck.

:)

starquake, 2022-01-20:
I have the same  issue. RG351MP. Will not boot 3.5.2, 3.6 or nightly-20220118. Tested multiple SD cards. With and without wireless.
If I reboot a bunch of times it will start resizing the filesystem. But after a few seconds it crash.

Any ideas on what to try next?

zongyuanliu-bio, 2022-02-02:
> I have the same issue. RG351MP. Will not boot 3.5.2, 3.6 or nightly-20220118. Tested multiple SD cards. With and without wireless. If I reboot a bunch of times it will start resizing the filesystem. But after a few seconds it crashes.
> 
> Any ideas on what to try next? EDIT: I see some people specifically mentioning using Samsung or SanDisk. I have tried a bunch of brands but not those. Will try those. EDIT2: Nope Samsung does not fix it

Yes, same as mine, so I choose to use ArkOS, waiting for future versions of Lakka.

blindm0use, 2022-02-22:
After flashing a new SD card of a capacity not less than 2gb with Lakka (I used latest development image and a 4gb SD card) remount it on your computer and do the following:
1. Download latest stable release of 351ELEC
2. Navigate to the downloaded .gz archive and open it (can be done with 7-Zip if on Windows).
3. Open .img within the archive
4. Navigate to "0.fat"
5. Extract "rk3326-rg351mp-linux.dtb"
6. Navigate to the root of your Lakka SD card and delete every .dtb file
7. Place .dtb file from the **Step 5** into the root of Lakka SD card
8. Navigate to "extlinux" folder of Lakka SD card
9. Leave only "extlinux.conf" & "rk3326-rg351mp-linux.dtb.conf" files in the folder
10. Edit each of these two files with text editor (Notepad) and replace the 4th line starting with "APPEND  " with the following:
_**APPEND boot=UUID=2102-1725 disk=UUID=677fc905-b613-44b1-ba7a-55441415562e quiet console=ttyFIQ0 console=tty0 net.iframes=0 fbcon=rotate:0 ssh consoleblank=0 systemd.show_status=0 loglevel=0 panic=20**_
11. Safely unmount the modified Lakka SD card and insert it into your RG351MP;

**After following the steps above, you will have a fully bootable device into Lakka with correct battery management, working wifi dongle, and great RetroArch frontend. _The only problem is volume buttons not working and low volume, but these can be assigned to hotkey and a key combination as an alternative until the fix arrives._**

closed 2022-05-13 by zyliu-OU