libretro/Lakka-LibreELEC #1552: Can not pair any xbox controller to any build

smf-gitwit, 2021-11-22:
- Which version of Lakka are you using? Lakka-RPi3.aarch64-nightly-20211121-465562e,Lakka-RPi3.aarch64-3.6, or Lakka-RPi2.arm-3.6 
- What system hardware are you using? Pi 3 B v1.2
- What did you do? Access Lakka file system via SAMBA and add file named xpadneo.conf with following contents to the share Configfiles → subfolder modprobe.d:  "options bluetooth disable_ertm=1" > Restart the box.
- What did you expect to happen? The controller would pair 
- What happened instead? "Connected" was displayed but the controller still flashed and did not pair
- ALL the hardware works just fine with Reacalbox. The hardware is fine it all paired.
- 
It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

[log-2021-11-22-04.42.09.zip](https://github.com/libretro/Lakka-LibreELEC/files/7578266/log-2021-11-22-04.42.09.zip)

arunningpir8, 2021-11-24:
I also have this issue.  It appears to be a problem with series x/s controllers, as the original controllers for Xbox one work fine.  To fix it the image needs to be recompiled with "Privacy = device" in bluetooth/main.conf.

Seeing as how the filesystem is read only, a new image needs to be built.  I am trying to get my dev machine up and running to test this but reports online show that is the fix.

ToKe79, 2021-11-24:
should be fixed now with 9e4af94432e9a176f2b42dcc7785f961306df11a.
you can retest with next nightly build.
you might need to clean the bluetooth cache (iirc `rm -rf /storage/.cache/bluetoothctl`).
also test if `disable_ertm` is needed, but I think it is (due to older kernel).

smf-gitwit, 2021-11-26:
Sadly not. Installed a fresh install of 9e4af94 and paired a xbox one s controller with the current firmware the first time.Installed a second copy of 9e4af94 on a 2nd micro sd with the disable_ertm=1 in xpadneo.conf and tested. The controller would not pair.Went back to the fist install and the controller would no loner pair.Cleared the cache on first install rm -rf /storage/.cache/bluetoothctl and could not pair.Paired the controller back to the xboxI installed another fresh install of 9e4af94 and tried to pair the controller and could not replicate the first pair From: Tomáš KelemenSent: Wednesday, November 24, 2021 2:45 PMTo: libretro/Lakka-LibreELECCc: smf-gitwit; AuthorSubject: Re: [libretro/Lakka-LibreELEC] Can not pair any xbox controller to any build (Issue #1552) should be fixed now with 9e4af94.you can retest with next nightly build.you might need to clean the bluetooth cache (iirc rm -rf /storage/.cache/bluetoothctl).also test if disable_ertm is needed, but I think it is (due to older kernel).

gouchi, 2022-03-02:
@smf-gitwit Is still an issue with [nightly builds](https://nightly.builds.lakka.tv/) ?

Thank you.

beyondmeat, 2022-03-05:
Yes this still the issue with v.3.7.

I have a white xbox one x BT controller with the latest firmware. I also have a xbox one Core (series) controller with latest version.

gouchi, 2022-03-05:
Thank you for the feedback.

Not sure if it will help but you may try [those steps](https://github.com/atar-axis/xpadneo/issues/287#issuecomment-846156904).

Also if you have the opportunity to make a test with another [BT](https://atar-axis.github.io/xpadneo/#bt-dongles) [dongle](https://github.com/atar-axis/xpadneo/issues/93) ?

beyondmeat, 2022-03-05:
@gouchi i have  pi 400 (kb)

ToKe79, 2022-04-07:
@beyondmeat pairing should be fixed in latest [Lakka 4.1-rc](https://lakka.tv/articles/2022/04/03/lakka-4.1-rc/#bluetooth-pairing). Can you please confirm it is working also for you?

beyondmeat, 2022-04-08:
@ToKe79 unfortunately it's still an issue. I updated to 1Lakka (official): devel-20220407004438-f6d6477 (RPi4.aarch64)`
I used the spacebar to clear all the pairings, I rebooted via the menu. I tried to pair 3 different versions of the xbox one controller. The first bluetooth one controller (microusb port), a white usb-c BT one controller, and a new 2022 Series X black controller (share button). All 3 controllers have had latest firmware installed (updated via xbox accessories on pc or series x). Back when I first commented, I had the latest firmware for them at the time. I did a firmware check today and only 1 variant had an update, didn't make a difference.

All 3 controller variant will keep flashing the xbox button like it's in sync mode. Retroarch will freeze for about 2-3 minutes whole the UI says connected. The controller sync eventually times out, controller goes back to last paired device, like the pairing with lakka never took or happened.

I tried disabling all bluetooth PCs (because BT smart connect was popping up on 2 of the controllers for Windows 11), thinking it was somewhere intercepting it and causing the issue, but that's not the case.

I took a brand new Nintendo Switch Online SNES controller and genesis controller, both unpaired to anything and tried to pair with lakka, didn't work either. Same issue Lakka says connected while controller still in sync mode.

I believe today's nightly should have everything in 4.1rc just  a few commits newer.

I can do a fresh clean install with the latest nightly but not sure if that will fix it.

I can provide logs, have ssh access. Let me know what you need.

I was able to pair all my controllers to my PC, even the switch controllers. Seems to be a lakka issue with the pi 400?

ToKe79, 2022-04-08:
Hi @beyondmeat,

sorry to hear that, maybe the driver we include is not working or is conflicting with something.

To troubleshoot a bit deeper, please follow these steps:

1. Boot Lakka, make sure you are connected to a network (Wi-Fi / Ethernet) - check device IP via `Main Menu` &rarr; `Information` &rarr; `Network Information`
2. Enable `SSH` via `Settings` &rarr; `Services`
3. Connect via `ssh` to your device with user `root` and password `root`, e.g. `ssh root@<device_ip>`
4. Re-mount FAT32 `LAKKA` partition in read-write mode, so we can store log and configuration files there: `mount -o remount,rw /flash`
5. Store the current system log: `dmesg -c > /flash/dmesg-1-before_pair.txt`
6. Stop the Bluetooth service: `systemctl stop bluetooth`
7. Wipe all Bluetooth settings: `rm -rf /storage/.cache/bluetooth`
8. Start the Bluetooth service: `systemctl start bluetooth`
9. Start Bluetooth command-line utility: `bluetoothctl`
10. Put your gamepad into pairing mode
11. Scan for devices: `scan on` and wait until your device shows up, take a note of the gamepad MAC address
12. Set the device into pairing mode: `pairable on`
13. Start pairing with your gamepad: `pair <MAC_address>`
14. Set the trust to gamepad: `trust <MAC_address>`
15. Connect the gamepad: `connect <MAC_address>`

At this point your gamepad should be paired. If not, get the system log again: `dmesg -c > /flash/dmesg-2_after_pair.txt`. Also copy the Bluetooth config files to the FAT32 partition:

```
mkdir /flash/bt-files
for f in $(find /storage/.cache/bluetooth -type f) ; do target_file="${file//:/_}" && target_file="${target_file//\//-}" && cp "${file}" "/flash/bt-files/${target_file}" ; done
```

Then get the created logs and Bluetooth configuration from the `LAKKA` partition, check them for any sensitive information, redact if necessary, zip them and upload them.

Thank you for your time.

beyondmeat, 2022-04-09:
@ToKe79 Thank you will follow those steps this weekend. Take a look at [this posting](https://www.reddit.com/r/linux_gaming/comments/lenhbc/comment/i293rwt) about this issue.

Ntemis, 2022-06-10:
This is stalled issue so closing down

closed 2022-06-10 by Ntemis