libretro/Lakka-LibreELEC #1553: config.txt settings do not prevent 4K resolution on Pi 4

Brunnis, 2021-11-22:
Since a few releases back, the config.txt file on the Raspberry Pi 4 builds contain the following two lines that are supposed to prevent the Pi 4 from running at 4K 30Hz when connected to a 4K display:
```
hdmi_max_pixel_freq:0=200000000
hdmi_max_pixel_freq:1=200000000
```
However, these lines have no effect when using the full KMS driver, so the Pi still happily starts in 4K 30 Hz. As far as I know, there's no native Linux way of putting an upper limit on the resolution. For my own system, I ended up writing an autostart script that handles this and configures RetroArch's resolution before launch. The upside of that is that it works for all hardware. Maybe something to consider? 

The autostart.sh script I'm using can be seen below:

```shell
# NOTE: Put this autostart.sh file in the /storage/.config directory.

# Use the variables below to set the desired maximum resolution of Lakka.
maxWidth=1920
maxHeight=1080

# Get the current console framebuffer resolution.
framebufferWidth=$(fbset | awk 'NR==2{sub(/.*mode "/,"");sub(/x.*/,"");print;}')
framebufferHeight=$(fbset | awk 'NR==2{sub(/.*x/,"");sub(/-.*/,"");print;}')

# Get the current configured Lakka resolution from retroarch.cfg.
currentLakkaWidth=$(grep video_fullscreen_x /storage/.config/retroarch/retroarch.cfg | cut -d'"' -f2)
currentLakkaHeight=$(grep video_fullscreen_y /storage/.config/retroarch/retroarch.cfg | cut -d'"' -f2)

if [ $((framebufferWidth * framebufferHeight)) -gt $((maxWidth * maxHeight)) ]
then
    # Console framebuffer resolution exceeds the desired maximum resolution. Write the desired 
    # maximum resolution to retroarch.cfg, but don't do this if retroarch.cfg is already set to the 
    # correct resolution (prevents unnecessary writes).
    if ! [ $((currentLakkaWidth * currentLakkaHeight)) -eq $((maxWidth * maxHeight)) ]
    then
        sed -i "s/video_fullscreen_x = \"[0-9]\+\"/video_fullscreen_x = \"${maxWidth}\"/g" /storage/.config/retroarch/retroarch.cfg
        sed -i "s/video_fullscreen_y = \"[0-9]\+\"/video_fullscreen_y = \"${maxHeight}\"/g" /storage/.config/retroarch/retroarch.cfg
    fi
else
    # Console framebuffer resolution is less than or equal to the desired maximum resolution. Write 
    # the console framebuffer resolution to retroarch.cfg, but don't do this if retroarch.cfg is 
    # already set to the correct resolution (prevents unnecessary writes).
    if ! [ $((currentLakkaWidth * currentLakkaHeight)) -eq $((framebufferWidth * framebufferHeight)) ]
    then
        sed -i "s/video_fullscreen_x = \"[0-9]\+\"/video_fullscreen_x = \"${framebufferWidth}\"/g" /storage/.config/retroarch/retroarch.cfg
        sed -i "s/video_fullscreen_y = \"[0-9]\+\"/video_fullscreen_y = \"${framebufferHeight}\"/g" /storage/.config/retroarch/retroarch.cfg
    fi
fi
```

RobLoach, 2021-12-09:
I've heard there's been a bunch of changes in LibreELEC 10 that help with 4K on Pi 4. It may be worth waiting for an update to LibreELEC 10 to look for 4K support on the Pi4 in Lakka:
https://libreelec.tv/2021/11/03/libreelec-matrix-10-0-1/

jcbente, 2021-12-13:
With my rpi4, the only config that worked to ignore 4K30Hz was setting `hdmi_ignore_edid=0xa5000080` in config.txt.

gouchi, 2022-03-02:
@Brunnis is this still an issue ?

Brunnis, 2022-03-02:
> @Brunnis is this still an issue ?

Should be. I mean, since these config.txt commands are not supported with the full KMS video driver, it’s not going to start working. These commands worked as long as the firmware was responsible for the display. With full KMS that has changed and we need to solve it in some other way. I personally did it with the script in my first post above, but I’m not sure what would be an acceptable solution to actually integrate into Lakka.

kekopop, 2022-04-12:
Had the same problem with the 4.1 release and found this thread. My cheap HDMI switch wouldn't handle 4k optimally though, so wanted lower res all the way through the boot process. Did some further googling. Turns out HDMI settings are done with kernel parameters now:

https://www.kernel.org/doc/Documentation/fb/modedb.txt

To force 720p from boot, I appended:

`video=HDMI-A-1:1280x720M@60`

...to the existing kernel parameters string in /flash/cmdline.txt.

This has the same effect as forcing it using hdmi_mode in config.txt with older kernels/drivers.

closed 2022-06-10 by Ntemis