libretro/Lakka-LibreELEC #1555: further performance regressions on NUC11 (Bean Canyon) intel system / gl render issues?

auxfam, 2021-11-24:
>   Which version of Lakka are you using?
    Lakka-Generic.x86_64-nightly-20211119-d3a7bbf.img.gz

>    What system hardware are you using?
    intel NUC8 Intel® Core™ i5-8259U Processor (6M Cache, up to 3.80 GHz)
    note: this has 128 EDRAM as a L4 cpu/GPU cache

>    What did you do?
    installed latest nighties as above
    run flycast
    with gl backend (also tested with glcore)
    load Dead or Alive
    let intro run
    play a couple of cpu matches

 >   What did you expect to happen?
    in game match (v cpu) is expected to perform around 58-60. when with following deviations from default
   power profile: max_performence
    Internal resolution: 1280x960
    accuracy: medium
    latency number of buffers: 3

    What happened instead?
    intro sees drops to 40+ fps
    matches sees drops to 40+ fps

    other notes:
somewhat of a perfomance regression here.    I now need to go down to 
    Internal resolution: 640x480
    accuracy: high
    latency number of buffers: 3
to get somewhat 58 fps.    on medium accuracy I get some fps back but polygon clipping and glitch occurs on modles (flicker in and out)   I am reporting this to note as it likely is driver/firmware changes for the builds in the last few months

other titles:
Citra -> Resident evil revalations -> x2 resolution -> used to be aprpox 57-59 fps
now I have to drop down to x1 resolution and even then unstable

This report is more of a note of the odd performence reductions in intel NUC8 over the past few weeks of nightlies.  of which I can only observe kernal nad driver changes being possible causes (not that we can necessarily do anything about Mesa et al)

auxfam, 2022-01-21:
I am closing this since haven't been able to find any further possible causes.  it does appear to be driver related as the for example the vulkan driver for `Flycast` performs notably worse on 60fps titles like DOA (glcore: 58-> 60 -> vulkan:47-52).  but for anyone also using intel nucs I guess you may also be experiencing regression on some of the heavier cores

closed 2022-01-21 by auxfam

Ntemis, 2022-01-21:
If you still need this fixed you better report it to the individual cores github page