libretro/Lakka-LibreELEC #1556: Snes9x (2010) fails to load save states on RG351V

FishBulb32, 2021-11-25:
- Which version of Lakka are you using?
3.6 and the latest nightly as of today.

- What system hardware are you using?
Anbernic RG351V

- What did you do?
Run a SNES rom using Snes9x (2010). Create a save state. Then load the save state.

- What did you expect to happen?
Save state should load.

- What happened instead?
System crashed as the save state was selected. Retro arch reloads.

It would be great if snes9x (2010) would run ok on these devices as it’s seems the best all round core for snes games.

ToKe79, 2022-03-07:
@FishBulb32 can you test with [latest release](https://le-builds.lakka.tv/OdroidGoAdvance.arm/)? maybe it was bug in the core itself that was fixed in later version of the core. thank you!

ToKe79, 2022-04-01:
refers to discontinued branch, only updates to RetroArch / libretro-cores. could be also bug in the core (which could be fixed with an update of the core in the future).

closed 2022-04-01 by ToKe79