libretro/Lakka-LibreELEC #1558: RG351V battery indicator and charging.

Lucius1202, 2021-11-28:
I would like to report a problem with the battery indicator.

- Which version of Lakka are you using? 
- Lakka (official) 3.6 
- What system hardware are you using?
- Anbernic RG351V
- What did you do?
- Charge the device.
- What did you expect to happen?
- The battery normally charges up to 100%
- What happened instead?
- The battery gauge does not rise above 43 percent and the LED behaves accordingly (orange to red).

"Long explanation".
With the original system received with console, the battery normally charges up to 100% and discharges accordingly, the LED behaves accordingly, green, yellow, orange and finally red below 20%. When the system is 100% charged with green LED, I turn off the console, replace with official Lakka 3.6, and the battery reports 43% and orange LED. The console is new as well as the SD cards.

Graham15, 2021-12-06:
I have noticed the same with my Anbernic RG351MP. 351ELEC reports 100% but Lakka 3.6 reporting 39%.

Lucius1202, 2021-12-29:
After a few days I thought my console was broken and asked for a new unit. After a first charge and discharge the problem has reoccurred, indicator at 37% but battery that seems fully charged. The legacy system is retropie based and works normally. Hope this information will help.

Graham15, 2021-12-29:
Mine is now showing 0% even though it's 100% charged. Works fine in 351Elec. Tried Lakka 4 (dev) but unfortunately, same issue.

Lucius1202, 2022-01-11:
I was thinking if it is possible to get the battery level through ssh? Any system files to read with grep or cat? Maybe with other commands?

ToKe79, 2022-04-01:
not fixing this. there is plan to port RK3326 and all the consoles (Odroid Go Advance/Super, Anbernic RG351 series, ...) to the 4.x branch in the future, so focus on the port, not fixing old bug, sorry. As a workaround you may try to use .dtb file from 351ELEC build (helps with RG351MP at least).

closed 2022-04-01 by ToKe79