libretro/Lakka-LibreELEC #1559: Raspberry Pi 4. Can't connect to Wifi

Cibomatto2002, 2021-11-30:
- Which version of Lakka are you using? 3.6
- What system hardware are you using? Raspberry Pi 4
- What did you do? try to use Wifi
- What did you expect to happen? Wifi to work
- What happened instead?  had problems connecting to Wifi I could see the networks, and I put in the password for my network but it didn't see to connect)

RobLoach, 2021-12-24:
There's been some changes to the network stuff in the latest RetroArch. Perhaps an update will fix this... I've found I've had to have some patience when using the wifi menu, but the latest Lakka has worked for me....

Try a 2.4GHz network?

Cibomatto2002, 2021-12-24:
Thanks for the reply i don't see the 2.4Ghz option it just shows NETGEAR

RobLoach, 2022-01-09:
Just tried on a RPi3 and couldn't open the Wifi menu. Will have to do more testing and get the logs.

loki666, 2022-02-04:
had the same issue with mine,
seems related to wifi power saving
try adding file /storage/.config/udev.rules.d/81-wifi-powersave.rules

`ACTION=="add", SUBSYSTEM=="net", KERNEL=="wl*", RUN+="/usr/sbin/iw dev $name set power_save off"`

ToKe79, 2022-02-21:
was this fixed with the addition of `HAVE_WIFI=1` to the `retroarch` build?

RobLoach, 2022-02-21:
I tested the latest on a RPi3, and can confirm I was able to connect to wifi.

loki666, 2022-02-21:
but the original issue was on Lakka 3.6 where Wifi options / drivers was still fine

loki666, 2022-02-21:
issue looked like bad Wifi reception (all RPi seems affected by this) that's why I suggested to disable power saving on the wifi interface

ToKe79, 2022-03-02:
no response from OP. closing.