libretro/Lakka-LibreELEC #1560: Flycast Libretro Upstream Repo

Mr-Jake, 2021-12-01:
Lakka still builds Flycast from the Libretro repo:
'''
packages/libretro/flycast/package.mk 

PKG_NAME="flycast"
PKG_VERSION="beb9a61"
PKG_ARCH="any"
PKG_LICENSE="GPLv2"
PKG_SITE="https://github.com/libretro/flycast"
'''

However, there is a notice in the about section in the Libretro repo.
'''
About

Flycast is a multiplatform Sega Dreamcast emulator. NOTE: No longer actively developed, use upstream repo for libretro from now on - https://github.com/flyinghead/flycast 
'''

Should Lakka switch to the Flycast Libretro upstream repo?

Ntemis, 2021-12-01:
We will if this is valid. Thanks for noticing it

RobLoach, 2021-12-07:
It looks like most of libretro references the libretro/flycast repo:
https://github.com/libretro/libretro-super/blob/master/recipes/linux/cores-linux-x64-generic#L114

gouchi, 2021-12-07:
Yes we will have to switch to upstream repository  [https://github.com/flyinghead/flycast](https://github.com/flyinghead/flycast).

It will be done when we release the new Lakka version with RA version 1.9.14.

RobLoach, 2021-12-22:
Thanks for providing the updated package file :+1:

closed 2021-12-22 by RobLoach