libretro/Lakka-LibreELEC #1561: WiFi/Ethernet Issues with NUC11

sharkytowers, 2021-12-06:
Lakka does not recognize Ethernet or Wifi adapter on NUC11 (1165G7).  Details in the thread below.  I'm fairly confident this is a Lakka issue as everything else seems to recognize them (other linux installs, windoze, etc)

https://forums.libretro.com/t/nuc-11-networking-not-working-wireless-or-ethernet/35837

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?  Latest stable and tried the most recent nightly
- What system hardware are you using? NUC 11 i7 (1165g7) - 16gb/1tb
- What did you do? Tried to use Lakka like a nice little boy.
- What did you expect to happen? Thought I'd be in retro heaven.
- What happened instead? I have to use a separate USB-C ethernet adapter (the HORROR!!!)

Links to logs in the above thread.  Also links to logs where I generated them while not using a network adapter (significantly "out-kicking my coverage" technology speaking.)

gouchi, 2021-12-06:
We will have to enable [CONFIG_IGC](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Generic/linux/linux.x86_64.conf#L2011) in the kernel config for the Ethernet.

For the WiFi it seems we are missing some firmware for iwlwifi driver.

```
[    5.909670] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    5.911090] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-59.ucode failed with error -2
[    5.911103] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-58.ucode failed with error -2
[    5.911115] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-57.ucode failed with error -2
[    5.911126] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-56.ucode failed with error -2
[    5.911138] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-55.ucode failed with error -2
[    5.911149] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-54.ucode failed with error -2
[    5.911160] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-53.ucode failed with error -2
[    5.911171] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-52.ucode failed with error -2
[    5.911182] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-51.ucode failed with error -2
[    5.911193] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-50.ucode failed with error -2
[    5.911206] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-49.ucode failed with error -2
[    5.911217] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-48.ucode failed with error -2
[    5.911228] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-47.ucode failed with error -2
[    5.911240] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-46.ucode failed with error -2
[    5.911251] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-45.ucode failed with error -2
[    5.911264] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-44.ucode failed with error -2
[    5.911275] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-43.ucode failed with error -2
[    5.911285] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-42.ucode failed with error -2
[    5.911293] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-41.ucode failed with error -2
[    5.911299] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-40.ucode failed with error -2
[    5.911306] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-39.ucode failed with error -2`
```

We will have to update [iwlwifi firmware](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-firmware/iwlwifi-firmware) package to [latest commit](https://github.com/LibreELEC/iwlwifi-firmware).

gouchi, 2021-12-07:
@sharkytowers Can you make test with this [image](https://nightly.builds.lakka.tv/members/gouchi/1561/Lakka-Generic.x86_64-3.6-devel-20211206230642-9e4af94.img.gz) ?

Thank you.

sharkytowers, 2021-12-11:
> @sharkytowers Can you make a test with this [image](https://nightly.builds.lakka.tv/members/gouchi/1561/Lakka-Generic.x86_64-3.6-devel-20211206230642-9e4af94.img.gz) ?
> 
> Thank you.

No prob.  Sorry, just seeing this.  I'll report back momentarily.

sharkytowers, 2021-12-11:
@gouchi Ethernet is all good.  Works a treat.  Wireless still doesn't work.  When I "Connect to Wifi" it says it is scanning (but doesn't) then says no networks found and won't even pretend to start to scan again (if that makes sense).  I disabled wifi and re-enabled it.  Also logged out and in.  

Having Ethernet is great though.  Let me know if there is anything else you want me to do or if you want logs or anything.  Happy to do it.

gouchi, 2021-12-11:
Thank you for your test.

May you try to provide some log with [createlog](https://www.lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) ?

Thank you.

sharkytowers, 2021-12-11:
No problem @gouchi   Here is a link to the most recent log.  To try and recreate the issue I just toggled the Wifi on and off and tried to connect.  Hope it is helpful.  

https://drive.google.com/file/d/19neDhda0OvT67LGJ9PoLMEC2haNfXzTr/view?usp=sharing

gouchi, 2021-12-11:
Thank you for test.

Strange it seems it does not succeed to load iwlwifi ?

Can you try to copy iwlwifi firmwares (`iwlwifi-*.ucode`) from [linux-firware](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/snapshot/linux-firmware-20211027.tar.gz) in [/storage/.config/firmware](https://wiki.libreelec.tv/how-to/add-firmware) ?

Don't forget to reboot one time after copying iwlwifi firmwares.

Thank you.

sharkytowers, 2021-12-15:
I'm sorry @gouchi .  Where exactly do you want me to copy the iwlwifi-*.ucode files?  I'm not seeing a /storage/.config/firmware directory.  Should I create this directory on the USB stick and move them there?  Sorry for being so dense.

gouchi, 2021-12-15:
No problem, sorry I forgot to mention you have to create the directory if it is not existing.

Thank you.

sharkytowers, 2021-12-24:
Okay I copied all the iwlwiw-*.ucode files to a directory (that I created) at /storage/.config/firmware/

And...  it works!  Kind of...  it does find the wireless networks now when searching for them (which it didn't do before!). However, once I enter my password for the network is fails. Or at least it says it does.  I rebooted with the ethernet cable disconnected and it found one of the previous wireless networks.  So it is saying "failed" but it's actually working?

Seems like progress.

ToKe79, 2022-03-07:
@sharkytowers Lakka/RetroArch shows "Failed" message after connecting to an AP for the first time, but it actually connects - you can check via main menu &rarr; information &rarr; network information &rarr; wlan0. see [this](https://nightly.builds.lakka.tv/members/vudiq/lakka-wifi-connect.mkv) low quality video.

@gouchi we do not include `iwlwifi-*` in `x86_64.dat` - did you do that in your test images above?

gouchi, 2022-03-08:
@ToKe79 IIRC I only enabled [CONFIG_IGC](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Generic/linux/linux.x86_64.conf#L2011), I don't recall enabling `iwlwifi-*` in `x86_64.dat`.

Yes I think we should try to update iwlwifi-firmware until the [commit to the kernel version we are using](https://github.com/LibreELEC/iwlwifi-firmware/commits/master).

ToKe79, 2022-03-21:
@sharkytowers it would be very kind of you, if you could test with [this image](https://nightly.builds.lakka.tv/PRs/pr_1641_000479/Generic.x86_64/Lakka-Generic.x86_64-4.0-devel-20220320160024-586f9f6.img.gz), so we know that our fix is working :-)
Thank you in advance!

PR #1641 (2022-03-21): fix broken ethernet and wifi for NUC11
closes: #1561