libretro/Lakka-LibreELEC #1562: build failure: px68k is missing a commit?

theflyingape, 2021-12-09:
**There is no commit 'ea6e5a7' on branch 'master' of package 'px68k'! Aborting!**

Weird, is this a github package issue?  I am pasting the console output:
```
$ git pull
Already up to date.

$ DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=aarch64 make image

INSTALL      libzip (target)
[317/363] [DONE] install libzip
GET      px68k (git)
    GIT CLONE      px68k
Cloning into '/home/theflyingape/src/Lakka-LibreELEC/sources/px68k/px68k-ea6e5a7'...
There is no commit 'ea6e5a7' on branch 'master' of package 'px68k'! Aborting!
FAILURE: scripts/install px68k has failed!
[318/363] [FAIL] install px68k

The following logs for this failure are available:
  stdout: /home/theflyingape/src/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-3.6-devel/.threads/logs/325/stdout
  stderr: /home/theflyingape/src/Lakka-LibreELEC/build.Lakka-RPi4.aarch64-3.6-devel/.threads/logs/325/stderr

parallel: This job failed:
package_worker 1 325 363 'install px68k'
```

Demon000, 2021-12-19:
I believe their repo got force-pushed.

Demon000, 2021-12-20:
https://github.com/libretro/Lakka-LibreELEC/pull/1566/commits/c30bfcce8a183076acb1acdf078f50709ff54862

RobLoach, 2021-12-22:
The `libretro-update` branch is targeting this one: https://github.com/libretro/px68k-libretro/commit/b309941140f03b3d52e4c1720d6f4cbc0581f301

https://github.com/libretro/Lakka-LibreELEC/pull/1564

closed 2021-12-22 by RobLoach