libretro/Lakka-LibreELEC #1563: Recording / Streaming is not working. (lakka 3.6 or latest nightly build)

ups-ups, 2021-12-10:
- Which version of Lakka are you using?
-  Lakka-RPi3.aarch64-nightly-20211125-9e4af94.img.gz
- What system hardware are you using?
-  Raspberry Pi 3B plus
- What did you do?  
-  I set my twitch streaming url in 'Recording' config. After loading contents, I started streaming via quick menu. 
- What did you expect to happen?
-  Streaming to twitch. I checked the streaming url is correct. I can stream with same url with retroarch in 64bit windows 10.
- What happened instead?
-  It was not working and retroarch rebooted.
-  Same trouble is occored in lakka for 64bit Generic PC.
-  ’Start recording' is also same behavior.

logs:
[INFO] [CORE]: No content, starting dummy core.
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 1374, Frames dropped: 0.
[INFO] [Config]: Loading config from: "/storage/.config/retroarch/config/RetroArch-1210-014321.cfg".
[INFO] [Config]: Saved new config to "/storage/.config/retroarch/config/RetroArch-1210-014321.cfg".
[INFO] [Environ]: SET_SUBSYSTEM_INFO.
[INFO] [CONTENT LOAD]: Updating firmware status for: /tmp/cores/fbneo_libretro.so on /tmp/system
[INFO] [CORE]: Using content: /storage/roms/downloads/alienar.zip.
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 1496, Frames dropped: 0.
[INFO] RetroArch 1.9.13 (Git fca72f67d1)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Nov 13 2021
[INFO] Version: 1.9.13
[INFO] Git: fca72f67d1
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/fbneo_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/FinalBurn Neo/FinalBurn Neo.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/FinalBurn Neo/downloads.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/FinalBurn Neo/alienar.cfg".
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/alienar.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/alienar.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_MESSAGE_INTERFACE_VERSION.
[INFO] [Environ]: SET_AUDIO_BUFFER_STATUS_CALLBACK.
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[libretro INFO] Setting save dir to /storage/savefiles
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] Setting system dir to /tmp/system
[INFO] [Environ]: SET_CONTROLLER_INFO.
[libretro INFO] Initialize DIP switches.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2.
[libretro INFO] [FBNeo] Searching all possible locations for romset alienar
[libretro INFO] [FBNeo] No patched romset found at /tmp/system/fbneo/patched/alienar
[libretro INFO] [FBNeo] Romset found at /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] No romset found at /tmp/system/fbneo/alienar
[libretro INFO] [FBNeo] No romset found at /tmp/system/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0x6feb0314 and name aarom10 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0xae3a270e and name aarom11 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0x6be9f09e and name aarom12 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0xbb0c21be and name aarom01 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0x165acd37 and name aarom02 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0xe5d51d92 and name aarom03 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0x24f6feb8 and name aarom04 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0x5b1ac59b and name aarom05 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0xda7195a2 and name aarom06 from archive /storage/roms/downloads/alienar
[libretro INFO] [FBNeo] Using ROM with known crc 0xf9812be4 and name aarom07 from archive /storage/roms/downloads/alienar
[... 151 lines cut ...]
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Environ]: SET_SUBSYSTEM_INFO.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[libretro INFO] [FBNeo] Unknown device type for port 0, forcing "Classic" instead
[libretro INFO] [FBNeo] Unknown device type for port 1, forcing "Classic" instead
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/config/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/config/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/config/content_video_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/config/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/config/content_favorites.lpl].
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [KMS]: New FB: 1280x720 (stride: 5120).
[INFO] [KMS]: New FB: 1280x720 (stride: 5120).
[INFO] [recording] Custom timing given: FPS: 60.0000, Sample rate: 48000.0000
[INFO] [recording] Recording to rtmp://live.twitch.tv/app/live_******************************@ 292x240. (FB size: 292x292 pix_fmt: 2)

[end of log file]
# Streaming url is masked.

bonswouar, 2021-12-10:
After some time digging into that (and asking for help on discord) I was finally going to create an issue, good to see I'm not the only one having this problem!

So some complementary details:
- It's actually a `Segmentation Fault`
- Here is the gdb dump :
```bash
#0  0x0000000000000000 in ?? ()
No symbol table info available.
#1  0x0000000000424fab in record_driver_init_first (backend=0xc61540 <recording_state>, data=0xc61548 <recording_state+8>, params=0x7ffe976de3c0) at record/record_driver.c:153
        handle = <optimized out>
        i = 1
#2  recording_init () at record/record_driver.c:410
        output = "/storage/.config/retroarch/records/10-Yard Fight (USA)-211203-193421.mkv\000G\325 \366\177\000\000\020\347m\227\376\177\000\000z\330.!\366\177\000\000\375\377\377\377\000\000\000\000v", '\000' <repeats 24 times>, "z\353\001\000\000\000\000\300Y\330 \366\177\000\000S", '\000' <repeats 39 times>, "\240\037\000\000\377\377", '\000' <repeats 18 times>...
        buf = "10-Yard Fight (USA)-211203-193421.mkv", '\000' <repeats 11 times>, "@\256\025\002", '\000' <repeats 28 times>, "\340\372m\227", '\000' <repeats 12 times>, "\330\b\000\000\000\000\001\000\000\000\000\000\001", '\000' <repeats 19 times>, "/tmp/cores/bnes_libretro.so", '\000' <repeats 45 times>...
        params = {fps = 60.099826520671044, samplerate = 32000, filename = 0x7ffe976de420 "/storage/.config/retroarch/records/10-Yard Fight (USA)-211203-193421.mkv", 
          config = 0x7ff61e8090b2 "/storage/roms/downloads/twitch.cfg", audio_resampler = 0x7ff61e7f5e59 "sinc", out_width = 256, out_height = 240, fb_width = 256, fb_height = 240, channels = 2, 
          video_record_scale_factor = 1, video_stream_scale_factor = 1, video_record_threads = 2, streaming_mode = 0, aspect_ratio = 1.06666672, preset = RECORD_CONFIG_TYPE_RECORDING_MED_QUALITY, 
          pix_fmt = FFEMU_PIX_RGB565, video_gpu_record = false}
        settings = <optimized out>
        video_st = <optimized out>
        av_info = <optimized out>
        runloop_st = <optimized out>
        video_gpu_record = <optimized out>
        video_force_aspect = true
        current_core_type = <optimized out>
        video_driver_pix_fmt = <optimized out>
        recording_st = 0xc61540 <recording_state>
        recording_enable = true
#3  0x000000000041977e in command_event (cmd=<optimized out>, data=<optimized out>) at retroarch.c:4744
        boolean = false
        p_rarch = 0xc57f80 <rarch_st>
        runloop_st = 0xc39d80 <runloop_state>
        access_st = 0xc39d60 <access_state_st>
        menu_st = 0xcf87a0 <menu_driver_state>
        video_st = 0xc9ed00 <video_driver_st>
        settings = 0x7ff61e7f4010
        recording_st = 0xc61540 <recording_state>
```
- Someone suggested to add `x264` to my Lakka build (**_shouldn't this be included by default?_**). I've enabled it in the `ffmpeg` package and added the library `x264` build itself (and did the same for `libmp3lame` just in case), but I still had segfaults
- Unfortunately I can't build Lakka with debug on my laptop _(not enough ram)_, and I have issues when building it on my server _(had to change dependencies and other stuff because of some errors, even had to remove `mrboom` package because of some "incompatible build options")_ so I finally gave up. **The _(older)_ debug build I used here has been provided by "garvin" on Discord, and doesn't include my modifications with `ffmpeg`/`x264`, thus I can't know what's causing the segfault with my build**.
- Playing with gdb I've been able to go a bit deeper with the `non-x264` debug build, and identified the error occured in `record/drivers/record_ffmpeg.c`, specifcally `L888`, because `ctx->oformat` is `0x0`. Taking a look at what's returned by `avformat_alloc_context()` it seems to confirm `ffmpeg` wasn't initialized properly because of codecs missing, which fits the original assumption:
```gdb
(gdb) p *ctx
$6 = {av_class = 0x7f6020273d80 <av_format_context_class>, iformat = 0x0, oformat = 0x0, priv_data = 0x0, pb = 0x0, ctx_flags = 0, nb_streams = 0, streams = 0x0, 
  filename = "/storage/.config/retroarch/records/10-Yard Fight (USA)-211210-143558.mkv", '\000' <repeats 951 times>, url = 0x0, start_time = 0, duration = 0, bit_rate = 0, packet_size = 0, max_delay = -1, 
  flags = 2097152, probesize = 5000000, max_analyze_duration = 0, key = 0x0, keylen = 0, nb_programs = 0, programs = 0x0, video_codec_id = AV_CODEC_ID_NONE, audio_codec_id = AV_CODEC_ID_NONE, 
  subtitle_codec_id = AV_CODEC_ID_NONE, max_index_size = 1048576, max_picture_buffer = 3041280, nb_chapters = 0, chapters = 0x0, metadata = 0x0, start_time_realtime = -9223372036854775808, fps_probe_size = -1, 
  error_recognition = 1, interrupt_callback = {callback = 0x0, opaque = 0x0}, debug = 0, max_interleave_delta = 10000000, strict_std_compliance = 0, event_flags = 0, max_ts_probe = 50, avoid_negative_ts = -1, 
  ts_id = 0, audio_preload = 0, max_chunk_duration = 0, max_chunk_size = 0, use_wallclock_as_timestamps = 0, avio_flags = 0, duration_estimation_method = AVFMT_DURATION_FROM_PTS, skip_initial_bytes = 0, 
  correct_ts_overflow = 1, seek2any = 0, flush_packets = -1, probe_score = 0, format_probesize = 1048576, codec_whitelist = 0x0, format_whitelist = 0x0, internal = 0x2a49e40, io_repositioned = 0, 
  video_codec = 0x0, audio_codec = 0x0, subtitle_codec = 0x0, data_codec = 0x0, metadata_header_padding = -1, opaque = 0x0, control_message_cb = 0x0, output_ts_offset = 0, dump_separator = 0x2a08530 ", ", 
  data_codec_id = AV_CODEC_ID_NONE, open_cb = 0x0, protocol_whitelist = 0x0, io_open = 0x7f60201b53e0 <io_open_default>, io_close = 0x7f60201b53d0 <io_close_default>, protocol_blacklist = 0x0, 
  max_streams = 1000, skip_estimate_duration_from_pts = 0, max_probe_packets = 2500}
```
- I have no idea if that's a Lakka or RetroArch problem, but seeing x264 isn't included in Lakka I'd guess it's more likely a Lakka problem

GavinDarkglider, 2022-02-24:
This is fixed with this PR https://github.com/libretro/Lakka-LibreELEC/pull/1617 That being said, I didnt port it to 3.x tree yet, but it should be easy enough to do..... Give me an hour, and I will submit a patch there as well for that.

gouchi, 2022-02-27:
@ups-ups @bonswouar PR has been merged please report if it is working with [nightly builds](https://nightly.builds.lakka.tv/latest/) so that we can close this issue.

Thank you.

Kind regards,

bonswouar, 2022-02-27:
Amazing, thanks for this fix @GavinDarkglider and the rest of the team!

@gouchi I confirm it works on `4.x` branch (that I built myself earlier), although not on `3.x` using Online Updater, but I believe the nightly build happened before the merge (I'm seeing Feb 26 2022 as the Build Date for version `b71be7efd2`)?

smilerliu, 2022-02-28:
I have a GPI case + Pi Zero 2 W. I updated to the latest nightly build. When I tried to record a GB game using the gambatte core, the game ran fine except for some very loud background white noise, but no files were written to the recordings directory.

gouchi, 2022-02-28:
@bonswouar @smilerliu Thank you for the tests. I am glad it is working.

@GavinDarkglider Thank you for the fix.

closed 2022-02-28 by gouchi

ups-ups, 2022-02-28:
Thank you for the fix.
I checked x86-64 or rpi3 latest nightly build.
It completely worked.
Streaming to twitch is also worked.