libretro/Lakka-LibreELEC #1567: TP-LINK UB500 Bluetooth adapter

Torrens1, 2021-12-21:
#Lakka 3.6 #PC #Add Bluetooth device
Hello, i tray to add external bluetooth device to my old pc but lakka propably didint support new bluetooth adapter TP-LINK UB500. 
I tray [https://askubuntu.com/questions/1370663/bluetooth-scan-doesnt-detect-any-device-on-ubuntu-21-10](url) but i faild.

Torrens1, 2021-12-24:
[log-2021-12-24-08.02.45.zip](https://github.com/libretro/Lakka-LibreELEC/files/7773332/log-2021-12-24-08.02.45.zip)

gouchi, 2021-12-24:
It seems this device has been supported using [mainline Linux](https://github.com/torvalds/linux/blob/v5.16-rc1/drivers/bluetooth/btusb.c#L446) (v5.16 when it is released). It seems to be available also on [Linux stable version](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/bluetooth/btusb.c?h=v5.15.11#n437) (v5.15).

Otherwise we might try to include this [patch](https://www.spinics.net/lists/linux-bluetooth/msg94476.html) as we are running Linux v5.10 (longterm version).

[Source Kernel version](https://www.kernel.org/).

Thank you.

closed 2022-03-21 by Ntemis

Torrens1, 2022-08-26:
Hi, 
I left this problem for a while. But I went into the project and discovered that the latest version of lakka supports this device. however, I was very upset that it does not work properly. and still does not detect bluetooth devices.

Torrens1, 2022-08-30:
> Hi, I left this problem for a while. But I went into the project and discovered that the latest version of lakka supports this device. however, I was very upset that it does not work properly. and still does not detect bluetooth devices.

[log-2022-08-29-12.17.02.zip](https://github.com/libretro/Lakka-LibreELEC/files/9451204/log-2022-08-29-12.17.02.zip)

by-gnome, 2022-11-17:
After updating kernel-firmware to 20221109 (PR #1778) which includes `rtl_bt/rtl8761b_fw.bin` this issue should be solved.

gouchi, 2023-02-14:
@Torrens1 Will it possible to make a test with [Lakka 4.3](https://le-builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-4.3.img.gz)  ?  If it is still not working you can try [Lakka 5.x dev build](https://nightly.builds.lakka.tv/latest/Generic.x86_64/).

Thank you.

gouchi, 2023-09-20:
@Torrens1 Did you have the opportunity to make some test ?

gouchi, 2023-12-19:
Please open a new issue if it is still not working.

PR #1645 (2022-03-21): kernel/Lakka: add support for ASUS BT500 and TP-LINK UB500
fixes: #1567