libretro/Lakka-LibreELEC #1568: Raspberry Pi Zero 2 - what image should I use?

armadillu, 2021-12-21:
Hi, I'm trying to get Lakka running on a Raspberry Pi Zero 2. I tried the latest dev and the last release for Pi Zero 1, but It doesn't seem to boot?

I think the Pi Zero 2 should show up on this list in the download page as well... http://www.lakka.tv/get/macos/

RobLoach, 2021-12-22:
Don't believe we're building Zero 2 builds right now. Looks like LibreELEC doesn't even have them? https://libreelec.tv/downloads/raspberry/ Could you try the 9.2 build of LibreELEC?

smilerliu, 2021-12-26:
If you are using Pi Zero 2 W with GPI case: https://nightly.builds.lakka.tv/members/vudiq/rpi02gpi/

armadillu, 2021-12-27:
I just tried it, it seems to get stuck at the rainbow square boot screen...

armadillu, 2022-01-09:
A little more on this, it seems there's a working WIP version based on libreELEC 11... But it seems lakka is on 9, targeting 10?

https://forum.libreelec.tv/thread/24778-raspberry-pi-zero-2-w/?pageNo=1

smilerliu, 2022-01-11:
I've been using the image I linked for a long time. Don't use any image in the sub-directories; they don't work.

armadillu, 2022-01-12:
I've tried this image, but I'm not targeting the GPI case case. I modified the `distroconfig.txt` and `config.txt` files to match my setup, but right after the lakka logo appears, my hdmi screen goes to sleep and can't really tell what's going on.

smilerliu, 2022-01-17:
If you are not using the GPI case, have you tried the pi 2/3 arm build or pi 3 aarch64 build? I thought zero 2 is basically a pi 3.

RobLoach, 2022-03-01:
Zero 2 is up! http://lakka.tv/articles/2022/03/01/lakka-3.7/

closed 2022-03-01 by ToKe79

ustix, 2022-03-07:
> Zero 2 is up! http://lakka.tv/articles/2022/03/01/lakka-3.7/

Not working with my zero 2 without GPI case, I don't know if that is an issue, or what, but it's not loading at all. I've made it with raspberry imager.

armadillu, 2022-03-07:
Same here, I could get to the Lakka booting logo after tweaking the config.txt file to match my screen, but it never gets to Retroarch. Because I cant config wifi (maybe there's a way to set it up through the config file?), I cant ssh to get to logs or anything.

ToKe79, 2022-03-07:
please, use RPi3.aarch64 nightly build for RPi Zero 2 without GPICase mod.
https://nightly.builds.lakka.tv/latest/

ustix, 2022-03-08:
> please, use RPi3.aarch64 nightly build for RPi Zero 2 without GPICase mod. https://nightly.builds.lakka.tv/latest/

That would have been my other choice, and it's working like a charm. But it would be good if there could be an optimized  version for zero 2 without the case.