libretro/Lakka-LibreELEC #1577: TinkerBoard: mame2003-plus fail

RobLoach, 2022-01-09:
https://nightly.builds.lakka.tv/2022-01-09/0_logs/TinkerBoard.arm-2022-01-09.html

``` bash
src/lib/zlib/zutil.c:20:4: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   20 |    "insufficient memory", /* Z_MEM_ERROR     (-4) */
      |    ^~~~~~~~~~~~~~~~~~~~~
src/lib/zlib/zutil.c:21:4: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   21 |    "buffer error",        /* Z_BUF_ERROR     (-5) */
      |    ^~~~~~~~~~~~~~
src/lib/zlib/zutil.c:22:4: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   22 |    "incompatible version",/* Z_VERSION_ERROR (-6) */
      |    ^~~~~~~~~~~~~~~~~~~~~~
src/lib/zlib/zutil.c:23:4: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   23 |    ""};
      |    ^~
Linking mame2003_plus_libretro.so...
platform unix
/home/lakkabuildbot/nightly/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm/toolchain/lib/gcc/armv7ve-libreelec-linux-gnueabi/9.4.0/../../../../armv7ve-libreelec-linux-gnueabi/bin/ld.gold: error: src/machine/namcos1_machine.o: file is empty
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:897: mame2003_plus_libretro.so] Error 1
make[1]: Leaving directory '/home/lakkabuildbot/nightly/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm/mame2003-plus-5ad3eaf'
FAILURE: scripts/install mame2003-plus has failed!
[333/342] [FAIL] install mame2003-plus

The following logs for this failure are available:
  stdout: /home/lakkabuildbot/nightly/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm/.threads/logs/279/stdout
  stderr: /home/lakkabuildbot/nightly/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm/.threads/logs/279/stderr

parallel: This job failed:
package_worker 26 279 342 'install mame2003-plus'
Parallel build failure - see log for details. Time of failure: Sun Jan  9 03:48:17 CET 2022
make: *** [Makefile:12: image] Error 1
```

RobLoach, 2022-01-09:
@mahoneyt944 Would you happen to know why this would fail? I see `namcos1_machine` in the Makefile.

mahoneyt944, 2022-01-09:
I haven't worked with these boards before unfortunately. Is this a new found issue or a first attempt to build? `namcos1_machine` hasn't been altered recently so not sure if this file is relevant or not, may just be the lucky number it stopped on from a build environment issue. Memory? I've also seen fails with regards to line endings before https://github.com/libretro/mame2003-plus-libretro/pull/1141 might be worth verifying but unlikely.

RobLoach, 2022-01-16:
@spleen1981 Thanks for fixing that up! Looks like the builds are going through now :rocket:

closed 2022-01-16 by RobLoach