libretro/Lakka-LibreELEC #1581: Please include bluetooth firmware for Qualcomm Atheros QCA6174 from upstream

adamrdrew, 2022-01-15:
Hi all! Love the project having a blast with it. On my system with a Qualcomm Atheros QCA6174 WIFI / Bluetooth combo card Bluetooth didn't work. It would scan and never find devices. SSH'ed in and saw the problem was the firmware binaries weren't included:

From dmesg:
```
[    3.702427] bluetooth hci0: Direct firmware load for qca/rampatch_usb_00000302.bin failed with error -2
[    3.702431] Bluetooth: hci0: failed to request rampatch file: qca/rampatch_usb_00000302.bin (-2)
[    3.713954] bluetooth hci0: Direct firmware load for qca/rampatch_usb_00000302.bin failed with error -2
[    3.713959] Bluetooth: hci0: failed to request rampatch file: qca/rampatch_usb_00000302.bin (-2)
```
In bluetoothctl:
```
[bluetooth]# power on
No default controller available
```
I wade /var/lib/firmware/qca and then grabbed the rampatch and NVM files from kernel.org:
```
Lakka:/var/lib/firmware/qca # wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firm
ware.git/plain/qca/rampatch_usb_00000302.bin
Lakka:/var/lib/firmware/qca # wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firm
ware.git/plain/qca/nvm_usb_00000302.bin
```

Reloaded btusb and its working great now:
```
Lakka:/var/lib/firmware/qca # bluetoothctl
Agent registered
[bluetooth]# power on
Changing power on succeeded
[CHG] Controller E4:AA:EA:BF:C4:30 Discovering: yes
```

So, yeah - if y'all could include these firmwares that would be awesome! Thanks!!!!

Ntemis, 2022-01-21:
What is your system? Because we do ship those firmwares

adamrdrew, 2022-01-21:
Hi thanks for the response! Yeah, something must be up because those firmware files my system needed (`rampatch_usb_00000302.bin` & `nvm_usb_00000302.bin`) weren't in `/var/lib/firmware/qca` on a fresh install of Lakka. My BASH isn't so strong these days so I can't fully understand that script, but it looks like it iterates over a firmware list file. Maybe those files aren't in that list file on the version of kernel you're based on? You'd know better than me, but I figured I'd suggest it.

I'm on a Lenovo Ideacentre 3 with an AMD Athlon Silver 3050U and a  Qualcomm Atheros QCA6174 WIFI / Bluetooth combo card. Everything else works perfectly out of the box. I will be happy to provide any debug data you request. Thanks for your work!

Ntemis, 2022-01-21:
Both are there, plot thickens

gouchi, 2022-01-24:
We might have to add `qca/` in [x86_64.dat file](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-firmware/kernel-firmware/firmwares/x86_64.dat).

Or you can add the firmware to [/storage/.config/firmware](https://wiki.libreelec.tv/how-to/add-firmware).

ToKe79, 2022-03-07:
firmware is now included in the 4.x images:

https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/packages/linux-firmware/kernel-firmware/firmwares/x86_64.dat#L5

@adamrdrew you can test with latest [nightly build](https://nightly.builds.lakka.tv/latest/)

closed 2022-03-07 by ToKe79