libretro/Lakka-LibreELEC #1586: Database: Directory Not Found

RobLoach, 2022-01-16:
Seems the RDBs in the database are not there on install.

>Which version of Lakka are you using? 

Lakka-RPi3.aarch64-nightly-20220116-8bb17e6.img.gz

> What system hardware are you using?

RPi3

> What did you do?

Install the latest build of Lakka, and then visit Information > Database Manager. Find that there are no files. "Directory Not Found"

> What did you expect to happen?

For there to be a list of .rdb files.

> What happened instead?

There are no .rdb files listed in the Database Manager list.

RobLoach, 2022-01-16:
It could be my SD card though, so if someone else could verify the database still works, that'd be incredible.

Ntemis, 2022-01-17:
From what package we take the rdb files from?

RobLoach, 2022-01-17:
libretro-database over at https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/libretro/libretro-database/package.mk

gouchi, 2022-01-17:
I don't reproduce the issue on Generic x86_64 nightly-20220116-8bb17e6.img.gz

RobLoach, 2022-01-17:
Good to hear! Thanks for the testing :+1:

closed 2022-01-17 by RobLoach