libretro/Lakka-LibreELEC #1589: SwanStation Core: per game config overrides do not load

auxfam, 2022-01-21:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
nightlies: https://nightly.builds.lakka.tv/2022-01-16/Generic.x86_64/Lakka-Generic.x86_64-nightly-20220116-8bb17e6.img.gz

- What system hardware are you using?
intel NUC8 Intel® Core™ i5-8259U Processor (6M Cache, up to 3.80 GHz)
note: this has 128 EDRAM as a L4 cpu/GPU cache

- What did you do?
1. load title (ridge racer 1 jp) with core: Swanstation
2. change retroarch video-> scaling -> aspect ratio to 16:9
3. save per game override

- What did you expect to happen?
I expect that if I close the title and then load again that the per game config is loaded and I see `16:9` as the saved asepct ratio

- What happened instead?
it used the saved Core aspect ratio instead (4:3 in my case)

note:  I repeated similar steps for another core (Dolphin) and pr game configs seem to work there.

Ntemis, 2022-01-21:
Core related issue, take it there.

closed 2022-01-21 by Ntemis

evbo, 2022-02-05:
@Ntemis I am noticing this same behavior with duckstation on lakka, but it appears to be archived on github. Where exactly should this issue be "taken" for the duckstation core?

Thanks!

Ntemis, 2022-02-06:
Interesting question indeed , maybe try duckstation discord for this one

Sanaki, 2022-02-06:
DuckStation has been archived, they severed ties with libretro a while back, and attempting to contact them over this will invariably lead to a ban from their server. The core itself is dead. As for SwanStation, @DarthMew is the most likely candidate to know something about what's going on.

Ntemis, 2022-02-06:
Then R.I.P DuckStation. I wont be accepting any bug issues after this

DarthMew, 2022-02-06:
I've tried it myself using a live USB of the latest version of Lakka, but on my end SwanStation saves and loads .cfg and .opt override files just fine.

DarthMew, 2022-02-10:
So you're not saving the core option overrides as either a game or content override through the "manage core options" submenu inside the core options tab?

evbo, 2022-02-10:
Correct, you can add "Overrides" to the quick menu and from there is where I'm clicking on saving a game override. To make it appear you have to enable it from user interface settings.

Sanaki, 2022-02-10:
Just to be clear, opt files aren't overrides, those are core options, saved at a core/directory/game level. Overrides are cfg files and can be saved at the same three levels.

evbo, 2022-02-12:
Ok, it just so happens it's the "core" options I want to override. So are you saying overriding core options is expected to not work? for instance, I want Duckstation (or Swanstation) to use a higher resolution for certain games that can run fast enough, but then switch to a lower resolution for games that can't.

DarthMew, 2022-02-12:
What you do is set the higher resolution (or any option you want to be different from the main .opt), go to "manage core options", and then select "save game/content/core options".

evbo, 2022-02-12:
Ok, strangely there is an "overrides" sub menu you can optionally add to the "Quick menu". I will try to do it the way you suggest and see if it's different.

evbo, 2022-02-12:
> You cannot save core options to an override

@Sanaki so it sounds like based on that there is no way to tell swanstation "use this resolution for this particular game and then use a lower resolution for that one"

What's confusing I think for the O.P. and me both is that other cores seem to allow overrides to core options on a per game basis, or maybe we're both confused with what qualifies as "core" options. For instance, O.P. references how in Dolphin core they are able to set per game options.

auxfam, 2022-02-14:
I am surprised that this got reopened but I think you are refering to something else.  if it;s per game resolution (or other actual core options like disc read speed etc...) then usually

1. load your game using swanstation core
2. go to quick menu -> options -> manage core options
3.  if no game  options saved then select "save game option"
4. this creates a per game template specifically for that title
5. now under quick menu -> options -> enhancement settings -> internal resolution scale,  set whatever you want 
6. close the core

this should do it