libretro/Lakka-LibreELEC #1601: Missing firmware for Realtek RTL8821CE Wireless Adapter

GPDP1, 2022-02-07:
I added a Wireless AC adapter + Bluetooth combo card to my Lakka x86 PC, and while Bluetooth works, it does not recognize WiFi capabilities of any kind. Wireless is confirmed working on both Manjaro and Batocera using the same machine. Log is provided below:

[log-2022-02-07-21.28.28.zip](https://github.com/libretro/Lakka-LibreELEC/files/8018930/log-2022-02-07-21.28.28.zip)

For good measure, here is the relevant output from lspci -k:

> 03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
> 	Subsystem: AzureWave RTL8821CE 802.11ac PCIe Wireless Network Adapter
> 	Kernel modules: wl, rtw88_8821ce

And from dmesg | grep firmware:

> [    2.724097] rtw_8821ce 0000:03:00.0: Direct firmware load for rtw88/rtw8821c_fw.bin failed with error -2
> [    2.724100] rtw_8821ce 0000:03:00.0: failed to request firmware
> [    2.725824] rtw_8821ce 0000:03:00.0: failed to load firmware

Seems the necessary firmware is not included.

i30817, 2022-02-20:
This wifi device has a lot of problems. IIRC, to prevent some crashing in hibernate i had to black list both 'kernel drivers' for it and use a out of tree one.

I put in this into the blacklist at `/etc/modprobe.d/blacklist.conf`
```
blacklist rtw88_8821ce
blacklist 8821ce
```

Then i built and installed this:
https://github.com/tomaspinho/rtl8821ce

It was annoying.

Ntemis, 2022-02-23:
Read this https://github.com/endlessm/linux-firmware/blob/master/rtw88/README

closed 2022-02-23 by Ntemis

GPDP1, 2022-03-05:
> Read this https://github.com/endlessm/linux-firmware/blob/master/rtw88/README

Thank you, adding this firmware to the appropriate folder fixed the issue.