libretro/Lakka-LibreELEC #1610: please add the emulation station UI to lakka

coreybruce, 2022-02-19:
Hi just wanted to ask if you could add the emulation station UI to lakka as it has so many UI options would would work so much better with Lakka and its features not to mention would be a Far better looking/running UI for handheld devices like the GPI cases and other hand held devices.

RobLoach, 2022-02-21:
The aim of Lakka is to be Just Enough OS for RetroArch. Emulation Station may be out of scope for that. I'd recommend trying out RetroPie or Batocera if you're looking to use EmulationStation as a launcher.

coreybruce, 2022-02-22:
> The aim of Lakka is to be Just Enough OS for RetroArch. Emulation Station may be out of scope for that. I'd recommend trying out RetroPie or Batocera if you're looking to use EmulationStation as a launcher.

It wouldn't be any different the the already existing UI choices on Lakka like xmb, rgui, ozone or fluid. All I am asking for is to add the emulation station UI to lakka :)

GavinDarkglider, 2022-02-24:
> > The aim of Lakka is to be Just Enough OS for RetroArch. Emulation Station may be out of scope for that. I'd recommend trying out RetroPie or Batocera if you're looking to use EmulationStation as a launcher.
> 
> It wouldn't be any different the the already existing UI choices on Lakka like xmb, rgui, ozone or glui. All I am asking for is to add the emulation station UI to lakka :)

Um, this is actually completely different. Retroarch is one program, Emulation Station is a different program. Both programs do the same thing essentially. But comparing  EmulationStation to xmb, rgui, ozone or glui which are basically skin engines in retroarch, is rediculous. Lakka is designed to be just enough OS to run retroarch, not retroarch and ...

coreybruce, 2022-02-25:
> Um, this is actually completely different. Retroarch is one program, Emulation Station is a different program. Both programs do the same thing essentially. But comparing EmulationStation to xmb, [...]

Would it be to much to even add it if it's different UI/launcher? it still would be just enough of a OS just another gui option.

RobLoach, 2022-02-25:
Yes...
- Allow configuration of which application to launch on boot
- New documentation for how to configure it
- Adding all the packages to build it properly
- Maintaining the upgrades for the packages
- Setting up the launch configuration for RetroArch across all platforms, along with which core is used
- Manage support requests surrounding emulation station
- etc

Again, if you're looking to use EmulationStation as a Frontend Launcher rather than RetroArch out of the box, I'd recommend looking at Recalbox, Batocera, or Retropie. There are a few others out there too... The other thing you could try is making your own fork of Lakka to try it out yourself!

coreybruce, 2022-02-28:
Understandable, thanks for the chat.

closed 2022-02-28 by coreybruce

ShadowrunHerzogZwei, 2023-12-21:
I support. I thought about this too and googled this topic. This should have been done a long time ago. Why do you need two shells, one inside the other? It would be much more convenient if we removed all menu items from EmulationStation and replaced them with RetroArch. And then hide all unnecessary menu items in RetroArch. Plus make a console installer for Raspberry and Radxa and PC x86.