libretro/Lakka-LibreELEC #1622: Lakka 3.7 RPi4 64 Bits => GameMode unsupported

MopheusDG, 2022-03-02:
Lakka stays on Logo and running verbose mode shows:

```
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroar                    ch.cfg".
[WARN] [GameMode]: GameMode cannot be enabled on this system ("dlopen faile                    d - libgamemode.so: cannot open shared object file: No such file or directo                    ry.") https://github.com/FeralInteractive/gamemode needs to be installed.
[WARN] [Config]: GameMode unsupported - disabling...
[INFO] RetroArch 1.10.0 (Git b71be7efd2)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Feb 28 2022
[INFO] Version: 1.10.0
[INFO] Git: b71be7efd2
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "V3D 4.2".
[INFO] [Vulkan]: Using GPU index 0.
[ERROR] [Vulkan]: Failed to create KHR_display surface.
[ERROR] [Vulkan]: Failed to set video mode.
[ERROR] [Video]: Cannot open threaded video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds                    .
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 0, Frames dropped: 0.

```

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 3.7 Stable
- What system hardware are you using? RPi4 4 Gb 64 Bits
- What did you do? Plain install, was working, updated to 3.7 and stuck on logo now.
- What did you expect to happen? Boot to menu.
- What happened instead? Stuck on logo.

MopheusDG, 2022-03-03:
Fixed the gamemode error adding the line to the config, but still having problems with Vulkan it seems.

Line to fix:

**gamemode_enable = "false"**

ToKe79, 2022-03-03:
is gamemode on by default? I did not experience this issue on rpi400. vulkan on rpi does not work with all cores.

MopheusDG, 2022-03-04:
I didn't have the option on the retroarch.cfg, I added but still have issues, so I decided to fully remove it and let it create again. Now it's working fine.

closed 2022-03-04 by ToKe79