libretro/Lakka-LibreELEC #1626: Anbernic RG351 Update bug

Cibomatto2002, 2022-03-10:
- Which version of Lakka are you using? 3.7
- What system hardware are you using? Anbernic RG351
- What did you do? try to update it after it downlodads the update it never updates
- What did you expect to happen? get updated
- What happened instead? nothing

this is been going on since Lakka 3.5

i have to reflash when I want the newest Lakka

ToKe79, 2022-03-14:
Can you be more specific? Did you reboot after download? Were there some messages on the screen during the next boot? The update is applied during booting (the init script checks if there is an update available in /storage/.update).

Cibomatto2002, 2022-03-14:
I rebooted it there was no messages of any kind it just shows lakka 3.6 I updated to 3.7 but I had to do a fresh install.

ToKe79, 2022-04-01:
refers to discontinued branch, only updates to RetroArch / libretro-cores.

Updating via RetroArch requires lot of RAM, as the image/update file is downloaded to RAM and then written to storage. These RK3326 devices have "only" 1GB of RAM, image/update file is more than 600MB. As a workaround place the update file directly into the /storage/.update folder:

- use SCP/SFTP (requires WLAN), or
- use the SD card in SD card reader on a computer with OS that supports ext4fs and put the update file into `.update` folder on the `LAKKA_DISK` partition, or
- use SSH (requires WLAN) and run `lakka-update.sh`.

closed 2022-04-01 by ToKe79