libretro/Lakka-LibreELEC #1632: [Request] - Add SameDuck core (Mega Duck/Cougar Boy)

Levine91, 2022-03-15:
Hello, I'm just wanting to request the addition of the SameDuck core to Lakka, as it would make things far smoother for the very few of us who would love to play Mega Duck/Cougar Boy without going through MAME. Thanks!

gouchi, 2022-03-15:
It seems there is a [branch SameDuck](https://github.com/LIJI32/SameBoy/tree/SameDuck).

Also @ToKe79 we might need to switch to [buildbot tree](https://github.com/libretro/SameBoy/tree/buildbot) from the Libretro repo as it is the default one. 

I don't know the difference between the SameDuck libretro core and the upstream one ?

ToKe79, 2022-03-15:
I think that builds SameBoy, but the [upstream branch](https://github.com/LIJI32/SameBoy/tree/SameDuck) seems to be the right spot. I will push PR, package is ready.

ToKe79, 2022-03-15:
@Levine91 can you test image from [here](http://nightly.builds.lakka.tv/PRs/pr_1633_000476/), if the core works as you expect?

PR #1633 (2022-03-15): sameduck: add new libretro core
closes #1632
commit c8540160 (2022-03-15): sameduck: add new libretro core (#1633)

closes #1632
commit 1927d3fa (2022-04-08): sameduck: add new libretro core (#1633)

closes #1632