libretro/Lakka-LibreELEC #1634: usbhid.quirks and Twin Usb Joystick

denser, 2022-03-15:
While use LAKKA 2.2 and Twin USB Joystick - device act as two-player  and in Input settings i can choose #1 and #2.

After reinstall LAKKA 3.6 in Input settings i can see only one Twin USB Joystick.

Digging lead me to 
https://github.com/libretro/RetroArch/issues/3084
and i try to use some cheat :) add this in syslinux.cfg: 
`usbhid.quirks=0x2563:0x0555:0x040`
and reboot. Now I can see both #1 and #2.

Pls explain to me is it issue or what?

gouchi, 2022-03-15:
Hi,

Some devices require to add/modify some "quirk" because they have some strange behavior otherwise.

Thank you.

Kind regards,

closed 2022-03-15 by gouchi

denser, 2022-06-26:
If there any way to make it in Settings? I mean to do not edit boot files.