libretro/Lakka-LibreELEC #1637: Black screen when booting Lakka USB

RedCoder09, 2022-03-16:
I'm trying to install Lakka on an old Dell Inspiron PC, but when I boot live or installer, the screen just goes black. No berry, no nothing. It has an Intel Pentium inside, and it seems that it's using the wrong video output. I have Windows 10 on it, and have tried the stable and nightly builds. Any help appreciated!

ToKe79, 2022-03-17:
Hi @RedCoder09,
you mentioned, that you suspect that it is outputting to to wrong video output - can you be more specific on your hardware, mostly GPU(s)? Can you try the other video outpu?
Thanks!

RedCoder09, 2022-03-17:
I have an ATI Mobility Radeon HD 4300 Series GPU, and I unfortunately don't have the cable needed for connecting to another display.

gouchi, 2022-03-17:
You may try to [ssh to the station](https://www.lakka.tv/doc/Finding-the-IP-of-your-Lakka-box/) if you plugged Ethernet cable.

RedCoder09, 2022-03-19:
It's not showing up either.

gouchi, 2022-03-19:
Strange try to add `debugging` so boot with `live ssh textmode debugging`

Thank you.

RedCoder09, 2022-03-19:
Nope, still not working.

gouchi, 2022-03-20:
You should try to get another cable or adapter of the the other output. You tried DVI output but you don't have VGA cable ?

RedCoder09, 2022-03-20:
No, I can't try either.

gouchi, 2022-03-20:
As you can't use other cable You may try to force video put using something like depending of your cable 
Force DVI output and disable VGA 
`video=DVI-I-1:1024x768@85 video=VGA-1:d`

Or force VGA output and disable DVI
`video=VGA-1:1024x768@85 video=DVI-I-1:d`

[Source](https://wiki.archlinux.org/title/kernel_mode_setting#Forcing_modes)

gouchi, 2022-04-07:
@RedCoder09 Did you have the opportunity to make some test ?

Ntemis, 2022-06-10:
Unsupported gpu

RedCoder09, 2022-06-11:
I believe it was working fine before, when I tried to reinstall Lakka is when the issues started. As such, I have reason to believe that the GPU is supported.

Ntemis, 2022-06-11:
Try this one but with a clean image https://nightly.builds.lakka.tv/2022-06-05/Generic.x86_64/

Ntemis, 2022-06-16:
No feedback closing down permanently

closed 2022-06-16 by Ntemis

RedCoder09, 2022-10-11:
Unfortunately I did not.