libretro/Lakka-LibreELEC #1640: IMX6.arm does not boot

kr43xpo, 2022-03-20:
I'm trying to install Lakka on an iMX6.arm (IMX6QUAD udoo). 
I have tried several versions, but all with the same result.
Only a very old version works for me (2.3)

With the latest version (Lakka-iMX6.arm-nightly-v4.x-20220318), activating the logs at boot, I see the following:

[image]

```
imx_usb 2184000.usb: No over current polarity defined
mxs_phy 20c9000.usbphy: Data pin can't make good contact.
imx_usb 2184200.usb: No over current polarity defined 
```

I am using this extlinux.conf file:
```
LABEL Lakka
  LINUX /KERNEL
  FDTDIR /
  APPEND boot=UUID=1803-3837 disk=UUID=1abbd6bc-1ff7-4f72-9418-8f09e3224853 console=ttymxc0,115200 console=tty0
```

ToKe79, 2022-03-20:
Sorry to hear that, but unfortunately none of the active contributors has access to this board. However, as we use LibreELEC as our foundation, you may check if LibreELEC nightly builds work and if these work, support in future is possible - https://test.libreelec.tv/

kr43xpo, 2022-03-20:
There is a nightly build version (LibreELEC-iMX6.arm-11.0-nightly-20220319-5591832-udoo.img), but I get the same error.
I will test in the future with new versions (if any).
Thanks.

ToKe79, 2022-03-20:
OK, so I'm closing this issue then, as we cannot fix it currently. I do not think anything will change until there is somebody with a fix in the upstream.

closed 2022-03-20 by ToKe79

MrCapone, 2022-10-24:
A little bump for this issue. I edited some package.mk file a little and changed kernel to 5.4 mainline for imx6. After that i compiled image and was able to boot on my udoo quad. Maybe edit imx6 target config to use kernel 5.4 may be temporary solution?