libretro/Lakka-LibreELEC #1646: No Dolphin Standalone Core

michaelsmolinski, 2022-03-20:
Dolphin downloads, but doesn’t appear on ‘Standalone Cores’ menu.

Running Lakka 4.0 release on Odroid N2+

ToKe79, 2022-03-23:
I do not think Lakka supports standalone dolphin emulator. Did you download the file from https://buildbot.libretro.com/assets/system/Dolphin.zip? If so, these are only assets for the dolphin core, which are already included in the image and there is no need to download them.

ToKe79, 2022-03-24:
I discussed this with the RetroArch devs - the archives in the [`system` folder on buildbot](https://buildbot.libretro.com/assets/system/) are supplementary files to various cores, some of which are "standalone" cores, but require additional data (game files) to run. The Dolphin archive contains files for the libretro dolphin core. However, these files are already included in the Lakka image. It is strongly advised to Lakka users to not download files from this folder, as they may be incompatible with the core-version provided in the Lakka image, especially in the case of the `PPSSPP` core. Starting from next Lakka release we will include (and already include in the [nightly builds](https://nightly.builds.lakka.tv/latest/)) files needed for `nxengine`, `prdoom`, `ecwolf`, `cannonball`, `dinothawr`. We already include files for `scummvm`, `PPSSPP`, `dolphin`, `mame2003-plus`, `fbneo` and `uae4arm`.

Same applies to other files available on the libretro buildbot like database files, cheats, overlays, shaders, etc. The Lakka image already includes all these files, so it can work also without internet connection.

closed 2022-03-24 by ToKe79

michaelsmolinski, 2022-03-24:
Both, thank you for the quick turnaround.

@Mr-Jake Indeed. I was under the mistaken impression that the "standalone cores" menu was for standalone emulators (i.e. not using libretro). I was trying to see if any noticeable performance and compatibility differences between dolphin standalone emulator vs. dolphin libretro core.

@ToKe79  Thank you for those details. Just for clarification, I downloaded dolphin.zip from within Lakka, which may or may not be the same as your link above.

Agreed to close issue. Thanks again!