libretro/Lakka-LibreELEC #1651: PCSX ReARMed Crashing (Lakka 4.0 RPi4)

mediamoshpit, 2022-03-22:
- Which version of Lakka are you using?
4.0
- What system hardware are you using?
RPi4b 4GB
- What did you do?
Launch Games
- What did you expect to happen?
Games to run in stable manner
- What happened instead?
Games run for various lengths of time before crashing.

Below is a log of 'Wipeout 3: Special Edition' running and crashing. I thought it might be a good example as it always crashes early, before gameplay can even be reached.

`[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Mar 13 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] Cannot resolve save file path.
[INFO] Cannot resolve save state file path.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [SoftFilter]: Found plug: Normal2x Height (normal2x_height).
[INFO] [SoftFilter]: Found plug: Blargg NTSC SNES (blargg_ntsc_snes).
[INFO] [SoftFilter]: Found plug: SuperEagle (supereagle).
[INFO] [SoftFilter]: Found plug: Upscale_240x160-320x240 (upscale_240x160_320x240).
[INFO] [SoftFilter]: Found plug: Normal2x Width (normal2x_width).
[INFO] [SoftFilter]: Found plug: LQ2x (lq2x).
[INFO] [SoftFilter]: Found plug: Super2xSaI (super2xsai).
[INFO] [SoftFilter]: Found plug: Normal4x (normal4x).
[INFO] [SoftFilter]: Found plug: Darken (darken).
[INFO] [SoftFilter]: Found plug: Dot Matrix 4x (dot_matrix_4x).
[INFO] [SoftFilter]: Found plug: EPX (epx).
[INFO] [SoftFilter]: Found plug: Picoscale_256x-320x240 (picoscale_256x_320x240).
[INFO] [SoftFilter]: Found plug: Upscale1.5x (upscale_1_5x).
[INFO] [SoftFilter]: Found plug: Upscale_256x-320x240 (upscale_256x_320x240).
[INFO] [SoftFilter]: Found plug: Dot Matrix 3x (dot_matrix_3x).
[INFO] [SoftFilter]: Found plug: Grid3x (grid3x).
[INFO] [SoftFilter]: Found plug: 2xSaI (2xsai).
[INFO] [SoftFilter]: Found plug: Normal2x (normal2x).
[INFO] [SoftFilter]: Found plug: Gameboy4x (gameboy4x).
[INFO] [SoftFilter]: Found plug: Gameboy3x (gameboy3x).
[INFO] [SoftFilter]: Found plug: Phosphor2x (phosphor2x).
[INFO] [SoftFilter]: Found plug: 2xBR (2xbr).
[INFO] [SoftFilter]: Found plug: Scale2x (scale2x).
[INFO] [SoftFilter]: Found plug: Scanline2x (scanline2x).
[INFO] [SoftFilter]: Found plug: Grid2x (grid2x).
[INFO] Using 1 threads for softfilter.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 38 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 23.976080 Hz
[... 327 lines cut ...]
[INFO] [RCHEEVOS]: Found AUDIO track 13 (first sector 0, sector size 2352, 150 pregap sectors)
[INFO] [RCHEEVOS]: Found AUDIO track 14 (first sector 0, sector size 2352, 150 pregap sectors)
[INFO] [RCHEEVOS]: Opened WipEout 3 - Special Edition (Europe) (En,Fr,De,Es,It) (Track 01).bin
[INFO] [RCHEEVOS]: Opened track 1 (sector size 2352)
[INFO] [RCHEEVOS]: Opened WipEout 3 - Special Edition (Europe) (En,Fr,De,Es,It).cue
[INFO] [RCHEEVOS]: Found MODE2/2352 track 1 (first sector 0, sector size 2352, 0 pregap sectors)
[INFO] [RCHEEVOS]: Found AUDIO track 2 (first sector 0, sector size 2352, 150 pregap sectors)
[INFO] [RCHEEVOS]: Opened WipEout 3 - Special Edition (Europe) (En,Fr,De,Es,It) (Track 02).bin
[INFO] [RCHEEVOS]: Opened track 2 (sector size 2352, 150 pregap sectors)
[INFO] [RCHEEVOS]: Not a PC-FX CD
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [Video]: Frame delay decrease by 1 to 7 due to frame time: 24999 > 16666.
[INFO] [Video]: Frame delay decrease by 1 to 6 due to frame time: 24999 > 16666.
[INFO] [Video]: Frame delay decrease by 1 to 5 due to frame time: 24999 > 16666.
[INFO] [Video]: Frame delay decrease by 1 to 4 due to frame time: 24999 > 16666.
[INFO] [Video]: Frame delay decrease by 1 to 3 due to frame time: 24999 > 16666.
[INFO] [Video]: Frame delay decrease by 1 to 2 due to frame time: 24999 > 16666.
[INFO] [Video]: Frame delay decrease by 1 to 1 due to frame time: 24999 > 16666.
[INFO] [RCHEEVOS]: pseudoscience logged in successfully
[INFO] [RCHEEVOS]: Game could not be identified
[INFO] [Video]: Frame delay decrease by 1 to 0 due to frame time: 24999 > 16666.
[INFO] [Environ]: SET_GEOMETRY: 256x240, Aspect: 1.333.
[INFO] [Environ]: SET_GEOMETRY: 640x448, Aspect: 1.333.
[INFO] [Environ]: SET_GEOMETRY: 320x240, Aspect: 1.333.`

ToKe79, 2022-03-25:
@mediamoshpit can you test with this image - RA 1.10.2 + updated cores https://nightly.builds.lakka.tv/PRs/pr_1661_000494/RPi4.aarch64/
Thank you!

mediamoshpit, 2022-03-26:
Still experiencing crashes in PCSX ReARMed. I can provide a log if necessary. Interestingly I'm also also noticing the graphical glitches that were seen in DOSBox Pure. Since that issue closed I have also updated my LibreElec install and have noticed similar glitches afterward there as well. While they're more infrequent and visually a little different, I now can't help but think the issue comes from upstream.

DrunkenMonk0001, 2022-03-30:
Disabling real bios allows games to load without crashing, however not using real bios disables too many other features (e.g consistent sound in games)

Using SCPH5002 with Digimon World 1 (EUR) as an example, It will seemingly work perfect at first, until the emulator freezes when using a bios and attempting to access the save/load menu and without the bios it functions fine, but has almost a complete loss of all sound.

DrunkenMonk0001, 2022-03-31:
> I can confirm this behavior on Generic 64-bit. The crash is not at a specific time/place in content, sometimes after item in main menu, sometimes able to play through 1 stage/level and then crashes during reading memory card... Seems to be related to dynarec. When I switch dynarec off, I did not encounter such crash during 15 minutes. Below are logs.
> 
> dynarec enabled:
> 
> ```
> [INFO] [Config]: Loading default config.
> [INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
> [INFO] RetroArch 1.10.2 (Git de4e56e)
> [INFO] === Build =======================================
> [INFO] CPU Model Name: Intel(R) Celeron(R) CPU 3215U @ 1.70GHz
> [INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2
> [INFO] Built: Mar 31 2022
> [INFO] Version: 1.10.2
> [INFO] Git: de4e56e
> [INFO] =================================================
> [INFO] [Input]: Found input driver: "udev".
> [INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
> [INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
> [INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
> [INFO] [Video]: Set video size to: fullscreen.
> [INFO] [Video]: Starting threaded video driver ...
> [INFO] [DRM]: Found 2 connectors.
> [INFO] [DRM]: Connector 0 connected: yes
> [INFO] [DRM]: Connector 0 has 1 modes.
> [INFO] [DRM]: Connector 0 assigned to monitor index: #1.
> [INFO] [DRM]: Connector 1 connected: no
> [INFO] [DRM]: Connector 1 has 0 modes.
> [INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.020424 Hz
> libGL: Can't open configuration file /etc/drirc: No such file or directory.
> libGL: Can't open configuration file /storage/.drirc: No such file or directory.
> libGL: Can't open configuration file /etc/drirc: No such file or directory.
> libGL: Can't open configuration file /storage/.drirc: No such file or directory.
> libGL: Can't open configuration file /etc/drirc: No such file or directory.
> libGL: Can't open configuration file /storage/.drirc: No such file or directory.
> libGL: Can't open configuration file /etc/drirc: No such file or directory.
> libGL: Can't open configuration file /storage/.drirc: No such file or directory.
> [INFO] [GL]: Found GL context: "kms".
> [INFO] [GL]: Detecting screen resolution: 1920x1080.
> [INFO] [EGL] Found EGL client version >= 1.5, trying eglGetPlatformDisplay
> [INFO] [EGL]: EGL version: 1.5
> [INFO] [EGL]: Current context: 0x7f0304049db0.
> [INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
> [INFO] [GL]: Vendor: Intel, Renderer: Mesa Intel(R) HD Graphics (BDW GT1).
> [INFO] [GL]: Version: 4.6 (Compatibility Profile) Mesa 22.0.1.
> [INFO] [GL]: Using resolution 1920x1080.
> [INFO] [GL]: Default shader backend found: glsl.
> [INFO] [Shader driver]: Using GLSL shader backend.
> [INFO] [GLSL]: Checking GLSL shader support ...
> [WARN] [GL]: Stock GLSL shaders will be used.
> [INFO] [GLSL]: Found GLSL vertex shader.
> [INFO] [GLSL]: Found GLSL fragment shader.
> [INFO] [GLSL]: Linking GLSL program.
> [INFO] [GLSL]: Found GLSL vertex shader.
> [INFO] [GLSL]: Found GLSL fragment shader.
> [INFO] [GLSL]: Linking GLSL program.
> [INFO] [GLSL]: Found GLSL vertex shader.
> [INFO] [GLSL]: Found GLSL fragment shader.
> [INFO] [GLSL]: Linking GLSL program.
> [INFO] Setting up menu pipeline shaders for XMB ...
> [INFO] [GLSL]: Compiling ribbon shader..
> [INFO] [GLSL]: Found GLSL vertex shader.
> [INFO] [GLSL]: Found GLSL fragment shader.
> [INFO] [GLSL]: Linking GLSL program.
> [INFO] [GLSL]: Compiling simple ribbon shader..
> [INFO] [GLSL]: Found GLSL vertex shader.
> [INFO] [GLSL]: Found GLSL fragment shader.
> [INFO] [GLSL]: Linking GLSL program.
> [INFO] [GLSL]: Compiling snow shader..
> [INFO] [GLSL]: Found GLSL vertex shader.
> [INFO] [GLSL]: Found GLSL fragment shader.
[... 1115 lines cut ...]
> [libretro INFO] port: 5  device: standard
> [libretro INFO] port: 6  device: none
> [libretro INFO] port: 7  device: none
> [libretro INFO] port: 8  device: none
> [INFO] [Playlist]: Loading history file: "/home/vudiq/.config/retroarch/content_history.lpl".
> [INFO] [Playlist]: Loading history file: "/home/vudiq/.config/retroarch/content_music_history.lpl".
> [INFO] [Playlist]: Loading history file: "/home/vudiq/.config/retroarch/content_video_history.lpl".
> [INFO] [Playlist]: Loading history file: "/home/vudiq/.config/retroarch/content_image_history.lpl".
> [INFO] [Playlist]: Loading favorites file: "/home/vudiq/.config/retroarch/content_favorites.lpl".
> [INFO] [Environ]: RETRO_ENVIRONMENT_SET_MINIMUM_AUDIO_LATENCY.
> [INFO] [Environ]: SET_GEOMETRY: 640x478, Aspect: 1.333.
> [INFO] [GLX]: Resized fullscreen resolution to 1920x1080.
> [INFO] [Environ]: SET_GEOMETRY: 512x256, Aspect: 1.333.
> [INFO] [GLX]: Resized fullscreen resolution to 1920x1080.
> Lightrec RAM usage: IR 265 KiB, CODE 1105 KiB, MIPS 146 KiB, TOTAL 6785 KiB, avg. IPI 7.543721
> Lightrec RAM usage: IR 294 KiB, CODE 1213 KiB, MIPS 161 KiB, TOTAL 6939 KiB, avg. IPI 7.495568
> Lightrec RAM usage: IR 302 KiB, CODE 1240 KiB, MIPS 165 KiB, TOTAL 6977 KiB, avg. IPI 7.492233
> Lightrec RAM usage: IR 308 KiB, CODE 1263 KiB, MIPS 168 KiB, TOTAL 7010 KiB, avg. IPI 7.482701
> Lightrec RAM usage: IR 331 KiB, CODE 1344 KiB, MIPS 178 KiB, TOTAL 7124 KiB, avg. IPI 7.522274
> Lightrec RAM usage: IR 350 KiB, CODE 1438 KiB, MIPS 191 KiB, TOTAL 7248 KiB, avg. IPI 7.515500
> Lightrec RAM usage: IR 353 KiB, CODE 1506 KiB, MIPS 199 KiB, TOTAL 7329 KiB, avg. IPI 7.534581
> Segmentation fault (core dumped)
> ```

I had noted the issue be repeatable specifically on the switch version, it would hang entirely when accessing the memory card menu. This was very consistent when testing, switching between utilizing the bios and using HLE #1671 This same issue was not repeatable using retroarch + pcsx rearmed on a windows x64 system. Issues may or may not be related, seems likely they are since it's memory card access related? 🤷

ToKe79, 2022-03-31:
I had such crashes also during normal gameplay / not accessing memory card (or not at least intentionally, not sure if the game was accessing it).
This probably needs bisecting - do you know date/commit hash of the core you used on windows, which did not show this behavior? With Lakka and Ubuntu I tested the most recent version as of today (https://github.com/libretro/pcsx_rearmed/commit/22b488f32195ac270935f2760d91501c8fa0275a)

ToKe79, 2022-03-31:
Since this is not a Lakka bug, I'm closing this issue, the bug is already reported in the core repo - https://github.com/libretro/pcsx_rearmed/issues/626

closed 2022-03-31 by ToKe79

notaz, 2022-04-05:
I can tell from the logs the core was not compiled correctly for ARM, DYNAREC=ari64 must be used.

As for x86, whomever added the dynarec for it is not maintaining it, so it should not be used (DYNAREC=0 or whetever).

ToKe79, 2022-04-06:
@notaz thank you for your comment - indeed, we build `arm` with `DYNAREC=ari64`, but for other archs (`aarch64`, `x86_64` and `i386`) we do not add `DYNAREC=...` to make options, so the default dynarec is probably built.

I see that now when `lightrec` is being compiled for e.g. `aarch64`, an error is thrown (no `DYNAREC=` in make options):

```
libpcsxcore/lightrec/plugin.c:19:2: error: #error "Lightrec should not be used on ARM (please specify DYNAREC=ari64 to make)"
   19 | #error "Lightrec should not be used on ARM (please specify DYNAREC=ari64 to make)"
      |  ^~~~~
```

After adding e.g. `DYNAREC=none` to make options for `aarch64`, seems like the lightrec is really excluded. Can we use `ari64` on `aarch64`? Should we avoid using `lightrec` (or any DYNAREC) on `x86_64` and `i386`?

ToKe79, 2022-04-06:
@notaz based on https://github.com/libretro/pcsx_rearmed/issues/640#issuecomment-1089365376 no dynarec for x86_64 / i386 it is.

ToKe79, 2022-04-06:
@mediamoshpit should be fixed in next nightly build (updated core + correct dynarec). you can either wait for next nightly build / release or manually copy core from https://nightly.builds.lakka.tv/members/vudiq/pcsx_rearmed/

mediamoshpit, 2022-04-06:
Tested it on several titles and everything seems to work well now. Gotta hand it to you guys and the overall project; I make use of 48 cores and during my migration from 32-bit, only two had issues. Pretty great work and much appreciated.

KOPRajs, 2023-04-17:
@ToKe79 Since the https://github.com/libretro/pcsx_rearmed/issues/626 was resolved, could the dynarec be re-enabled for x86? Or there are any other problems with the dynarec enabled?