libretro/Lakka-LibreELEC #1653: DOSBox Pure Failing to Launch (Lakka 4.0 RPi4)

mediamoshpit, 2022-03-22:
- Which version of Lakka are you using?
4.0
- What system hardware are you using?
RPi4b 4GB
- What did you do?
Launch Games
- What did you expect to happen?
Games to launch
- What happened instead?
Core exits immediately with `[ERROR] Error(s): /tmp/cores/dosbox_pure_libretro.so: cannot open shared object file: No such file or directory`

Below is the full log launching 'Archon'.

`[INFO] [Config]: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Playlist]: Written to playlist file: /storage/playlists/DOS.lpl
[ERROR] Failed to open libretro core: "/tmp/cores/dosbox_pure_libretro.so"
[ERROR] Error(s): /tmp/cores/dosbox_pure_libretro.so: cannot open shared object file: No such file or directory
[INFO] [Core]: Using content: "/storage/roms/PI_EXTERNAL/retropie-mount/roms/pc/archon/archon.bat".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 5623, Frames dropped: 0.
[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Mar 13 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/dosbox_pure_libretro.so"
[ERROR] Failed to open libretro core: "/tmp/cores/dosbox_pure_libretro.so"
Error(s): /tmp/cores/dosbox_pure_libretro.so: cannot open shared object file: No such file or directory
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core symbols..
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [SoftFilter]: Found plug: Normal2x Height (normal2x_height).
[INFO] [SoftFilter]: Found plug: Blargg NTSC SNES (blargg_ntsc_snes).
[INFO] [SoftFilter]: Found plug: SuperEagle (supereagle).
[INFO] [SoftFilter]: Found plug: Upscale_240x160-320x240 (upscale_240x160_320x240).
[INFO] [SoftFilter]: Found plug: Normal2x Width (normal2x_width).
[INFO] [SoftFilter]: Found plug: LQ2x (lq2x).
[INFO] [SoftFilter]: Found plug: Super2xSaI (super2xsai).
[INFO] [SoftFilter]: Found plug: Normal4x (normal4x).
[INFO] [SoftFilter]: Found plug: Darken (darken).
[INFO] [SoftFilter]: Found plug: Dot Matrix 4x (dot_matrix_4x).
[INFO] [SoftFilter]: Found plug: EPX (epx).
[INFO] [SoftFilter]: Found plug: Picoscale_256x-320x240 (picoscale_256x_320x240).
[INFO] [SoftFilter]: Found plug: Upscale1.5x (upscale_1_5x).
[INFO] [SoftFilter]: Found plug: Upscale_256x-320x240 (upscale_256x_320x240).
[INFO] [SoftFilter]: Found plug: Dot Matrix 3x (dot_matrix_3x).
[INFO] [SoftFilter]: Found plug: Grid3x (grid3x).
[INFO] [SoftFilter]: Found plug: 2xSaI (2xsai).
[INFO] [SoftFilter]: Found plug: Normal2x (normal2x).
[INFO] [SoftFilter]: Found plug: Gameboy4x (gameboy4x).
[INFO] [SoftFilter]: Found plug: Gameboy3x (gameboy3x).
[INFO] [SoftFilter]: Found plug: Phosphor2x (phosphor2x).
[INFO] [SoftFilter]: Found plug: 2xBR (2xbr).
[INFO] [SoftFilter]: Found plug: Scale2x (scale2x).
[INFO] [SoftFilter]: Found plug: Scanline2x (scanline2x).
[INFO] [SoftFilter]: Found plug: Grid2x (grid2x).
[INFO] Using 1 threads for softfilter.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[... 74 lines cut ...]
[INFO] [udev]: Keyboard #5: "  Wireless Air mouse Keyboard Consumer Control" (/dev/input/event8).
[INFO] [udev]: Keyboard #6: "  Wireless Air mouse Keyboard System Control" (/dev/input/event9).
[INFO] [udev]: Mouse #0: "Compx 2.4G Wireless Receiver Mouse" (REL) /dev/input/event1
[INFO] [udev]: Mouse #1: "  Wireless Air mouse Keyboard Mouse" (REL) /dev/input/event7
[INFO] [udev]: Pad #0 (/dev/input/event10) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event10) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Display]: Found display driver: "gl".
[ERROR] Failed to open libretro core: "/tmp/cores/dosbox_pure_libretro.so"
[ERROR] Error(s): /tmp/cores/dosbox_pure_libretro.so: cannot open shared object file: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed.
[INFO] [SRAM]: SRAM will not be saved.
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).`

ToKe79, 2022-03-22:
I just checked the core in the image:
```
$ pwd
/home/lakkabuildbot/release/Lakka-LibreELEC/build.Lakka-RPi4.aarch64/image/system/usr/lib/libretro
$ file dosbox_pure_libretro.so 
dosbox_pure_libretro.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=c87415c7a624138e2a67b1b2fba2eb18854e766d, stripped
```

For some reason it is built as x86_64 binary, not aarch64. we will look into it, thank you for your report.

ToKe79, 2022-03-22:
You can test with [this](https://nightly.builds.lakka.tv/members/vudiq/dosbox_pure_libretro.so) core:
- scp to `/tmp/cores` OR
- copy via samba to `Cores` share OR
- copy to the second partition on the SD card to folder `cores`

Thank you.

mediamoshpit, 2022-03-22:
The core will now load without content, but still fails to launch with anything. I tried a few variations with different exe and bat files, all with the same result. Below is a log with a bat file launching 'Archon Ultra'

Thank you for your time.

`[INFO] [Config]: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Content]: Updating firmware status for: /tmp/cores/dosbox_pure_libretro.so on /storage/roms/PI_EXTERNAL/retropie-mount/BIOS
[INFO] [Core]: Using content: "/storage/roms/PI_EXTERNAL/retropie-mount/roms/pc/archon/archon.bat".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 7863, Frames dropped: 0.
[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Mar 13 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/dosbox_pure_libretro.so"
[INFO] [Overrides]: Core-specific overrides found at "/storage/.config/retroarch/config/DOSBox-pure/DOSBox-pure.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/DOSBox-pure/archon.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/DOSBox-pure/archon.cfg".
[INFO] [Config]: Appending config "/storage/.config/retroarch/config/DOSBox-pure/DOSBox-pure.cfg".
[INFO] [Environ]: SET_SUPPORT_NO_GAME: yes.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Remaps]: Core-specific remap found at "/storage/remappings/DOSBox-pure/DOSBox-pure.rmp".
[INFO] [Overrides]: Redirecting save file to "/storage/roms/PI_EXTERNAL/retropie-mount/roms/pc/archon/archon.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/PI_EXTERNAL/retropie-mount/roms/pc/archon/archon.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: SET_KEYBOARD_CALLBACK.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/PI_EXTERNAL/retropie-mount/BIOS".
[INFO] [Environ]: GET_VFS_INTERFACE. Core requested version >= V3, providing V3.`

ToKe79, 2022-03-22:
I do not see any error in the log - did it just hang?
anyway - [here](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-dosbox_pure/v2-compiler+no_optimize/dosbox_pure_libretro.so) is another build of the core for RPi4.aarch64 - without optimizations - we use gcc10 and the Makefile states that there are some issues with gcc10 + -O2, but it also mentions arm7l, so not sure if aarch64 is affected or not, and if disabling optimization can help.

ToKe79, 2022-03-22:
I also built two versions from the upstream repository https://github.com/schellingb/dosbox-pure:
- [forced toolchain compiler](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-dosbox_pure/v3-upstream-compiler_fix/dosbox_pure_libretro.so)
- [forced toolchain compiler + no optimizations](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-dosbox_pure/v4-upstream-compiler_and_optimization/dosbox_pure_libretro.so)

ToKe79, 2022-03-23:
added another build attempts [here](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-dosbox_pure/) - `v5` through `v8`. if you can also test these, if any of those works for you. thanks a lot!

mediamoshpit, 2022-03-23:
Not a problem. The previous builds all produced a crash. I'll test this new batch before long.

mediamoshpit, 2022-03-23:
Well, I hate to report that all eight were a no go as well. Each one I tried with multiple exe files. Every time it results in Lakka/RA itself crashing rather than just the core.

ToKe79, 2022-03-23:
Thank you for your time.

So it seems that it is related to the compiler then. I will merge at least my changes to the package, so the taget-toolchain-g++ is used (to build binary for the target system).

ToKe79, 2022-03-23:
I just noticed that the CPUFLAGS were not added (hat to modify the Makefile to see the actual build commands to notice this). can you try the [9th iteration](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-dosbox_pure/v9-added_cpuflags/dosbox_pure_libretro.so)? :-D

mediamoshpit, 2022-03-23:
Content is now launching in a stable manner, so that is very positive. However, every couple of minutes the screen shows unusual graphical glitches. More disturbingly, I notice a performance hit in the neighborhood of 60-70 percent. I saw the same drop in performance when testing DOSBox/DBSVN, but I chalked that up to development regression. As it stands it would seem all three DOS cores are experiencing a dramatic drop in performance across the board.

ToKe79, 2022-03-23:
that is positive news... what I did also was that I changed `-O2` to `-O3`, [here](https://nightly.builds.lakka.tv/members/vudiq/RPi4.aarch64-dosbox_pure/v10-rollback-O3/dosbox_pure_libretro.so) is this change reverted, if you are still up for testing.

mediamoshpit, 2022-03-23:
OK, The new build fails to launch. On the bright side, I found that the severe performance issues from the previous build stemmed from a setting I had left in place to test a particular feature between builds. Sorry for the confusion there. I guess this means the only remaining issues are graphical glitches. I'll keep testing through my library to see what I notice.

ToKe79, 2022-03-23:
ok, so for now i'll go to the 9th iteration and lets hope the glitches will be solved by a core update. thank you for your patience :)

ToKe79, 2022-03-23:
for the graphical glitches - if you can confirm that it is only Lakka related (it's not a core issue), please reopen this issue. thanks!

closed 2022-03-23 by ToKe79

mediamoshpit, 2022-03-23:
Sure. Just to leave a few observations for anyone in passing. My perspective in this is not only in moving from Lakka 3.7 to 4.0, but also migrating from a 32 to 64-bit installation for whatever that's worth. Also, it would seem that these graphical issues are only found when running certain types of games that appear to be terminal-driven graphics. 

Lastly, these glitches can be alleviated and perhaps eliminated by adjusting the performance core option to a fixed and or lower setting. This was not necessary for the 32-bit implementation, but with my very limited testing seems to work well enough as long as the software can operate at the new setting. Thanks again ToKe79!

PR #1657 (2022-03-23): dosbox_pure: use correct g++
closes: #1653