libretro/Lakka-LibreELEC #1656: SwanStation Failing to Launch (Lakka 4.0 RPi4)

mediamoshpit, 2022-03-22:
I had previously opened a virtually identical issue for DuckStation, after reading somewhere that development had resumed there. At any rate they both appear to crash under the same circumstances in Lakka 4.0.

- Which version of Lakka are you using?
4.0
- What system hardware are you using?
RPi4b 4GB
- What did you do?
Launch games
- What did you expect to happen?
Games to launch
- What happened instead?
Core crashes immediately 

Below is a log of 'Alien Trilogy' attempting to Launch.

`[INFO] [Config]: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Core]: Using content: "/storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).cue".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 1661, Frames dropped: 0.
[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Mar 13 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/swanstation_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/SwanStation/SwanStation.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/SwanStation/Alien Trilogy (USA).cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/SwanStation/Alien Trilogy (USA).cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2_INTL.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: Redirecting save file to "/storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).state".
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Content]: Content loading skipped. Implementation will load it on its own.
[libretro INFO] [Main] retro_load_game(/storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).cue)
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[libretro INFO] [HostInterface] Boot Filename: /storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).cue
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[libretro INFO] [HostInterface] Creating 'Cubeb' audio stream, sample rate = 44100, channels = 2, buffer size = 2048
[libretro INFO] [System] Loading CD image '/storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).cue'...
[libretro INFO] [System] Auto-detected console NTSC-U region for '/storage/roms/PI_EXTERNAL/retropie-mount/roms/psx/Alien Trilogy (USA)/Alien Trilogy (USA).cue' (region NTSC-U)
[libretro INFO] [System] Console Region: NTSC-U/C (US, Canada)
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/PI_EXTERNAL/retropie-mount/BIOS".
[libretro INFO] [LibretroHostInterface] Running game changed: SLUS-00007 (Alien Trilogy (USA))
[libretro INFO] [Common::MemoryArena] File mapping name: duckstation_994_0x7f78a75888
[libretro INFO] [Bus] RAM is 2097152 bytes at 0x7f91bb6000
[libretro INFO] [GPUBackend] GPU thread started.
[libretro INFO] [Bus] Fastmem base (software): 0x7f89dc9010
[libretro INFO] [CDROMAsyncReader] Read thread started with readahead of 8 sectors
[libretro INFO] [CDROM] Inserting new media, disc region: NTSC-U, console region: NTSC-U
[INFO] [Environ]: GET_PREFERRED_HW_RENDER, video driver name: gl.
[... 101 lines cut ...]
[INFO] [DRM]: Mode 33: (640x480) 640 x 480, 75.000000 Hz
[INFO] [DRM]: Mode 34: (640x480) 640 x 480, 72.808800 Hz
[INFO] [DRM]: Mode 35: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 36: (640x480) 640 x 480, 59.940475 Hz
[INFO] [DRM]: Mode 37: (720x400) 720 x 400, 70.081665 Hz
[INFO] [GL]: Found GL context: "kms".
[INFO] [GL]: Detecting screen resolution: 1920x1080.
[INFO] [EGL] Found EGL client version >= 1.5, trying eglGetPlatformDisplay
[INFO] [EGL]: EGL version: 1.4
[ERROR] [EGL]: #0x3009, EGL_BAD_MATCH
[ERROR] [Video]: Cannot open video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [RCHEEVOS]: Asked the load tasks to terminate
[INFO] [RCHEEVOS]: Error resolving hash: Load aborted
[INFO] [RCHEEVOS]: Error logging in: Load aborted
[libretro INFO] [LibretroHostInterface] Hardware context destroyed
[INFO] [Core]: Unloading game..
[libretro INFO] [GPUBackend] GPU thread stopped.
[libretro INFO] [LibretroHostInterface] Running game changed:  ()
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Core]: Saved core options file to "/storage/.config/retroarch/config/SwanStation/SwanStation.opt".
[ERROR] failed_to_start_audio_driver
[INFO] [Video]: Frame delay reset to 8.`

ToKe79, 2022-03-22:
can you test with `vulkan` driver selected in Settings &rarr; Drivers &rarr; Video?

mediamoshpit, 2022-03-23:
It appears to crash with Vulkan as well.

ToKe79, 2022-03-23:
according to report on discord:

> jdalmanza - 03/20/2022
FYI for those trying to use swanstation on 4.0, you need to manually edit the SwanStation.opt file and set the GPU.Rendered="Vulkan" manually or it will crash

Can you try this? I think the file should be somewhere in `/storage/.config/retroarch`, while `/storage/.config` path is also accessible via samba share `Configfiles`.

mediamoshpit, 2022-03-23:
Manually editing the .opt file did the trick.

ToKe79, 2022-04-01:
does not happen on nightly build 2022.04.01

closed 2022-04-01 by ToKe79