libretro/Lakka-LibreELEC #1659: Freeze on lakka logo

SajjadSabzkar, 2022-03-23:
i bootable my sd card with etcher and plug sd card in my device boot successfully.  first step resize partition and reboot after 15 second is ok but after reboot device lock on lakka image and no happen anything.

- Which version of Lakka are you using? 4.0
- What system hardware are you using? x96mini 2g ram / 16g rom
- What did you do?  truning on device
- What did you expect to happen? boot to main menu lakka
- What happened instead? freeze at lakka logo

ToKe79, 2022-03-23:
Hi, can you gather some logs by following [this guide](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka#troubleshooting-stuck-on-lakka-flower-logo) - as you have Amlogic device, you will have to edit `uEnv.ini` in the root of the SD card.

SajjadSabzkar, 2022-03-23:
> Hi, can you gather some logs by following [this guide](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka#troubleshooting-stuck-on-lakka-flower-logo) - as you have Amlogic device, you will have to edit `uEnv.ini` in the root of the SD card.

Ok i'll check it

cesartaqui, 2022-03-23:
Same insue here. Lakka pc 64bits USB image.

ToKe79, 2022-03-23:
> Same insue here. Lakka 4.0 pc 64bits USB image.

If you have Nvidia GPU, we have a bug, you can try image from here - http://nightly.builds.lakka.tv/PRs/pr_1655_000488/

cesartaqui, 2022-03-23:
> > Same insue here. Lakka 4.0 pc 64bits USB image.
> 
> If you have Nvidia GPU, we have a bug, you can try image from here - http://nightly.builds.lakka.tv/PRs/pr_1655_000488/

Yes! I have Nvidia GPU. Thanks!

SajjadSabzkar, 2022-03-24:
> Hi, can you gather some logs by following [this guide](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka#troubleshooting-stuck-on-lakka-flower-logo) - as you have Amlogic device, you will have to edit `uEnv.ini` in the root of the SD card.

i followed  the guide and now i have shell / terminal for use . which command to start lakka main menu ?

SajjadSabzkar, 2022-03-24:
i run command retroarch -v and save response to log.txt 
i also test Lakka-S905.arm-2.3.1.img.gz  version of lakka and everything is ok but in latest image this problem happen .
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/8338710/log.txt)

ToKe79, 2022-03-24:
can you also give me output of `ldd /usr/bin/retroarch` and also `LIBGL_DEBUG=verbose retroarch -v`. Thank you!

ToKe79, 2022-03-24:
okay, I think we would need full logs - after you get to the shell prompt and execute `systemctl start retroarch.target` (there will be no output to this command, only back to shell prompt), please run `createlog` to collect all logs. As you will have no network and probably do not have the possibilty to read the ext4 partition, select the option "2" - to store the archive with logs on the FAT32 partition. Then upload this file (will be named `log-<year>-<month>-<day>-<hour>-<minute>-<second>.zip`). Thank you!

ToKe79, 2022-04-03:
thank you - can you try image from here:
https://nightly.builds.lakka.tv/members/vudiq/AMLGX+mali_fbdev/

which DTB are you using?

ToKe79, 2022-04-07:
@SajjadSabzkar you can ignore the previous comment, `mali_fbdev` would not help, as we are using `lima` opensource driver. but I made a build where RetroArch is configured for GLESv2 (instead of GLESv3): https://nightly.builds.lakka.tv/members/vudiq/AMLGX.aarch64-GLESv2only/

Can you please test with this build and provide logs?

I noticed in the `01_RetroArch.log` missing verbose output from libGL.

Thank you!

ToKe79, 2022-04-07:
Additional question - is [Lakka 3.7](https://le-builds.lakka.tv/AMLGX.arm/Lakka-AMLGX.arm-3.7-box.img.gz) working on your TV Box?

SajjadSabzkar, 2022-04-09:
> Thank you!

Hi , Yes of Course , Sorry to see your Comment now

SajjadSabzkar, 2022-04-09:
> stion -

I'll test this version too

SajjadSabzkar, 2022-04-12:
> Additional question - is [Lakka 3.7](https://le-builds.lakka.tv/AMLGX.arm/Lakka-AMLGX.arm-3.7-box.img.gz) working on your TV Box?

I tested this version and this version is no work on my tvbox too

igyenes, 2022-04-13:
@SajjadSabzkar Hello, can you please give this a try? 
https://drive.google.com/file/d/1gSzDkveByanIleJQ9eeuPswG0iV4jxSi/view?usp=sharing

SajjadSabzkar, 2022-04-13:
> @SajjadSabzkar Hello, can you please give this a try? https://drive.google.com/file/d/1gSzDkveByanIleJQ9eeuPswG0iV4jxSi/view?usp=sharing

Yes Sure!

SajjadSabzkar, 2022-04-13:
> @SajjadSabzkar Hello, can you please give this a try? https://drive.google.com/file/d/1gSzDkveByanIleJQ9eeuPswG0iV4jxSi/view?usp=sharing

Wow This Version is ok but Why ? :)

igyenes, 2022-04-13:
Thanks for the feedback! 
Disabled the composite output, so it forces retroarch to start on the HDMI.

SajjadSabzkar, 2022-04-14:
> Thanks for the feedback! Disabled the composite output, so it forces retroarch to start on the HDMI.

Thanks .

PR #1684 (2022-04-16): Amlogic: disable composite output on tvboxes
Disable composite output to force retroarch to start on HDMI.
Fixes #1659
closed 2022-04-16 by Ntemis