libretro/Lakka-LibreELEC #1660: flycast only sound ,no screen (Lakka 3.7 & 4.0 Pi4 arm64)

atlus666, 2022-03-23:
Which version of Lakka are you using?
    3.7 & 4.0 (both have problem)
    What system hardware are you using?
    RPi4b 4GB
    What did you do?
    Launch Games
    What did you expect to happen?
    Games to launch
    What happened instead?
    the screen can't show game , just black screen ,but have game sound ( video driver "gl " )

gouchi, 2022-03-24:
Please try to provide some [logs](https://lakka.tv/doc/Troubleshooting-Lakka/) with `createlog` ?

Did you try with other video driver ?

Also if possible you can make some test with [nightly builds](https://nightly.builds.lakka.tv/latest/RPi4.aarch64/) ?

Thank you.

atlus666, 2022-03-25:
I tried "gl "  "glcore' no one works

atlus666, 2022-03-26:
hi , here is the log with lakka 4.0 use driver "gl"

[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] 容量:  ASIMD
[INFO] Built: Mar 14 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: libretro API 版本: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 31 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 6: (1680x1050) 1680 x 1050, 59.883251 Hz
[INFO] [DRM]: Mode 7: (1280x1024) 1280 x 1024, 75.024673 Hz
[INFO] [DRM]: Mode 8: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 9: (1440x900) 1440 x 900, 59.901459 Hz
[INFO] [DRM]: Mode 10: (1280x960) 1280 x 960, 60.000000 Hz
[INFO] [DRM]: Mode 11: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 14: (1024x768) 1024 x 768, 75.028580 Hz
[INFO] [DRM]: Mode 15: (1024x768) 1024 x 768, 70.069359 Hz
[INFO] [DRM]: Mode 16: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 17: (832x624) 832 x 624, 74.551270 Hz
[INFO] [DRM]: Mode 18: (800x600) 800 x 600, 75.000000 Hz
[INFO] [DRM]: Mode 19: (800x600) 800 x 600, 72.187569 Hz
[INFO] [DRM]: Mode 20: (800x600) 800 x 600, 60.316540 Hz
[INFO] [DRM]: Mode 21: (800x600) 800 x 600, 56.250000 Hz
[INFO] [DRM]: Mode 22: (720x576) 720 x 576, 50.000000 Hz
[INFO] [DRM]: Mode 23: (720x480) 720 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 24: (720x480) 720 x 480, 59.940060 Hz
[INFO] [DRM]: Mode 25: (640x480) 640 x 480, 75.000000 Hz
[INFO] [DRM]: Mode 26: (640x480) 640 x 480, 72.808800 Hz
[INFO] [DRM]: Mode 27: (640x480) 640 x 480, 66.666664 Hz
[INFO] [DRM]: Mode 28: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 29: (640x480) 640 x 480, 59.940475 Hz
[INFO] [DRM]: Mode 30: (720x400) 720 x 400, 70.081665 Hz
[INFO] [GL]: Found GL context: "kms".
[INFO] [GL]: Detecting screen resolution: 1920x1080.
[INFO] [EGL] Found EGL client version >= 1.5, trying eglGetPlatformDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7fa0006b20.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: Broadcom, Renderer: V3D 4.2.
[INFO] [GL]: Version: OpenGL ES 3.1 Mesa 22.0.0.
[INFO] [GL]: Using resolution 1920x1080.
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[... 206 lines cut ...]
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event* and /run/udev/?
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro INFO] core/rend/gles/gles.cpp:534 N[RENDERER]: Open GLES version 3.1
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is already enabled
[INFO] [Display]: Found display driver: "gl".
ALSA lib /home/lakkabuildbot/release/Lakka-LibreELEC/build.Lakka-RPi4.aarch64/build/alsa-lib-1.2.4/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed.
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Playlist]: Written to playlist file: /storage/.config/retroarch/content_history.lpl
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

atlus666, 2022-03-26:
hi ,here is lakka 4.0 with driver " glcore"  => no game screen output

[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] 容量:  ASIMD
[INFO] Built: Mar 14 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: libretro API 版本: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 31 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 6: (1680x1050) 1680 x 1050, 59.883251 Hz
[INFO] [DRM]: Mode 7: (1280x1024) 1280 x 1024, 75.024673 Hz
[INFO] [DRM]: Mode 8: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 9: (1440x900) 1440 x 900, 59.901459 Hz
[INFO] [DRM]: Mode 10: (1280x960) 1280 x 960, 60.000000 Hz
[INFO] [DRM]: Mode 11: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 14: (1024x768) 1024 x 768, 75.028580 Hz
[INFO] [DRM]: Mode 15: (1024x768) 1024 x 768, 70.069359 Hz
[INFO] [DRM]: Mode 16: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 17: (832x624) 832 x 624, 74.551270 Hz
[INFO] [DRM]: Mode 18: (800x600) 800 x 600, 75.000000 Hz
[INFO] [DRM]: Mode 19: (800x600) 800 x 600, 72.187569 Hz
[INFO] [DRM]: Mode 20: (800x600) 800 x 600, 60.316540 Hz
[INFO] [DRM]: Mode 21: (800x600) 800 x 600, 56.250000 Hz
[INFO] [DRM]: Mode 22: (720x576) 720 x 576, 50.000000 Hz
[INFO] [DRM]: Mode 23: (720x480) 720 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 24: (720x480) 720 x 480, 59.940060 Hz
[INFO] [DRM]: Mode 25: (640x480) 640 x 480, 75.000000 Hz
[INFO] [DRM]: Mode 26: (640x480) 640 x 480, 72.808800 Hz
[INFO] [DRM]: Mode 27: (640x480) 640 x 480, 66.666664 Hz
[INFO] [DRM]: Mode 28: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 29: (640x480) 640 x 480, 59.940475 Hz
[INFO] [DRM]: Mode 30: (720x400) 720 x 400, 70.081665 Hz
[INFO] [GLCore]: Found GL context: "kms".
[INFO] [GLCore]: Detecting screen resolution: 1920x1080.
[INFO] [EGL] Found EGL client version >= 1.5, trying eglGetPlatformDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7fa4019220.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GLCore]: Vendor: Broadcom, Renderer: V3D 4.2.
[INFO] [GLCore]: Version: OpenGL ES 3.1 Mesa 22.0.0.
[INFO] [GLCore]: Using resolution 1920x1080.
[INFO] [GLCore]: Loading stock shader.
[INFO] [slang]: Building pass #0 (N/A)
[INFO] [GLCore]: Not using frame history.
[INFO] [GLCore]: Not using framebuffer feedback.
[INFO] [Font]: Using font rendering backend: "freetype".
[... 135 lines cut ...]
[INFO] [Font]: Using font rendering backend: "freetype".
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event* and /run/udev/?
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro INFO] core/rend/gles/gles.cpp:534 N[RENDERER]: Open GLES version 3.1
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is already enabled
[INFO] [Display]: Found display driver: "glcore".
ALSA lib /home/lakkabuildbot/release/Lakka-LibreELEC/build.Lakka-RPi4.aarch64/build/alsa-lib-1.2.4/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed.
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: 正在讀取歷史檔案: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

atlus666, 2022-03-26:
it may not rom type problem , I try another rom use gdi type (not cdi) ,it just crash
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/Flycast/Flycast.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/Flycast/[GDI] Sonic Adventure 2 (US).cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/Flycast/Sonic Adventure 2 v1.008 (2001)(Sega)(NTSC)(US)(M5)[!].cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "11".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2_INTL.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: 重新轉向 save file to "/storage/savefiles/Sonic Adventure 2 v1.008 (2001)(Sega)(NTSC)(US)(M5)[!].srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/Sonic Adventure 2 v1.008 (2001)(Sega)(NTSC)(US)(M5)[!].state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[libretro INFO] shell/libretro/libretro.cpp:297 N[BOOT]: retro_init
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: SET_KEYBOARD_CALLBACK.
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Content]: 跳過內容載入。 實現將自行載入。
[libretro INFO] shell/libretro/libretro.cpp:1693 N[BOOT]: retro_load_game: /storage/roms/DreamCast/[GDI] Sonic Adventure 2 (US)/Sonic Adventure 2 v1.008 (2001)(Sega)(NTSC)(US)(M5)[!].gdi
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: SET_SAVE_STATE_IN_BACKGROUND: yes.
[libretro INFO] File extension is: .gdi
[INFO] [Environ]: GET_PREFERRED_HW_RENDER, video driver name: glcore.
[INFO] [Environ]: GET_PREFERRED_HW_RENDER - Context callback set to RETRO_HW_CONTEXT_OPENGL_CORE.
[INFO] [Environ]: SET_HW_RENDER, context type: gl.
[INFO] Requesting OpenGLES2 context.
[INFO] Reached end of SET_HW_RENDER.
[libretro ERROR] Verify Failed  : state == Init
 in init -> ../core/emulator.cpp : 387
[libretro ERROR] shell/libretro/libretro.cpp:2908 E[COMMON]: DEBUGBREAK!
Trace/breakpoint trap

atlus666, 2022-03-26:
I rollback to lakka 3.6 , the same sd card ,same rom , It work prefect 
here is log may help you to compare 

Lakka (official): 3.6 (RPi4.aarch64)
Lakka:~ # systemctl stop retroarch.service ; retroarch -vsystemctl stop retroarc
h.service ; retroarch -vsystemctl stop retroarch.service ; retroarch -v
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cf       g".
[INFO] RetroArch 1.9.13 (Git fca72f67d1)
[INFO] === Build =======================================
[INFO] 容量:  ASIMD
[INFO] Built: Nov 13 2021
[INFO] Version: 1.9.13
[INFO] Git: fca72f67d1
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] [Core]: Frontend is built for dynamic libretro cores, but path is not se       t. Cannot continue.
[ERROR] 接收資料時發生錯誤: "init_libretro_symbols()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core symbols..
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cf       g".
[INFO] RetroArch 1.9.13 (Git fca72f67d1)
[INFO] === Build =======================================
[INFO] 容量:  ASIMD
[INFO] Built: Nov 13 2021
[INFO] Version: 1.9.13
[INFO] Git: fca72f67d1
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] [Core]: Frontend is built for dynamic libretro cores, but path is not se       t. Cannot continue.
[ERROR] 接收資料時發生錯誤: "init_libretro_symbols()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core symbols..
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cf       g".
[INFO] RetroArch 1.9.13 (Git fca72f67d1)
[INFO] === Build =======================================
[INFO] 容量:  ASIMD
[INFO] Built: Nov 13 2021
[INFO] Version: 1.9.13
[INFO] Git: fca72f67d1
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] libretro API 版本: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 31 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 6: (1680x1050) 1680 x 1050, 59.883251 Hz
[INFO] [DRM]: Mode 7: (1280x1024) 1280 x 1024, 75.024673 Hz
[INFO] [DRM]: Mode 8: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 9: (1440x900) 1440 x 900, 59.901459 Hz
[INFO] [DRM]: Mode 10: (1280x960) 1280 x 960, 60.000000 Hz
[INFO] [DRM]: Mode 11: (1280x720) 1280 x 720, 60.000000 Hz
[... 246 lines cut ...]
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is already enabled
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Playlist]: 正在讀取歷史檔案: [/storage/.config/retroarch/content_history       .lpl].
[INFO] [Playlist]: 正在讀取歷史檔案: [/storage/.config/retroarch/content_music_h       istory.lpl].
[INFO] [Playlist]: 正在讀取歷史檔案: [/storage/.config/retroarch/content_video_h       istory.lpl].
[INFO] [Playlist]: 正在讀取歷史檔案: [/storage/.config/retroarch/content_image_h       istory.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/content_f       avorites.lpl].
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[libretro INFO] hw/maple/maple_devs.cpp:603 N[MAPLE]: Loaded VMU from file "/sto       rage/savefiles/CVS.A1.bin"
[libretro INFO] hw/maple/maple_devs.cpp:603 N[MAPLE]: Loaded VMU from file "/tmp       /system/dc/vmu_save_B1.bin"
[libretro INFO] hw/maple/maple_devs.cpp:603 N[MAPLE]: Loaded VMU from file "/tmp       /system/dc/vmu_save_C1.bin"
[libretro INFO] hw/maple/maple_devs.cpp:603 N[MAPLE]: Loaded VMU from file "/tmp       /system/dc/vmu_save_D1.bin"
[libretro INFO] hw/pvr/Renderer_if.cpp:104 N[PVR]: Creating Open GL per-triangle       /strip renderer
[libretro INFO] rend/gles/gles.cpp:413 N[RENDERER]: OpenGL version: OpenGL ES 3.       1 Mesa 21.2.5

ToKe79, 2022-04-01:
i just tried nightly 2022-04-01 build with vulkan / gl / glcore - works. probably some settings you have changed in RetroArch / core.
Also please note when you change a driver a restart of RetroArch UI is recommended.

closed 2022-04-01 by ToKe79