libretro/Lakka-LibreELEC #1664: Race Core Crashes when Loading State on GPi case + Pi Zero W/Pi Zero 2 W

smilerliu, 2022-03-26:
I'm using the nightly build for 20220321 for Pi02GPi.

smilerliu, 2022-03-26:
I verified that the same core for Android doesn't have this problem.

ToKe79, 2022-03-27:
can you please provide some logs to see if there is anything that would give any pointers?
see [troubleshooting guide for lakka](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka), best would be [archive created by `createlog` script](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka#automated-collection-of-relevant-logs).

smilerliu, 2022-03-28:
What I did was just to start the game and load state 0. Log:

***
[INFO] [Core]: Using content: "/storage/roms/ngp/Metal Slug - 1st Mission.ngp".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
[INFO] RetroArch 1.10.1 (Git 4a53c67)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Mar 13 2022
[INFO] Version: 1.10.1
[INFO] Git: 4a53c67
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/race_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/RACE/RACE.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/RACE/ngp.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/RACE/Metal Slug - 1st Mission.cfg".
[INFO] [Environ]: GET_VFS_INTERFACE. Core requested version >= V1, providing V3.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CONTENT_INFO_OVERRIDE.
[INFO] [Content Override]: File Extension: 'ngp' - need_fullpath: FALSE, persistent_data: FALSE
[INFO] [Content Override]: File Extension: 'ngc' - need_fullpath: FALSE, persistent_data: FALSE
[INFO] [Content Override]: File Extension: 'ngpc' - need_fullpath: FALSE, persistent_data: FALSE
[INFO] [Content Override]: File Extension: 'npc' - need_fullpath: FALSE, persistent_data: FALSE
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/Metal Slug - 1st Mission.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/Metal Slug - 1st Mission.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Content]: Loading content file: "/storage/roms/ngp/Metal Slug - 1st Mission.ngp".
[INFO] [Content]: Did not find a valid content patch.
[INFO] [Environ]: RETRO_ENVIRONMENT_GET_GAME_INFO_EXT.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_AUDIO_BUFFER_STATUS_CALLBACK.
[INFO] [Environ]: SET_MEMORY_MAPS.
[INFO] [SRAM]: Skipping SRAM load.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 43704.75 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: "egl_videocore".
[INFO] [GL]: Detecting screen resolution: 320x240.
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 320x240.
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: "freetype".
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event* and /run/udev/?
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 444 frames
[INFO] ALSA: Buffer size: 1411 frames
wifi is already enabled
[INFO] [Display]: Found display driver: "gl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_MINIMUM_AUDIO_LATENCY.
[INFO] [State]: Loading state "/storage/savestates/Metal Slug - 1st Mission.state", 51912 bytes.
[INFO] [State]: Saving state "RAM", 51912 bytes.
Segmentation fault
***

ToKe79, 2022-03-28:
did this happen when you tested the core on GPICase + Pi Zero in https://github.com/libretro/Lakka-LibreELEC/issues/1638? Is this specific to the Zero 2 GPICase build? Can you test with RPi3.aarch64 build without the GPICase mod?
Or maybe did the cleanup help somewhat - https://github.com/libretro/Lakka-LibreELEC/issues/1666 - https://nightly.builds.lakka.tv/members/vudiq/pi02gpi_cleanup/?

smilerliu, 2022-03-28:
The core behaves the same with GPi case + Pi Zero W, so I don't think the cleanup helps.

I tried with the kernel from the cleanup build and it behaves the same.

smilerliu, 2022-03-28:
I don't have the accessories to use Pi Zero W outside of the GPI case (no mini hdmi to hdmi connector), so I can't test the rpi3 aarch64 build.

smilerliu, 2022-03-28:
I didn't test the state loading/saving for #1638.

smilerliu, 2022-03-28:
Is there an easy way to build the core for 3.7? I can try it there to see whether the crash is related to the transition to 4.x.

ToKe79, 2022-03-28:
Here: https://nightly.builds.lakka.tv/members/vudiq/pi02-3.7.1-test/

smilerliu, 2022-03-28:
It worked briefly at the beginning, but if after I played a bit more and did a few save states, it started to fail in the same way as 4.x.

I suspect this is a core problem and not a Lakka problem... Will do more testing on other platforms.

smilerliu, 2022-03-29:
Tried more with retroarch for Android and retroarch for rg350, and none of them have this problem. So it seems to be a specific lakka for rpi problem.

ToKe79, 2022-04-03:
Tested on Generic and RPi4 (both 64-bit) on latest nightly (equiv. to 4.1-rc), the issue is not present.

Where do you store the savestates? Do savestates work in other cores?

smilerliu, 2022-04-08:
Tried latest 3.x nightly on an rg351p and got the same crashing behavior. So this is not an rpi only thing. Wondering whether there are more compilation flags to tweak...

smilerliu, 2022-04-17:
Here's the original report about the race core being added to libretro: https://www.libretro.com/index.php/libretro-core-progress-report-december-31-2019/

It does mention something about potentially having savestate and netplay issues.

Can we get a build of the core without DRZ80 so I can test it (I don't have the ability to build Lakka stuff)? I looked at the makefile in the libretro race core repo and it seems DRZ80 is only turned on for the vita platform. DRZ80 is not even on for the miyoo platform, which has a much weaker CPU (and I suspect the performance would really such there).

ToKe79, 2022-04-19:
Hi @smilerliu,
I rebuild the core - modified the package to **not** add `DRZ80=1` to make options for arm platforms.
https://nightly.builds.lakka.tv/members/lakkabuildbot/race/
Let me know if it helps.

smilerliu, 2022-04-20:
Confirmed that the core without DRZ80=1 works in terms of loading and saving.

The core runs slower than with DRZ80=1; on GPICase.arm, for several games I tried (Metal Slug 1st Mission, Metal Slug 2rd Mission, King of Fighters R-2, Samurai Shodown 2), it runs at close to 60FPS (mostly >59FPS) if I set the power management to "manual" (i.e., the CPU runs between 700MHz and 1Ghz).

IMHO savestate working is more important than running at full speed and near-full speed is good enough. So I'd prefer to turn off DRZ80 for the core.

smilerliu, 2022-04-20:
I just tried the core with DRZ80 on again, and it actually doesn't run at full speed either. So now I definitely prefer to turn DRZ80 off.

ToKe79, 2022-04-21:
Thank you for your time and tests!

commit c96aa36f (2022-04-21): race: remove DRZ80 from make options

does not improve performance, creates issues with save states
closes #1664
commit bbdcbea4 (2022-04-21): race: remove DRZ80 from make options

does not improve performance, creates issues with save states
closes #1664
closed 2022-04-21 by ToKe79