libretro/Lakka-LibreELEC #1666: pi02gpi code cleanup

smilerliu, 2022-03-26:
The pi02gpi support was original provided by Cicirifu. I had a chat with Cicirifu and learned that some of the code was unnecessary and may be even harmful:

- Unused dts files should be removed:
https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/projects/RPi/devices/Pi02GPi/config/dpi24-gpi-overlay__.dts
https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/projects/RPi/devices/Pi02GPi/config/dpi24-gpi-overlay___.dts

- Patches that only contain code for temporary experimentation should be removed ("gpi copy.patch" seems unused anyway):
https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/projects/RPi/devices/Pi02GPi/patches/linux/gpi.patch
https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/projects/RPi/devices/Pi02GPi/patches/gpi%20copy.patch

I'll be happy to test any build that has these files removed.

ToKe79, 2022-03-27:
all files except gpi.patch are not used in the build. the patch is probably adding stuff for the GPICase, so it might be useful.
@Cicirifu can you, please, confirm?

ToKe79, 2022-03-27:
@smilerliu you can test [this image](https://nightly.builds.lakka.tv/members/vudiq/pi02gpi_cleanup/) based on https://github.com/libretro/Lakka-LibreELEC/tree/lakka4-pi02gpi_cleanup

Cicirifu, 2022-03-27:
The kernel patch was to add DPI display support for the VC4 driver. Since the build is not using VC4 KMS/DRM, it is unnecessary.

smilerliu, 2022-03-27:
I downloaded just the kernel and replaced my 20220321 build's kernel with it. Then I tried a game from each of the systems I have (gb, gba, md, neogeo, neogeo pocket, psx) and I had zero issues.

Is this enough testing for the kernel patch in question? If not, I can flash a brand new image and test again.

smilerliu, 2022-03-27:
Flashed the full image to a new card and tried a few games. Didn't notice any issue.

ToKe79, 2022-03-28:
closing with 288ebe24bc6648e6a82a370e779efd45b3842b16

closed 2022-03-28 by ToKe79