libretro/Lakka-LibreELEC #1667: Genesis Plus GX - Only the "Wide" variant available

jhpgcj, 2022-03-28:
Hello!

I'd like to report that, as of version 4.0, the regular Genesis Plus GX core is missing from Lakka, while only the Wide variant core remains. Browsing through the available cores, I can see that the regular Genesis Plus GX is "available", but in name only – loading a content reveals it's the Wide core instead. Everything was normal up to version 3.7, but that changed after I upgraded to version 4.0 using the " online updater" option.

- Which version of Lakka are you using?
  Version 4.0, aarch64.
- What system hardware are you using?
  Raspberry Pi 4.
- What did you do?
  Used the "online updater" to upgrade Lakka from version 3.7 to 4.0.
- What did you expect to happen?
  To keep the regular Genesis Plus GX core present in the OS.
- What happened instead?
  Regular Genesis Plus GX core was replaced by the Genesis Plus GX Wide core, even though all name references still point to the regular one.

ToKe79, 2022-03-28:
Try resetting the core association with the game in playlist.

ToKe79, 2022-03-28:
ok, found the error in the package.mk - the wide core was copied to the image with the "regular" name. fixed in #1661

jhpgcj, 2022-03-28:
Thank you for helping!

PR #1661 (2022-03-28): libretro update
- update RetroArch to 1.10.2 and drop patch merged upstream
- RetroArch: disable core system files downloader (included in image)
- update almost all cores to their latest version
- drop duckstation core
- modify libretro_update.sh script to allow for updates to latest tagged commit
- modify some packages to use latest tag during automated update
- update liblcf to 0.7.0 (EasyRPG dependency) and drop patches
- do not rename `genesis_plus_gx_wide_libretro.so` to `genesis_plus_gx_libretro.so` in the install path

fixes #1667
closed 2022-03-28 by jhpgcj